PixlOne/logiops #76: Problems with hires setting

muesli, 2020-06-15:
Since a recent Arch update, every time I reboot my machine, vertical scrolling with the wheel is super slow. I have to stop `logid`, edit the config file and set `hires` to `false`, then restart `logid` again to fix it. However, after a while (sometimes an hour or more) suddenly my scrolling will become super fast. Now I have to go back to the config, change `hires` to `true` again to fix it.

I can't make any sense of it, hope you got an idea. Running latest Arch and latest logiops from git.

PixlOne, 2020-06-15:
Is this only happening over Bluetooth? Linux 5.6 added support to some Bluetooth Logitech devices on hid-logitech-hidpp which means that scrolling now acts differently on those devices.

muesli, 2020-06-15:
No, I'm exclusively using the proprietary Logitech receiver.

zv0n, 2020-06-17:
I'm also experiencing some weird hi-res behavior. For me sometimes after I reboot, the scrolling is super fast. Turning the mouse off and on solves the problem (hooray for the oldest, most reliable solution to almost all problems! :D). If I had to guess, I'd say the mouse connects before logid properly starts, remembers that it's supposed to use hi-res, and when logid starts it adds another hi-res layer to that => super fast scrolling. This is purely a guess, I have no evidence to back it up, so feel free to disregard this.

myralchista, 2020-06-18:
Is there a way to make hires scroll not so fast inside this program? I found solution for solaar, but not for this program:
https://rhtenhove.nl/blog/hires-scrolling-logitech/hires-scrolling-logitech/

PixlOne, 2020-06-18:
@myralchista That's a udev rule so the same method should work with logid.

I'm also noticing the same issue occassionally although I don't quite know what triggers it. I'll try to investigate this further.

vredez, 2020-06-18:
I'm facing the same issue with a MX Master 3.
Another (annoying) workaround is to replug the connector.

vredez, 2020-06-25:
Update: I've found a better workaround that works for my needs, which is to completely remove the 'hires' section from the configuration file. Although a warning is shown regarding the missing section, the scrollwheel speed has been stable now for three days. This might give a hint where to track down the root cause of this issue.

Qwerty-Space, 2020-07-10:
> I'm also experiencing some weird hi-res behavior. For me sometimes after I reboot, the scrolling is super fast. Turning the mouse off and on solves the problem (hooray for the oldest, most reliable [...]

Similar issue here, except my scrolling is SUPER slow regardless of whether I have hires on or off, after rebooting.

Turning the mouse of and of fixed it though.  Thank you

muesli, 2020-12-22:
Fwiw, this issue has resolved itself here. Since it wasn't really reproducible I'm not sure if it was a logid or a kernel update. Closing this ticket hence.

closed 2020-12-22 by muesli