PixlOne/logiops #85: How to know if it is logid is working properly?

vitaminace33, 2020-06-28:
Hi there,

I've compiled, installed (symlinked `/usr/lib/libhidpp.so` to `/usr/local/lib/libhidpp.so`), copied [a basic .cfg](https://wiki.archlinux.org/index.php/Logitech_MX_Master#Examples) to /etc, and started the daemon. Everything seems to work smoothly except that it doesn't. Whatever I write in the configuration file seems to have no effect. I have written the following config file for testing purposes.
```
devices: ({
	name: "Logitech Wireless Mouse MX Master";

	smartshift: { on: true; threshold: 20; };
	hiresscroll: { hires: false; invert: false; target: false; };
	dpi: 1500;

	buttons: (
		{ cid: 0x50; action = { type: "Keypress"; keys: ["KEY_0"]; }; },
		{ cid: 0x51; action = { type: "Keypress"; keys: ["KEY_1"]; }; },
		{ cid: 0x52; action = { type: "Keypress"; keys: ["KEY_2"]; }; },
		{ cid: 0x53; action = { type: "Keypress"; keys: ["KEY_3"]; }; },
		{ cid: 0x56; action = { type: "Keypress"; keys: ["KEY_6"]; }; },
		{ cid: 0xc3; action = { type: "Keypress"; keys: ["KEY_A"]; }; },
		{ cid: 0xc4; action = { type: "Keypress"; keys: ["KEY_B"]; }; }
	);
});
```
I expected to get keystroke from the mouse buttons but nothing has changed. What I am doing wrong? How to know if it is logid is working properly?

I have a *Logitech Wireless Mouse MX Master* on Ubuntu 20.04 (fully updated) with Solaar 1.0.1.

BruceDLong, 2020-06-30:
I'm not an expert on this software, but I found that I had to turn the mouse off then on again.

vitaminace33, 2020-06-30:
Switching on/off has no effect. Thx for the idea, though.

Service is enabled and daemon is working from boot 'tll shutdown.

maxikd, 2020-07-12:
Check the name of your device when running `logid -v`.
For example, mine shows `[INFO] MX Master 3000 detected: device 1 on /dev/hidraw3`, then I have created a config file using "MX Master 3000" as "name".

stnieder, 2020-07-12:
I do have the same problem. For me there is no error showing up, that it doesnt work.
Checking  `logid -v` got me the following output:
```
[INFO] Found Logitech USB Receiver on /dev/hidraw7
[DEBUG] Receiver on /dev/hidraw7: Device 1 paired
```

I tried to change the name of the device to `Logitech USB Receiver`, `Receiver` and also to `Device 1`. But nothing seems to work properly. Was anybody able to solve this problem?

maxikd, 2020-07-20:
> I do have the same problem. For me there is no error showing up, that it doesnt work.
> Checking `logid -v` got me the following output:
>
> ```
> [INFO] Found Logitech USB Receiver on /dev/hidraw7
> [DEBUG] [...]

Is your device paired? Mine shows these exact same lines, plus on line showing the device:
`maximilian@desktop-max:~$ sudo logid -v
[INFO] Found Logitech USB Receiver on /dev/hidraw3
[DEBUG] Receiver on /dev/hidraw3: Device 1 paired
[INFO] MX Master 3000 detected: device 1 on /dev/hidraw3
`

gvasco, 2020-10-20:
I've found the scrollwheel to be a good indicator of when logid is working properly and my custome config working.

If running logid as a service vis systemd (which I assume you are) running "systemctl status logid" provides some debug info if you're having troubles with your config.

If you make any changes to the config or are having issues with your hardware, simply restarting the service should load the new config and sort out any buggy behaviour.

StefanLobbenmeier, 2020-10-24:
In addition to checking the logs I would recommend at first starting it yourself from the terminal - the output is very helpful as it has error messages when the cfg is not found, a device is found, devices could not be read etc.

Ashark, 2020-11-30:
I have a very similar problem. When I run logid -v, I got the following fragment:
```
[INFO] Device found: MX Vertical Advanced Ergonomic Mouse on /dev/hidraw0:4
[DEBUG] /dev/hidraw0:4 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] 0xfd | YES     |         | YES        | YES
```
This output shows all possible values, correct?

I have created a simple config like this:
```
devices: (
{
    name: "MX Vertical Advanced Ergonomic Mouse";

    buttons: (
        {
            cid: 0x??;
            action =
            {
                type: "Keypress";
                keys: ["KEY_A"];
            };
        }
    );
}
);
```
and instead of 0x?? I tried all from above, i.e 0x50, ..., 0xfd and relaunched logid every time. But when pressing thumb button, there is no any typing of "A". Am I doing something wrong?

Also, is there something similar to `xev` but for logiops? I want to press a buttons on mouse and see which events are detected with the Control IDs shown.

gvasco, 2021-01-14:
> I tried all from above, i.e 0x50, ..., 0xfd and relaunched logid every time. But when pressing thumb button, there is no any typing of "A". Am I doing something wrong?

Have you checked the logs or checked the output of systemctl status? Might be an issue with your config, I remeber having issues with my configuration because of missing comas or semicolons.

clappingmonkey, 2021-07-19:
Exactly. After you change your config file use: `sudo systemctl restart logid.service` to restart the deamon and then  use `sudo systemctl status logid.service` to print the status of the services. If everything is ok you'll see at the bottom that only the deamon was restarted if there is a problem you'll see that: 

> 10:45:47 TheMachine systemd[1]: Started Logitech Configuration Daemon.

> 10:45:47 TheMachine logid[24737]: [ERROR] Parse error in /etc/logid.cfg, line 96: syntax error

How you see even the line where the syntax error is is included.

Zanibali, 2024-09-29:
no error