PixlOne/logiops #86: Mx Master 2S compability

ding-ma, 2020-06-30:
Logid is not detecting all the possible buttons on my Mx master 2s. When I run `logid -v`, the CID of the left and right scroll are not showing up.

```
➜  ~ sudo logid -v
[INFO] Found Logitech USB Receiver on /dev/hidraw0
[DEBUG] Receiver on /dev/hidraw0: Device 1 paired
[INFO] MX Master 2S detected: device 255 on /dev/hidraw6
[DEBUG] Available CID: 0x50
[DEBUG] Available CID: 0x51
[DEBUG] Available CID: 0x52
[DEBUG] Available CID: 0x53
[DEBUG] Available CID: 0x56
[DEBUG] Available CID: 0xc3
[DEBUG] Available CID: 0xc4
[DEBUG] Available CID: 0xd7
[INFO] MX Master 2S detected: device 0 on /dev/hidraw6
[DEBUG] Available CID: 0x50
[DEBUG] Available CID: 0x51
[DEBUG] Available CID: 0x52
[DEBUG] Available CID: 0x53
[DEBUG] Available CID: 0x56
[DEBUG] Available CID: 0xc3
[DEBUG] Available CID: 0xc4
[DEBUG] Available CID: 0xd7
[DEBUG] Time out on default device: /dev/hidraw1 (possibly asleep)
```

This is my `logid.cfg` config file located at `/etc/logid.cfg`. I know my gesture button works since it is being detected by logid but my right and left scroll are not. Any help?
```
devices: ({
  name: "MX Master 2S";

  // A lower threshold number makes the wheel switch to free-spin mode
  // quicker when scrolling fast.
  smartshift: { on: true; threshold: 20; };

  hiresscroll: { hires: false; invert: false; target: false; };

  // Higher numbers make the mouse more sensitive (cursor moves faster),
  // 4000 max for MX Master 3.
  dpi: 1500;

  buttons: (
  
    //right scroll
    { cid: 0x5d; action = {
      	   	 	type: "Keypress";
			keys: ["KEY_LEFTCTRL", "KEY_TAB"];
      			};
    
    },

    //left scroll
    { cid: 0x5b; action = {
      	   	 	  type: "Keypress";
			  keys:["KEY_LEFTCTRL", "KEY_LEFTSHIFT", "KEY_TAB"];
      			  };

    },

    //gesture button
    { cid: 0xc3; action = {
      	   	 	  type: "Gestures";
      	   	 	  gestures:(
			   {
			   direction: "None";
			   mode: "OnRelease";
			   action = {
			   	  type: "Keypress";
				  keys: ["KEY_LEFTCTRL", "KEY_T"];
			   	  };
			   }
			  );
      	   	 	}

    }

  );
});
```

LuisMercadoS, 2020-07-01:
I'm having the same issue.

jneuhauser, 2020-07-01:
There is no cid 0x5b on MX Master 2S.

This are the available cids:

```
$ sudo logid -v
...
[INFO] MX Master 2S detected: device 1 on /dev/hidraw2
[DEBUG] Available CID: 0x50
[DEBUG] Available CID: 0x51
[DEBUG] Available CID: 0x52
[DEBUG] Available CID: 0x53
[DEBUG] Available CID: 0x56
[DEBUG] Available CID: 0xc3
[DEBUG] Available CID: 0xc4
[DEBUG] Available CID: 0xd7
```

ding-ma, 2020-07-01:
Yep exactly. Do you know if there is a fix? It is missing 0x5b and 0x5d

drahoja9, 2020-07-01:
https://github.com/PixlOne/logiops/issues/25 :)

PixlOne, 2020-07-04:
The horizontal scroll wheel does not have a CID, the reporting is accurate. The 0x5b and 0x5d CIDs you see in the wiki was just from what I've researched, it does not necessarily apply to all devices.

As far as I know, there is no way of remapping the horizontal scroll wheel via HID++ (and therefore logiops, currently); use xbindkeys for now. Eventually, I may add native support for rebinding the horizontal scrollwheel in logiops.

ding-ma, 2020-07-04:
> The horizontal scroll wheel does not have a CID, the reporting is accurate. The 0x5b and 0x5d CIDs you see in the wiki was just from what I've researched, it does not necessarily apply to all [...]

You have a small walkthrough to develop it? I'd be interested to work on it.