PixlOne/logiops #89: Profiles

SebastianLuebke, 2020-07-05:
Would be nice if we could have and easily switch between multiple profiles per device. Maybe by mapping a button to switch.
or profiles per application

PixlOne, 2020-07-05:
This feature's actually already in my plan, the main thing I'm concerned about right now is how it would affect compatibility with older config files.

Originally, the config file scheme I created was temporary and was subject to change. However, since this program now has an active userbase, I need to try to avoid breaking compatibility with older config files.

I'll think of a solution eventually, but this feature will come sometime after the refactor (and possibly IPC feature) is complete.

Qwerty-Space, 2020-07-12:
perhaps have a different config file per profile.

So, `logid.conf` is the default, and then `firefox.conf` might have a line in it for what `wm_class` to affect

closed 2022-12-30 by SebastianLuebke