PixlOne/logiops #90: mx master 3 buttons

SebastianLuebke, 2020-07-05:
logid -v shows me following buttons are available:
```
[DEBUG] Available CID: 0x50
[DEBUG] Available CID: 0x51
[DEBUG] 0x800 was pressed but no action was found.
[DEBUG] Available CID: 0x52
[DEBUG] Available CID: 0x53
[DEBUG] Available CID: 0x56
[DEBUG] Available CID: 0xc3
[DEBUG] Available CID: 0xc4
[DEBUG] Available CID: 0xd7
```

i cant find any id for the side scroll wheel and the cid in the wiki doesnt work. is there a way to monitor the which buttons i press and show the cid? Or does someone now the cid?

sdueringer, 2020-07-17:
I asked myself the same question and it would be awesome to find a solution for that

Elijas, 2020-07-21:
I control volume with side wheel by editing `~/.xbindkeysrc` file and launching `xbindkeys` application. For example, the command to lower volume is `pactl set-sink-volume @DEFAULT_SINK@ -1200` and <kbd>horizontal left scroll</kbd> is _button 6_, so the `~/.xbindkeysrc` file looks like this:

```
"pactl set-sink-volume @DEFAULT_SINK@ -1200"
  b:6
```

gdiShun, 2020-07-21:
logid doesn't do anything with the scroll wheel (yet?). So it can't help you on that front.

Elijas, 2020-07-21:
> logid doesn't do anything with the scroll wheels (yet?). So it can't help you on that front.

Still, it would be good to have a `side-wheel-tutorial.md` in this repo, since due to protocol/hardware limitations it's never going to be implemented and from our (user) perspective we still want to assign macros to it

gdiShun, 2020-07-21:
I'm not sure it is a limitation. Assuming logid gets it's XY coords from the same place `xinput query-state` queries, the scroll wheel coords are right there with the XY coords. But then again, if it were that simple, I'd imagine they'd have already implemented it.

I don't know about adding a side-wheel tutorial. It feels weird to me for a project to have a tutorial featuring other projects they have no control over. Maybe just links to other tutorials? But I guess to answer your original concern, I use this, but it doesn't work in wine for me.

``` 
"xte 'key XF86AudioRaiseVolume'"
b:7
"xte 'key XF86AudioLowerVolume'"
b:6
```

Elijas, 2020-07-21:
Maybe a separate repo with one file (README) with resources for Logitech
users on linux?

rtowsley, 2020-10-02:
Hello all, today I purchased an MX Master 3, great mouse and I am trying to set up this software. The documentation and example file has provided enough information, except for the thumb wheel. 

I noticed as well that "logid -v" does not list the left scroll and right scroll buttons, however I found in the wiki document there is a section describing the thumbwheel operation. There is no example for thumbwheel in the example configuration, so I am having trouble determining the correct syntax for the thumbwheel section to work. 

I have tried the following as a test, just to see if I could produce the correct syntax, but it is not working. 
```
    thumbwheel:
    {
        divert: true;
        left: 
        {
            type: "Keypress";
            keys: ["KEY_LEFTCTRL", "KEY_F10"];
        };
	right:
        {
            type: "Keypress";
            keys: ["KEY_LEFTCTRL", "KEY_F9"];
        };    
    };
```

Anyone that has a working thumbwheel configuration, I'd love to see an example of the correct syntax.

closed 2022-12-30 by SebastianLuebke