PixlOne/logiops #93: Is the Easy Switch action supported?

VikashKothary, 2020-07-08:
I've got the MX Master. The Easy Switch button (that allows me to switch between channels) is below the mouse.
I want to map it to a different button to make it easier to switch between them. 

Is this supported by this library? Is it even possible?

PixlOne, 2020-07-08:
It's possible, it just isn't supported at this time. Since this isn't terribly hard to implement, I could do it sometime in the future.

VikashKothary, 2020-07-12:
Hi @PixlOne, that's good to hear.

I'm not familiar with C++ but I'd like to help. Is there anything I can do?

I've been having a look and it seems the first thing is to add to the HID++ library? Then create a class similar to [this one](https://github.com/cvuchener/hidpp/blob/master/src/libhidpp/hidpp20/IAdjustableDPI.cpp)? Do you have any information on reverse engineering hardware drivers?

PixlOne, 2020-07-12:
[Someone recently documented the "Easy Switch" (Change host) feature,](https://drive.google.com/drive/folders/0BxbRzx7vEV7eWmgwazJ3NUFfQ28) so all that really needs to be done now is to implement a feature interface and add an action for switching hosts.

This shouldn't be terribly hard to do and if you want to work on this, base it on the drop-libhidpp branch and look at backend/hidpp20/features/SmartShift.h for an example on how to create a feature interface and actions/ChangeDPI.h on how to create an action.

Klap-in, 2020-07-14:
Btw, that info from the google drive is also used to create readable documentation in https://github.com/Logitech/cpg-docs . Helpful documentation can be requested there as well.

PixlOne, 2020-07-15:
I just implemented this feature in 0928b39. See `ChangeHost` in the wiki for details on how to configure it.

closed 2020-07-15 by PixlOne