PixlOne/logiops #94: Wiki update for keyboard layout

jfk88, 2020-07-08:
Great work. 
If I'm not mistaken, the "KEYPRESS" function is bound to a US (QWERTY) layout. Is this correct?
Perhaps you could precise this in the Wiki.

PixlOne, 2020-07-09:
There isn't really a keyboard layout bound to the keypress function, it just uses input event codes.

jfk88, 2020-07-09:
OK let me expose more clearly what I mean. My layout is swiss (QWERTZ):

```
$ localectl status
   System Locale: LANG=fr_CH.UTF-8
       VC Keymap: fr_CH
      X11 Layout: n/a
```

Then when I press "Y" (US "Z") I got:

`KeyRelease event, serial 40, synthetic NO, window 0x3800001,
    root 0x1b3, subw 0x3800002, time 9900369, (41,24), root:(41,1140),
    state 0x10, keycode 52 (keysym 0x79, y), same_screen YES,
    XLookupString gives 1 bytes: (79) "y"
    XFilterEvent returns: False`

Now here is part of my config using "Y":

`{direction: "Left"; mode: "OnRelease"; action = {type: "Keypress"; keys: ["KEY_LEFTSHIFT", "KEY_Y"];}}`

From this I got:

```
KeyPress event, serial 42, synthetic NO, window 0x3c00001,
    root 0x1b3, subw 0x0, time 10558966, (81,79), root:(81,1195),
    state 0x10, keycode 50 (keysym 0xffe1, Shift_L), same_screen YES,
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 42, synthetic NO, window 0x3c00001,
    root 0x1b3, subw 0x0, time 10558966, (81,79), root:(81,1195),
    state 0x11, keycode 29 (keysym 0x5a, Z), same_screen YES,
    XLookupString gives 1 bytes: (5a) "Z"
    XmbLookupString gives 1 bytes: (5a) "Z"
    XFilterEvent returns: False
```
In other words, when "KEYPRESS" fires "Y" my system takes this as a "Z". 

Don't you think there's something to say about this in your instructions?

PixlOne, 2020-07-09:
Sorry, I use a US keyboard, I didn't really know that was the case; this wasn't exactly noted in any documentation on libevdev or the linux-input-codes.h header. I would imagine X11 remaps the evdev device from a US layout to whatever X11 locale is set.

I'll update the wiki, thanks.

closed 2020-07-09 by PixlOne

jfk88, 2020-07-14:
Sorry to bother you with this again but I had a look at your comment and it's not exactly what happens. You wrote:

`Please note that these event codes work in a US (QWERTY) keyboard layout. If you have a locale set that does not use this keyboard layout, please map it to whatever key it would be on a QWERTY keyboard. (e.g. "KEY_Z" on a QWERTZ layout should be "KEY_Y")`

But it's not really a question of the function being bound to a certain layout. It's apparently a common (default?) keymap with XKB that anyone can modify on the system (but it's not really known that's why I suggested to append your wiki). I'll be honest I'm trying myself to figure out why the keymap is built that way and how to change it but the documentation is really messy.

PixlOne, 2020-07-14:
Yeah, what I suspected was that evdev used the US keyboard layout and then XKB converted that
 into your locale.

IIRC the input devices go something like hid driver -> evdev -> xkb (I may be oversimplying here). Usually keyboard locales are defined at the xkb level.