PixlOne/logiops #96: Delay switching from keyboard X mouse and mouse X keyboard

maxikd, 2020-07-12:
I have created a config file with settings that I want, started the logid from terminal, and used it fine, but I experience a delay of about 2 secs when I press a configured/special button on the mouse, if I'm using the keyboard, and vice versa.
My keyboard is a Microsoft Wireless Desktop 2000 and my mouse is a MX Master 3, using Logitech's USB adapter.
I'm currently running Pop!_OS 20.04. 
Any ideas on what may be going on?
If any other info is needed, I can provide.

maxikd, 2020-07-19:
I found out that this bug was way beyond logid. I managed to kind of fix it, and logid works fine!

closed 2020-07-19 by maxikd

yizdu, 2020-11-24:
Similar problem, I found that sometimes, when the keyboard is pressed, the system UI will not respond, and then wait for more than ten seconds for the UI to respond. The mouse and UI works fine when keyboard lagging. Wondering is Pop!_os or logid cause the problem.

yizdu, 2020-12-04:
> Similar problem, I found that sometimes, when the keyboard is pressed, the system UI will not respond, and then wait for more than ten seconds for the UI to respond. The mouse and UI works fine when keyboard lagging. Wondering is Pop!_os or logid cause the problem.

Well, I found out that it was an ibus problem

carvemerson, 2021-07-16:
@maxikd can you share with us, how did you fixed it? I'm getting the same problem on Ubuntu 20.04.

maxikd, 2021-08-06:
@carvemerson I think it was the Scroll Lock key on brazillian-portuguese keyboard layout. I managed to "disable" it and I think that this is what fixed this issue.

carvemerson, 2021-08-06:
Thank you @maxikd, it worked fine. 
Here is the [link](https://askubuntu.com/questions/1041335/ubuntu-18-04-input-slow-when-portuguese-brazil-layout-among-keyboard-input-sou) to disable this key:

maybe, can help someone else.