PixlOne/logiops #97: Windows/MacOS computability?

VikashKothary, 2020-07-12:
Hi,

Can this driver work on either Windows or MacOS machines?

If someone has done it before, could they provide a list of dependencies. I'll be trying it soon, so if I manage to get it working I'll also add it to this issue. I think it'll be nice to add it to the dependencies.

barolo, 2020-07-12:
What would be the point of that?

PixlOne, 2020-07-12:
I was thinking about supporting multiple platforms (more-so for BSD and other OSes) later. This isn't terribly hard as a task, all that really needs to be changed across OSes are the RawDevice, DeviceMonitor and InputDevice classes (on the drop-libhidpp branch).

My main issue with this right now is that I'm using DBus for the IPC feature which isn't really portable. Hopefully I'll be able to make it modular enough that a DBus backend can easily be exchanged with a different one for a different platform.

VikashKothary, 2020-07-12:
@barolo There is a limitation in the official Logitech driver, I'm hoping this software can solve. See https://github.com/PixlOne/logiops/issues/93.

@PixlOne Is this true for MacOS too? Would this work: https://formulae.brew.sh/formula/dbus?

I can see a issue regarding Windows compatibility here: https://github.com/cvuchener/hidpp/issues/5

barolo, 2020-07-12:
oh we're workin on #93 in solaar right now, we have the docs and working pull. It's truly annoying with mice

Flacedoo, 2020-12-05:
Would highly appreciate this as well - it's really annoying that you made a driver that's capable of switching the device without picking it up, and the official driver can't. Why are we paying so much to Logitech again? :D

CiriousJoker, 2020-12-11:
Tbh though, a community Windows driver would be pretty good since the official one has huge flaws with [sidescrolling](https://github.com/microsoft/vscode/issues/84136) in programs running inside an Electron (chrome) container as well as general scroll performance on Windows.

ArtisticZhao, 2021-09-06:
@Flacedoo Hi, 
I have built the [hidpp](https://github.com/cvuchener/hidpp) on Windows.
And you can call the change host feature by hidpp tools.
You can also remap the button in Logitech Options to call the hidpp tools to change the host.