PixlOne/logiops #98: Gesture mode does not pause mouse input and gestures don't work

tboudreaux, 2020-07-12:
Gestures are not working on my computer, this may be due to gesture not pausing mouse input. I am using a first generation MX Master and running Manjaro on 5.6.16, my logid.cfg file is copied below. I have also tested the config file from issue #9 and gestures also do not work with that config file. Additionally, I have tested the modes "OnFewPixels" and "Axis", with the same null result (gestures do not seem to do anything, and mouse input is not paused). It should be noted that when direction is "None" the action is preformed (in my config below for example pressing the thumb button does sent alt-s). Any help would be appreciated, thanks.

```
devices: (
{
    name: "MX Master";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_A"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_B"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_C"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_D"]
                        }
                    },
                    {
                        direction: "None"
                        mode: "OnRelease";
                        action =
                        {
                        	type: "Keypress";
                        	keys: ["KEY_LEFTALT", "KEY_S"];
                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "ToggleSmartshift";
            };
        },
	{
	    cid: 0x57;
	    action =
	    {
		type: "Keypress";
		keys: ["BTN_FORWARD"];
	    };
	}
    );
}
);
```

PixlOne, 2020-07-15:
Does this issue still occur on the latest commit?

tboudreaux, 2020-07-15:
On the latest commit mouse input does pause when running the service; however, it does not seem to be loading my config file anymore. When running $ /usr/local/bin/logid I get the following output
```
[INFO] Detected receiver at /dev/hidraw3
[INFO] Device Wireless Mouse MX Master not configured, using default config.
[INFO] Device found: Wireless Mouse MX Master on /dev/hidraw3:1
[WARN] Error adding device /dev/hidraw4: std::exception
```
This may be do to me mucking up uninstalling the old version, but I think that unlikely. Just to be clear though I first stopped then disabled the logid service. I the logid executable. I then build the new commit and ran make install. The new executable is where I expect it to be but when I run it I get the above exception.

PixlOne, 2020-07-15:
Yes, that's because device names have changed slightly in this release (it uses hidpp20 Name), try changing `MX Master` to `Wireless Mouse MX Master`. The good news now is that device names will be identical between Unifying and Bluetooth devices.

tboudreaux, 2020-07-15:
Ah yes, thanks. When I change that the mouse movement is no longer paused when the gesture button is depressed, the behavior is seemingly back to what it was on the old version I had installed. I do however now get the following message
```
[WARN] Wireless Mouse MX Master: Cannot divert raw XY movements for CID 0xc3
```

PixlOne, 2020-07-15:
That's odd. Launch logid with `-v DEBUG` and you should get a table like this:
```
[DEBUG] /dev/hidraw3:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
```
Can you post this table?

tboudreaux, 2020-07-15:
```
[DEBUG] /dev/hidraw3:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | 
[DEBUG] 0x53 | YES     |         | YES        | 
[DEBUG] 0x56 | YES     |         | YES        | 
[DEBUG] 0xc3 | YES     |         | YES        | 
[DEBUG] 0xc4 | YES     |         | YES        | 
[WARN] Wireless Mouse MX Master: Cannot divert raw XY movements for CID 0xc3
```

PixlOne, 2020-07-15:
What's the product ID of your MX Master? Perhaps older models use a different mechanism for defining gestures?

As of now, logid does not output the product ID (although I probably should do that), so you can get the product ID by using solaar and clicking the info button on the right side and reading the `Wireless PID` field on the right.

tboudreaux, 2020-07-15:
Solaar gives the Wireless PID of my mouse as 4041

PixlOne, 2020-07-15:
If anyone has an MX Master with the PID 4041, could you get a USB packet capture of the Logitech Options configuration sequence where 0xc3 (Gesture button) is mapped to a gesture?

This will help me further understand the gesture feature on previous iterations of the MX Master.

baldonp1, 2020-07-20:
Hey there! I'm experiencing the same issue. 

I'm on Nvidia Jetpack 4.4 (Ubuntu 18.04 derivative), confirmed my PID is 4041 using solaar. Getting the exact same output as @tboudreaux running `logid -v DEBUG`. 

Would really love to get this fixed. I'm happy to provide a usb packet capture. I have wireshark installed and copied the dissector script. 

I tried launching wireshark with the script as instructed, using `wireshark -X lua_script:hidpp-dissector.lua`, but I have no idea what to do from there. I've never used wireshark before. If you'd like to give me a few steps to follow I'll gladly follow up with the capture.

PixlOne, 2020-07-20:
@baldonp1 Look into [Wireshark's guide on capturing packets from USB devices](https://wiki.wireshark.org/CaptureSetup/USB). I'd like you to capture the packets from a unifying receiver with the 4041 MX Master on Windows/macOS with Logitech Options configured to have gestures on the mouse.

During the packet capture, I want you to turn the mouse off and then on (to grab the configuration packets), and I'd like you to use the gestures. Once you've completed this, post the wireshark pcap as well as an HID++ feature dump (the output of hidpp-list-features from [PixlOne/hidpp](https://github.com/PixlOne/hidpp)).

Please make sure to filter the USB packet capture to only the receiver USB device before posting it. It's not necessary but it helps with readability and file size.

baldonp1, 2020-07-21:
Ah, I'm all set up on Ubuntu. I'll have to work on this a bit later tonight on my mac. 

When I was looking at it on ubuntu I wasn't sure which interface to select (usb0, usbmon0, usbmon1, usbmon2) or which filters to apply. Maybe it'll make more sense once I try again on my mac. I'll follow up if I have any trouble. 

In the meantime, here's the hidpp-list-features output from ubuntu. 

> Logitech USB Receiver (046d:c52b) is a HID++ 1.0 device
> Register 0x00 read   3: 00 09 00
> Register 0x01 read   3: 00 00 00
> Register 0x02 read   3: 00 01 00
> Register 0x03 read   3: 00 00 00
> Register 0xb3 read  16: 0e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> Register 0xb5 read  16: Invalid value (0x03)
> Register 0xd0 read   3: 00 00 00
> Register 0xd4 read   3: 00 00 7c
> Register 0xe6 read   3: 00 00 00
> Register 0xf0 read   3: 00 00 00
> Register 0xf1 read   3: Invalid value (0x03)

PixlOne, 2020-07-21:
Oh, I forgot to mention to select the device index. Sorry, that's the feature dump of the receiver, run `hidpp-list-features -d <device index> <raw device path>` for the feature dump of the MX Master.

baldonp1, 2020-07-21:
Hey there! Attached is the capture. I think I got what you were looking for, not sure if I correctly filtered it. Let me know if anythings missing/if you need me to try again. In the capture I first turned the mouse off/on, gestured up, down, left, right, and then turned it off/on again. 

It doesn't look like the PixlOne/hidpp project is compatible with macOS (no libudev on mac), so I wasn't able to run the feature dump. The output from above is the feature dump from ubuntu. When I ran `hidpp-list-devices` it picked up the receiver on /dev/hidraw6 and mouse on /dev/hidraw6:1. `hidpp-list-features` would only run on /dev/hidraw6, which produced the output in my last post. 

Let me know if there's anything else you need!

[mxmaster4041_cap.pcapng.zip](https://github.com/PixlOne/logiops/files/4956026/mxmaster4041_cap.pcapng.zip)

baldonp1, 2020-07-31:
Just reread your instructions on hidpp-list-features... I missed the device ID. Here's the update output. 

Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x1d4b] Wireless device status
Feature 0x05: [0x0020] Reset
Feature 0x06: [0x1000] Battery status
Feature 0x07: [0x1814] Change host
Feature 0x08: [0x1b04] Reprog controls v4
Feature 0x09: [0x2201] Adjustable dpi
Feature 0x0a: [0x2100] Vertical scrolling
Feature 0x0b: [0x2110] Smart shift
Feature 0x0c: [0x2121] Hi-res wheel
Feature 0x0d: [0x6501] Gesture 2
Feature 0x0e: [0x00c1] DFUcontrol 2
Feature 0x0f: [0x1813] ? (hidden, internal)
Feature 0x10: [0x1830] ? (hidden, internal)
Feature 0x11: [0x1890] ? (hidden, internal)
Feature 0x12: [0x18a1] ? (hidden, internal)
Feature 0x13: [0x18c0] ? (hidden, internal)
Feature 0x14: [0x1df3] ? (hidden, internal)
Feature 0x15: [0x1e00] Enable hidden features (hidden)
Feature 0x16: [0x1eb0] ? (hidden, internal)
Feature 0x17: [0x1803] ? (hidden, internal)
Feature 0x18: [0x1861] ? (hidden, internal)
Feature 0x19: [0x9000] ? (hidden, internal)
Feature 0x1a: [0x9200] ? (hidden, internal)
Feature 0x1b: [0x9240] ? (hidden, internal)
Feature 0x1c: [0x1805] OOBState (hidden, internal)

szbergeron, 2020-08-11:
Am also experiencing this on 4041 and have a bit of time to throw at it, what additional info would be helpful here?

Conmy, 2020-10-10:
I would also like to get this to work with gestures. Is the output from Wireshark still needed?
@szbergeron, @PixlOne Did you make any progress?

RafayAK, 2021-06-14:
Can I help in some way to fast track this?

yash-fn, 2021-06-16:
I also have a 4041 and am happy to run any packet captures.

eth-hall, 2026-10-19:
I am having the same issue still in 2026!