PixlOne/logiops #100: drop-libhidpp/refactor branches: Not working at all

gdiShun, 2020-07-13:
Unsure where to post bugs for this branch. They aren't working at all for me though. Tried drop yesterday and refactor today. Same issues. Some of the errors:

> sudo logid -vV -c /etc/logid.cfg 
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [INFO] Detected receiver at /dev/hidraw2
> [WARN] Device /dev/hidraw4 timed out.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [WARN] Device /dev/hidraw3 timed out.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [WARN] Error adding device /dev/hidraw2: No DJ reports
> [WARN] Error adding device /dev/hidraw4: Device timed out
> Segmentation fault

> sudo logid -v
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [WARN] Device /dev/hidraw4 timed out.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [INFO] Detected receiver at /dev/hidraw2
> double free or corruption (out)
> Aborted

> sudo logid -v
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [WARN] Device /dev/hidraw4 timed out.
> [WARN] Device /dev/hidraw3 timed out.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [WARN] Error adding device /dev/hidraw4: Device timed out
> [WARN] Error adding device /dev/hidraw3: Device timed out

Keeps running, but log messages just stop there.
Running (early)Arch with an MX Master 3 and MX Keys on the receiver. Master configured in /etc/logid.cfg, Keys is not. Doesn't seem like it even loads the config file either way though. I mention I'm running an early install of Arch as perhaps it's missing libraries, although I have everything mentioned in the README. As well as git and dbus-cxx which appear to be necessary to install.

PixlOne, 2020-07-13:
Sorry about that, I was only running logid as a user with udev rules that allow me to access only Logitech devices. It seems that this issue occurs when DeviceManager scans non-Logitech devices. Hopefully this commit should fix it.

There's a slight bug where logid will sometimes hang completely for a few seconds at startup if a device times out. I'll see if I can fix it but realistically, it's not too much of a problem.

gdiShun, 2020-07-13:
That fixed the crashes, but it's still just sitting there doing nothing after the below. Turning off/on doesn't do anything. Unplugging/plugging in the receiver will cause it to re-detect it, but otherwise just sits there.

> [DEBUG] Unsupported device /dev/hidraw0 ignored
> [DEBUG] Unsupported device /dev/hidraw6 ignored
> [DEBUG] Unsupported device /dev/hidraw7 ignored
> [DEBUG] Unsupported device /dev/hidraw1 ignored
> [WARN] Device /dev/hidraw4 timed out.
> [INFO] Detected receiver at /dev/hidraw2
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [WARN] Error adding device /dev/hidraw2: No DJ reports
> [WARN] Error adding device /dev/hidraw4: Device timed out
> [WARN] Device /dev/hidraw3 timed out.
> [WARN] Error adding device /dev/hidraw3: Device timed out

PixlOne, 2020-07-13:
Can I get the product ID of your receiver (use lsusb, it should be something like 046d:c52b). Perhaps some receivers don't support DJ reports?

gdiShun, 2020-07-13:
That's the exact model actually.

> Bus 001 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver

I think the keyboard and mouse came with different ones. I'll see if I can find it.

PixlOne, 2020-07-13:
Hmm, is the `046d:c534` model the one that isn't playing nicely with logid or is it both models? I'm using the `046d:c52b` model and that's working just fine.

Can you post the report descriptors of both receivers? You can run `sudo usbhid-dump -d <vid>:<pid>` to get them (e.g. `sudo usbhid-dump -d 046d:c52b`).

gdiShun, 2020-07-14:
Both of them had the same "No DJ reports" error. I'm using the :c52b model as well. The only thing that did work was Bluetooth. Updated but still getting the same error. Here's the results from usbhid-dump:

> 001:003:002:DESCRIPTOR         1594687644.082342
>  06 00 FF 09 01 A1 01 85 10 75 08 95 06 15 00 26
>  FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
>  A1 01 85 11 75 08 95 13 15 00 26 FF 00 09 02 81
>  00 09 02 91 00 C0 06 00 FF 09 04 A1 01 85 20 75
>  08 95 0E 15 00 26 FF 00 09 41 81 00 09 41 91 00
>  85 21 95 1F 15 00 26 FF 00 09 42 81 00 09 42 91
>  00 C0
>
> 001:003:001:DESCRIPTOR         1594687644.086344
>  05 01 09 02 A1 01 85 02 09 01 A1 00 05 09 19 01
>  29 10 15 00 25 01 95 10 75 01 81 02 05 01 16 01
>  F8 26 FF 07 75 0C 95 02 09 30 09 31 81 06 15 81
>  25 7F 75 08 95 01 09 38 81 06 05 0C 0A 38 02 95
>  01 81 06 C0 C0 05 0C 09 01 A1 01 85 03 75 10 95
>  02 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
>  09 80 A1 01 85 04 75 02 95 01 15 01 25 03 09 82
>  09 81 09 83 81 60 75 06 81 03 C0 06 BC FF 09 88
>  A1 01 85 08 19 01 29 FF 15 01 26 FF 00 75 08 95
>  01 81 00 C0
> 
> 001:003:000:DESCRIPTOR         1594687644.089320
>  05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
>  75 01 95 08 81 02 81 03 95 05 05 08 19 01 29 05
>  91 02 95 01 75 03 91 01 95 06 75 08 15 00 26 A4
>  00 05 07 19 00 2A A4 00 81 00 C0

PixlOne, 2020-07-14:
Ah I see the problem, logid checks if the device supports DJ reports by checking if the report descriptor contains a certain block. While identical in function, the report descriptor of your receiver has two fields swapped, namely the report count and report size fields.
```
0x95, 0x0E,        //   Report Count (14)
0x75, 0x08,        //   Report Size (8)
```

This should be a fairly easy fix, I can just scan for an alternate report descriptor block (i.e. from the one you posted).

PixlOne, 2020-07-14:
@gdiShun Try it out now, it should be working with your receiver.

gdiShun, 2020-07-14:
That did it! Awesome. Thanks.

> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [DEBUG] Unsupported device /dev/hidraw0 ignored
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [INFO] Detected receiver at /dev/hidraw2
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [DEBUG] Unsupported device /dev/hidraw1 ignored
> [DEBUG] All workers were busy for 500ms, running task in new thread.
> [WARN] Error adding device /dev/hidraw4: std::exception
> [INFO] Device MX Keys Wireless Keyboard not configured, using default config.
> [INFO] Device found: MX Keys Wireless Keyboard on /dev/hidraw2:1
> [WARN] Error adding device /dev/hidraw3: std::exception
> [INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw2:2
> [WARN] Line 45: action is a required field, skipping.

PixlOne, 2020-07-14:
Sometimes exceptions are thrown and caught by exception handlers so they should only affect the device/task they're running on. Many of them can safely be ignored. That said, they should probably be handled better in the future.

If the exceptions are the ones you posted (i.e. `[WARN] Error adding device /dev/hidraw4: std::exception`), I'm assuming those are non-Logitech devices, it should be just fine.

closed 2020-07-14 by PixlOne