PixlOne/logiops #101: Fail to install (cmake)

marcosab1991, 2020-07-15:
Hi,

i download the zip file from github and unzip it, made da build file and try to run cmake .., but it failed, i tried ccmake .. and saw this:
 CMAKE_BUILD_TYPE                                                              
 CMAKE_INSTALL_PREFIX             /usr/local                                   
 EVDEV_INCLUDE_DIR                /usr/include/libevdev-1.0                    
 EVDEV_LIBRARY                    /usr/lib/x86_64-linux-gnu/libevdev.so        
 **HIDPP_INCLUDE_DIR                HIDPP_INCLUDE_DIR-NOTFOUND                   
 HIDPP_LIBRARY                    HIDPP_LIBRARY-NOTFOUND**                       
 HID_BACKEND                      linux        

i think the problem running cmake is with the hidpp library, which i cannot find, anyone with the same issue?
running ubuntu here

PixlOne, 2020-07-15:
Someone else had the same issue. The issue here is that the repository uses the hidpp submodule which doesn't work when you download it as a zip.

You can either use the refactor branch as a zip or use git clone to download it.

I'll be pushing an update soon to fix this.

closed 2020-07-15 by PixlOne

marcosab1991, 2020-07-15:
Ok i used git clone and i managed to install, but now when i start logid it gives me this error:
logid: error while loading shared libraries: libhidpp.so: cannot open shared object file: No such file or directory

PixlOne, 2020-07-15:
Try the latest commit, it no longer requires that library.