PixlOne/logiops #102: Cannot recognize MX master 3

InjecT1309, 2020-07-19:
After the newest update logid stopped working for my MX master 3. Journalctl logs:
```
-- Reboot --
lip 19 16:05:04 askl-pc systemd[1]: Started Logitech Configuration Daemon.
lip 19 16:05:06 askl-pc logid[890]: [WARN] Error adding device /dev/hidraw1: std::exception
```

thoastbrot, 2020-07-19:
My MX Master (1) stopped working, too - but I think, you're looking at the wrong line. In my case, hidraw0 is the receiver and why hidraw1 is reported, I don't know.

```
$ sudo /usr/bin/logid -v -c /etc/logid.cfg 
[WARN] Error adding device /dev/hidraw1: std::exception
[INFO] Detected receiver at /dev/hidraw0
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[INFO] Device Wireless Mouse MX Master not configured, using default config.
[INFO] Device found: Wireless Mouse MX Master on /dev/hidraw0:1
[DEBUG] /dev/hidraw0:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | 
[DEBUG] 0x53 | YES     |         | YES        | 
[DEBUG] 0x56 | YES     |         | YES        | 
[DEBUG] 0xc3 | YES     |         | YES        | 
[DEBUG] 0xc4 | YES     |         | YES        | 
```

And it's not doing anything anymore.

thoastbrot, 2020-07-19:
Since my last install (using logiops-git from the aur) is quite ancient (commit 36ish to 152 is a lot of code), I'll try to bisect. v0.1 was working, v0.2 not.

PixlOne, 2020-07-19:
Have you updated your config file? The device names have changed since then due to the use of the hidpp20 device name feature. Try changing `MX Master` to `Wireless Mouse MX Master` in your config file.

InjecT1309, 2020-07-19:
Changed from MX Master 3000 to Wireless Mouse MX Master, did not help.

PixlOne, 2020-07-19:
That's odd, can you post the log output now?

thoastbrot, 2020-07-19:
[INFO] Device found: Wireless Mouse MX Master on /dev/hidraw0:1

Didn't see that as device name, but as description... That's it, thank you!

@InjecT1309 check that Text, it's probably different for your model.

InjecT1309, 2020-07-19:
Its the same as before:
```
lip 19 21:05:46 askl-pc systemd[1]: Stopping Logitech Configuration Daemon...
lip 19 21:05:46 askl-pc systemd[1]: logid.service: Succeeded.
lip 19 21:05:46 askl-pc systemd[1]: Stopped Logitech Configuration Daemon.
lip 19 21:05:46 askl-pc systemd[1]: Started Logitech Configuration Daemon.
lip 19 21:05:48 askl-pc logid[10116]: [WARN] Error adding device /dev/hidraw1: std::exception
```

thoastbrot, 2020-07-19:
@InjecT1309 you need to run it in the shell with -v to get that information.

InjecT1309, 2020-07-19:
Ok, this fixes my issue, the device name was 'Wireless Mouse MX Master 3' (with a 3 at the end)

closed 2020-07-19 by PixlOne