PixlOne/logiops #103: No config change, but now I get a FileIOException

Qwerty-Space, 2020-07-20:
I have tried with the logiops-git and logiops AUR packages
```
[ERROR] I/O Error while reading ~/.config/logiops/logid.cfg: FileIOException
[INFO] Device Wireless Mouse MX Master 2S not configured, using default config.
[INFO] Device found: Wireless Mouse MX Master 2S on /dev/hidraw5:1
```

PixlOne, 2020-07-20:
Are you sure the file is in that path? Can you post the full output and the command you used to run it?

Qwerty-Space, 2020-07-20:
> Are you sure the file is in that path? Can you post the full output and the command you used to run it?

Thanks for the fast reply
It definitely is there

[image]

PixlOne, 2020-07-20:
hmm it seems to be reading that file just fine

Qwerty-Space, 2020-07-20:
It has for some reason stopped saying there's a FileIOException, but it says it's not configured

PixlOne, 2020-07-20:
Change `name: "MX Master 2S";` to `name: "Wireless Mouse MX Master 2S";`

Qwerty-Space, 2020-07-20:
Yeah that worked
I wonder why it was working before then?

PixlOne, 2020-07-20:
The latest update changed how names work in some cases.

Qwerty-Space, 2020-07-20:
Ah I see

Qwerty-Space, 2020-07-20:
After a restart it's happening again

[image]

PixlOne, 2020-07-21:
You can fix this by either using the full path (i.e. substitute `~` for `/home/<user>`) or removing the quotation marks.

When you run a command with the file path in quotation marks, the shell will ignore special characters like `~`. For example, if you try `ls "~"`, ls will respond by saying `ls: cannot access '~': No such file or directory`; if you run `ls ~`, it will read the path as your home folder and properly list the files in your home directory.

closed 2020-07-21 by PixlOne

Qwerty-Space, 2020-07-21:
Oh I didn't know that, thank you