PixlOne/logiops #104: GUI and Logo suggestions

PixlOne, 2020-07-20:
After I write an inter-process communication method for logid and write the logictl CLI front-end, I'm planning on writing a GUI to allow for an easier configuration method. Since I want to make this a fully-fledged Linux app, this project needs a logo as well.

While I currently have some vague ideas for the logo and the GUI, I'd like to ask the community for their input on this topic.

Please feel free to make any suggestions. Concepts for logo and GUI designs will be greatly appreciated.

In addition, I'm unsure about what GUI framework I should use. I've recently been eyeing [MauiKit](https://mauikit.org/) and while I like the look and feel of it, I don't exactly know if I want to use a Qt library; I'm already using GLib for DBus so using GTK will help reduce dependencies. I appreciate any advice on this topic.

Ishan333, 2020-08-03:
MauiKit looks very interesting indeed. Being a KDE user I wouldn't mind a Qt based UI, but that's your call. And GTK seems to make much more sens in your use case.

leifniem, 2020-10-19:
If you don't mind staring into a creepy face i mocked this up in 2 minutes :P

[image]

properlypurple, 2020-10-23:
GTK is a solid choice, since you're already using GLib. In addition, it makes easier to write responsive apps using Libhandy.

aragon999, 2020-12-17:
Please leave also the possibility to build it without a GUI :-)

tomekand1, 2021-01-03:
i think a fully-fledged app is an awesome idea, as there isn't anything like it out there, and i also think that GTK is a good choice, but then why not to try the MauiKit, as it looks more modern

frankk74, 2021-04-19:
Thanks guys for all the work on this project!!!! I have a MX ergo (trackball with scrollwheel) and nothing I could find made it work (xorg/imwheel/solaar)....

Just a quick question seems that running Solaar seems to interfere with logid? But it also seems it has the gui but now the backend isn't working as well as yours? My guess is merging the project might be too much work?

Anyway, thank you!!!

micrenda, 2021-07-28:
It is possible I can help you with a GUI.
There is a reference format for the config file? There are some templates for mouse like MX Everywhere 3? (I have this mouse).
I think, at a first moment, to have a separate development for the GUI (GTK or QT) and then, if it looks ok, to merge it to logiops.

SelfRef, 2021-12-13:
In my opinion choosing an UI framework should be based on stack underneath. In this case there is GLib already used and main target for app is Linux, where GTK works (still in my opinion) the most natively so it sounds like a natural choice. There is also new RAD tool for creating GTK 4/3 views: https://gitlab.gnome.org/jpu/cambalache

MauiKit makes more sense when multi-OS support is a primary goal and/or someone has already much experience with QT.

Greg0, 2022-01-27:
Maybe it's silly but what do you guys think about GUI in Flutter?

ciriousjoker, 2022-02-11:
[Sidekick](https://github.com/leoafarias/sidekick) is a desktop application to manage Flutter installations, written in Flutter.

Sewdohe, 2022-11-20:
I'd be willing to work on a GUI using tauri windowing (rust) and a web framework like Angular. We could change settings using rust RPC or even modify the actual .cfg file from the web frontend with file read/write access. If anyone is interested let me know and we can start a repo

Eko450-eng, 2023-03-17:
@leifniem is it ok if I copy the logo you have created for my app?

leifniem, 2023-03-17:
> @leifniem is it ok if I copy the logo you have created for my app?

Sure, go ahead. Feel free to credit me :) Sadly I do not have the vector file for this anymore, so you might need to reconstruct it if you need higher resolution.

Eko450-eng, 2023-03-17:
@leifniem thanks, and to anyone who's still searching for a GUI https://github.com/Eko450-eng/LogiOpsGui there you go, its not even remotely finished at this point I have spent maybe a total of 5 hours on it probably less
But it is at least working for the MX Master 3S. Feel free to open issues and PRs

NordLex, 2023-10-29:
Hi. I`m writing a graphical shell to control the LogiOps driver. I use gtk4 and libadwaita. I would like to clarify whether it is possible to request the current device settings using d-bus? I`m also interested in the ability to get the status of the device`s battery. Also, is it possible to use “LogiOps” as the name of the application?

NordLex, 2023-12-02:
Here is a link to the repository if you are interested
https://github.com/NordLex/LogiOps

matthewmoonen, 2024-09-25:
Hi all,

I've built a GUI using the Python CustomTkinter library. It's fully functional and working quite well.

https://github.com/matthewmoonen/LogiOpsGui

There are some inherent DPI scaling issues with the GUI library I've used, which make it challenging to achieve a consistent appearance across different monitors, distros, and desktop environments. So it probably won't look as good on your computer as it does on mine.

There are a few things I can do to improve this, such as building custom button classes using SVG files. I've already done this for the radio buttons. Ultimately, however, the best solution will be to rewrite everything in C++. That's my plan, but I'll need to brush up on my C++ skills, so it will take some time.

In the meantime, this tool is pretty effective for creating Logiops configuration files. It certainly speeds up the process as you no longer need to look up things like your device's button CIDs or the max values for various settings.

I’m hoping some of you will try it out and provide feedback. If enough people use it and find it helpful, I'll continue improving this version alongside working on the C++ rewrite. Feel free to reach out with any bugs you encounter or issues getting it to work.

matthewmoonen, 2024-09-25:
As for logos, I've created one using the letter "ტ" from the Georgian alphabet, which looks vaguely like a mouse or the letter "O" in English, but is actually a "T" sound. The word "tech" is spelled "ტექ", so combined, the logo says both "LogiOps" and "Logitech".

[image]