PixlOne/logiops #106: Disable MX Master 3 horizontal scroll

Elijas, 2020-07-21:
I'm currently using horizontal scroll wheel with ".xbindkeysrc" to control volume (Pop OS 20.04) but currently it activates horizontal scroll along also, and I don't want that. Is it possible to disable that?
```
"pactl set-sink-volume @DEFAULT_SINK@ -1200"
  b:6
```

mmsbrggr, 2020-07-22:
I am also using Pop OS 20.04 and have the same issue. Moreover, I mapped the gesture button to the Super Key via xbindkey which doesn't work properly. It seems that when I press the gesture button the system presses the Super key but also the default action which is Alt-Tab I think. Could it be that there is a default driver in the background competing with logiops which we have to remove?

maxikd, 2020-07-23:
I managed to use the thumb button to do gestures and to press the Super key. Check if my config helps you:

```
devices: (
{
    name: "MX Master 3000";
    
    smartshift:
    {
        on: true;
        threshold: 20;
    };
    
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                {
                    direction: "Left";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_LEFT"];
                    }
                },
                {
                    direction: "Right";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_RIGHT"];
                    }
                },
                {
                    direction: "Up";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_UP"];
                    }
                },
                {
                    direction: "Down";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
                        keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_DOWN"];
                    }
                },
                {
                    direction: "None";
                    mode: "OnRelease";
                    action: 
                    {
                        type: "Keypress";
	                keys: ["KEY_LEFTMETA"];
                    }
                }
                );
            };
        },
        {
            cid: 0x0053;
            action =
            {
                type: "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        },
        {
            cid: 0x0056;
            action =
            {
                type: "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        }
    );
}
);
```

PixlOne, 2020-07-29:
So I've just discovered some new documentation that may help with the whole thumb wheel issue.

[The documentation for HID++ 2.0 feature 0x2150 (Thumb wheel) was recently released.](https://drive.google.com/file/d/1E8iT-Vo53wczN8b8cRSeAWaQozqOCdBn/view) If the MX Master 3 has support, I could implement support for changing the thumb wheel via this feature. Sadly, this feature isn't available on my MX Master, so I won't be able to test the implementation, nor will other MX Master users be able to use this.

Anyways, could anyone with an MX Master 3 get a feature dump (using hidpp-list-features over in [PixlOne/hidpp](https://www.github.com/PixlOne/hidpp)) and post it?

Elijas, 2020-07-31:
> Anyways, could anyone with an MX Master 3 get a feature dump (using hidpp-list-features over in [PixlOne/hidpp](https://www.github.com/PixlOne/hidpp)) and post it?

```bash
~/hidpp $ sudo hidpp-list-devices
/dev/hidraw2: Logitech USB Receiver (046d:c52b) HID++ 1.0
/dev/hidraw2 (device 1): MX Master 3 (046d:4082) HID++ 4.5

~/hidpp $ sudo hidpp-list-features /dev/hidraw2
Logitech USB Receiver (046d:c52b) is a HID++ 1.0 device
Register 0x00 read   3: 00 09 00
Register 0x01 read   3: 00 00 00
Register 0x02 read   3: 00 01 00
Register 0x03 read   3: 00 00 00
Register 0xb3 read  16: 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Register 0xb5 read  16: Invalid value (0x03)
Register 0xd0 read   3: 00 00 00
Register 0xd4 read   3: 00 00 7c
Register 0xe6 read   3: 00 00 00
Register 0xf0 read   3: 00 00 00
Register 0xf1 read   3: Invalid value (0x03)
```

PixlOne, 2020-07-31:
> > Anyways, could anyone with an MX Master 3 get a feature dump (using hidpp-list-features over in [PixlOne/hidpp](https://www.github.com/PixlOne/hidpp)) and post it?
> 
> ```shell
> ~/hidpp $ sudo hidpp-list-devices
> /dev/hidraw2: Logitech USB Receiver (046d:c52b) HID++ 1.0
> /dev/hidraw2 (device 1): MX Master 3 (046d:4082) HID++ 4.5
> 
> ~/hidpp $ sudo hidpp-list-features /dev/hidraw2
> Logitech USB Receiver (046d:c52b) is a HID++ 1.0 device
> Register 0x00 read   3: 00 09 00
> Register 0x01 read   3: 00 00 00
> Register 0x02 read   3: 00 01 00
> Register 0x03 read   3: 00 00 00
> Register 0xb3 read  16: 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> Register 0xb5 read  16: Invalid value (0x03)
> Register 0xd0 read   3: 00 00 00
> Register 0xd4 read   3: 00 00 7c
> Register 0xe6 read   3: 00 00 00
> Register 0xf0 read   3: 00 00 00
> Register 0xf1 read   3: Invalid value (0x03)
> ```

That's a feature dump of the receiver, you can get a feature dump of the MX Master 3 by running `sudo hidpp-list-features -d 1 /dev/hidraw3`

Elijas, 2020-07-31:
> That's a feature dump of the receiver, you can get a feature dump of the MX Master 3 by running `sudo hidpp-list-features -d 1 /dev/hidraw3`

```
$ sudo hidpp-list-features -d 1 /dev/hidraw4
MX Master 3 (046d:4082) is a HID++ 4.5 device
Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x1d4b] Wireless device status
Feature 0x05: [0x0020] Reset
Feature 0x06: [0x0021] Crypto Identifier
Feature 0x07: [0x1000] Battery status
Feature 0x08: [0x1b04] Reprog controls v4
Feature 0x09: [0x1814] Change host
Feature 0x0a: [0x2250] ?
Feature 0x0b: [0x2201] Adjustable dpi
Feature 0x0c: [0x2110] Smart shift
Feature 0x0d: [0x2121] Hi-res wheel
Feature 0x0e: [0x2150] ?
Feature 0x0f: [0x2251] ?
Feature 0x10: [0x00c2] DFUcontrol 3
Feature 0x11: [0x1802] Device reset (hidden, internal)
Feature 0x12: [0x1803] ? (hidden, internal)
Feature 0x13: [0x1806] Configurable device properties (hidden, internal)
Feature 0x14: [0x1813] ? (hidden, internal)
Feature 0x15: [0x1805] OOBState (hidden, internal)
Feature 0x16: [0x1830] ? (hidden, internal)
Feature 0x17: [0x1890] ? (hidden, internal)
Feature 0x18: [0x1891] ? (hidden, internal)
Feature 0x19: [0x18a1] ? (hidden, internal)
Feature 0x1a: [0x1df3] ? (hidden, internal)
Feature 0x1b: [0x1e00] Enable hidden features (hidden)
Feature 0x1c: [0x1eb0] ? (hidden, internal)
Feature 0x1d: [0x1861] ? (hidden, internal)
Feature 0x1e: [0x9001] ? (hidden, internal)
Feature 0x1f: [0x9203] ? (hidden, internal)
Feature 0x20: [0x9205] ? (hidden, internal)
Feature 0x21: [0x9300] ? (hidden, internal)
```

PixlOne, 2020-08-01:
Oh good, it supports this feature. I'll see if I can get working on this, I won't be able to test it though since I only have an MX Master.

I'll start with supporting remapping vertical scrolling since I can actually test that out and then work on horizontal scrolling for the MX Master 3.

chepec, 2020-08-20:
I wonder, would this open up the possibility to invert the horizontal scrolling on MX Master 3? In my mind, the thumbwheel scrolls the "wrong way" by default.

solarchemist, 2020-08-27:
I tried to adapt your snippet above to invert the scroll direction on MX Master 3, but could not get it to take effect (I restart the systemctl job after each change to the config).

So I decided to see if I could just disable the thumbwheel. I'm sorry to report that did not work either.
I tried by including the above snippet both inside and outside the `buttons` section (I'm not sure where it's supposed to go). No effect, the thumb scroll worked as usual.
I also tried adding semicolons at the end of lines with assignments (I'm unsure of the syntax rules). Still no change.

Also, I did have Solaar running at the time, but I've also tested the above after first killing Solaar, and there was no change. So, no joy I'm afraid. Ideas?

```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: false; 
        invert: false; 
        target: false;
    };
    dpi: 1000;

    buttons: (
        {

            thumbwheel:
            {
                divert: true 
                invert: false
                left:
                {
                    mode: "NoPress"
                }
                right:
                {
                    mode: "NoPress"
                }
            }
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F10"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F7"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFT"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_RIGHT"];
                        }
                    },

                    {
                        direction: "None"
                        mode: "NoPress"
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
    );
}
);
```

PixlOne, 2020-08-27:
@solarchemist `thumbwheel` is not supposed to be inside `buttons`, it's supposed to be inside the `device` group, like how `hiresscroll` is.

solarchemist, 2020-08-27:
Thanks for the quick answer and the clarification :-)
But I've tried that too, as I was unsure where it was supposed to go. There was no effect.

PixlOne, 2020-08-27:
Can you post the log of `sudo logid -v DEBUG`?

solarchemist, 2020-08-27:
I get the following log (immediately after restarting the logid service):
```
$ sudo logid -v DEBUG
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[INFO] Detected receiver at /dev/hidraw2
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw2:1
[WARN] Line 73: action is a required field, skipping.
[DEBUG] /dev/hidraw2:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[WARN] Error adding device /dev/hidraw3: std::exception
```

with this `/etc/logid.cfg`:
```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: false;         invert: false;         target: false;
    };
    dpi: 1000;

	thumbwheel:
            {
                divert: true 
                invert: false
                left:
                {
                    mode: "NoPress"
                }
                right:
                {
                    mode: "NoPress"
                }
             }

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F10"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F7"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFT"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_RIGHT"];
                        }
                    },

                    {
                        direction: "None"
                        mode: "NoPress"
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
    );
}
);
```

PixlOne, 2020-08-27:
You are on the latest commit, right? If so, that's really weird, an info dump about the thumb wheel should appear when an MX Master 3 is connected.

solarchemist, 2020-08-27:
Alright, that was my bad (how do you make an embarrassed emoji?). I was on the latest *release*, not the latest *commit*. Now I'm on the latest commit `8348782`.

Same config as before, now the log:
```
$ sudo logid -v DEBUG
[sudo] password for taha: 
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[INFO] Detected receiver at /dev/hidraw2
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw2:1
[WARN] Line 73: action is a required field, skipping.
[DEBUG] /dev/hidraw2:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
[WARN] Error adding device /dev/hidraw3: std::exception
```

But the thumbwheel is still active. In fact, it's behaving just like I wanted (inverted scroll direction compared to the mouse's default). And I did quit Solaar, so that should not be interfering (before I closed Solaar, I reset all its toggles to their default setting). I have not restarted the computer, don't want to do that right now as I'm in the middle of other things. Shouldn't be necessary as long as I restart the `logid.service` though, right?

PixlOne, 2020-08-28:
Maybe I have invert and divert swapped? Try setting `invert: true` and `divert: false`.

solarchemist, 2020-08-28:
Thanks for the suggestions. I tried all four combinations of `invert: true|false` and `divert: true|false`, and I'm actually seeing no change at all between logid service restarts. The thumbwheel is active and scrolls in the default direction for all cases. (Which contradicts my earlier observation of reverse scrolling in the case of `divert: true; invert: false`. I can't explain that one).

This made me think that either my entire `/etc/logid.cfg` has no effect, or else the `thumbwheel` section is ignored. So I tested that assumption by trying to get rid of the warning about "action is required" in the service log (that warning is visible in my previous comment):
```
$ sudo journalctl -u logid.service -f
Aug 29 00:46:44 ubuntu systemd[1]: Stopping Logitech Configuration Daemon...
Aug 29 00:46:45 ubuntu systemd[1]: Stopped Logitech Configuration Daemon.
Aug 29 00:46:45 ubuntu systemd[1]: Started Logitech Configuration Daemon.
Aug 29 00:46:47 ubuntu logid[10339]: [WARN] Error adding device /dev/hidraw3: std::exception
```
Ok, so I successfully got rid of that warning by commenting out an (incomplete?) section in `/etc/logid.cfg`. That proves that the systemd job is indeed reading the config file, at least. 

I don't think the thumbwheel section has any effect, though.
At this point I think I've exhausted my usefulness to debug this issue. I hope someone else can try @PixlOne's suggested code and report back. Perhaps my system is not representative because I have installed Solaar (although it was never running during these tests). 

Running the latest commit of logiops on Ubuntu 18.04.5.
Sorry to make this comment so long, but just for completeness, here's my config:
```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: false;         invert: false;         target: false;
    };
    dpi: 1000;
                        
    thumbwheel:
    {
        divert: true
        invert: false
        left: {
            mode: "NoPress"
        }
        right: {
            mode: "NoPress"
        }
    }

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F10"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F7"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFT"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_RIGHT"];
                        }
                    }

                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
    );
}
);
```

gdiShun, 2020-08-29:
Doesn't work at all for me either.

peterbabic, 2020-09-03:
@solarchemist How come you got this level of DEBUG output? 

```
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```

I would love the **tap** action, but I always get only this:

```
[WARN] Error adding device /dev/hidraw1: std::exception
[INFO] Detected receiver at /dev/hidraw0
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
```

solarchemist, 2020-09-03:
@peterbabic I only got the Thumbwheel section in the debug log after I installed the current commit (note, not the current release) of Logiops. To be more precise, make sure you are running Logiops at commit [8348782](https://github.com/PixlOne/logiops/commit/8348782f27ecc998ddc4f86a5a892d33a7f21672) or later.

PixlOne, 2020-09-05:
Sorry about ignoring this for a while, I was a little busy.

I believe I've fixed this issue. @solarchemist I misaligned the arguments for setStatus, so divert was invert, and invert was out-of-bounds. I'm actually more shocked that it didn't just crash.

solarchemist, 2020-09-06:
I tried the latest commit, and if my test results are correct, there's good news and bad news. 

#### Here's the bad...
This thumbwheel snippet, copied from earlier example above, gives no horizontal scrolling at all:
```
thumbwheel:
{
   divert: true
   invert: false
   left: {
      mode: "Axis";
      axis: "REL_HWHEEL_HI_RES";
      axis_multiplier: -1;
   }
   right: {
      mode: "Axis";
      axis: "REL_HWHEEL_HI_RES";
      axis_multiplier: 1;
   }
}
```
and the debug log seems to indicate why:
```
[WARN] Line 21: Invalid axis REL_HWHEEL_HI_RES, skipping.
[WARN] Line 25: Invalid axis REL_HWHEEL_HI_RES, skipping.
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```

#### ... and my attempted fix (did not work)
so I tried dropping the `HI_RES` part of the axis name, (I checked the available axis names from [linux/input-event-codes.h](https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h) as suggested in the wiki):
```
thumbwheel:
{
    divert: true
    invert: false
    left: {
       mode: "Axis";
       axis: "REL_HWHEEL";            
       axis_multiplier: -1;
    }
    right: {
       mode: "Axis";
       axis: "REL_HWHEEL";
       axis_multiplier: 1;
    }
}
```
which gives no warnings in the debug log, but horizontal scroll is very choppy, and mostly not working at all.
```
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```

### The good news
As for the good news, disabling horizontal scroll now seems to work fine! This snippet:
```
thumbwheel:
{
   divert: true
   invert: false
   left: {
      mode: "NoPress";
   }
   right: {
      mode: "NoPress";
   }
}
```
gives no unexpected lines in the debug log, and successfully killed the thumbwheel. 

I really appreciate the level of support and feedback in this project. Kudos!

PixlOne, 2020-09-06:
@solarchemist Could you use your choppy horizontal scroll config and run `sudo evtest`, select `LogiOps Virtual Input`, scroll horizontally for a bit, and post the output? You can press Ctrl-C at any time while running that program to close it. Also, you should probably put it in a GitHub Gist since the output would be long.

gdiShun, 2020-09-06:
`Event: time 1599427802.682017, -------------- SYN_REPORT ------------
Event: time 1599427803.470002, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1599427803.470002, -------------- SYN_REPORT ------------
Event: time 1599427803.470025, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1599427803.470025, -------------- SYN_REPORT ------------
Event: time 1599427803.913990, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1599427803.913990, -------------- SYN_REPORT ------------
Event: time 1599427803.914008, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1599427803.914008, -------------- SYN_REPORT ------------
Event: time 1599427804.315989, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1599427804.315989, -------------- SYN_REPORT ------------
Event: time 1599427804.316007, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1599427804.316007, -------------- SYN_REPORT ------------
Event: time 1599427804.735993, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1599427804.735993, -------------- SYN_REPORT ------------
Event: time 1599427804.736010, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1599427804.736010, -------------- SYN_REPORT ------------
Event: time 1599427805.137989, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1599427805.137989, -------------- SYN_REPORT ------------
Event: time 1599427805.138006, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1599427805.138006, -------------- SYN_REPORT ------------
Event: time 1599427805.717989, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1599427805.717989, -------------- SYN_REPORT ------------
Event: time 1599427805.718007, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1599427805.718007, -------------- SYN_REPORT ------------
Event: time 1599427806.065967, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1599427806.065967, -------------- SYN_REPORT ------------
Event: time 1599427806.065983, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1599427806.065983, -------------- SYN_REPORT ------------
Event: time 1599427806.443983, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1599427806.443983, -------------- SYN_REPORT ------------
Event: time 1599427806.444001, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1599427806.444001, -------------- SYN_REPORT ------------
Event: time 1599427806.741989, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1599427806.741989, -------------- SYN_REPORT ------------
Event: time 1599427806.742015, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1599427806.742015, -------------- SYN_REPORT ------------
Event: time 1599427807.096002, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1599427807.096002, -------------- SYN_REPORT ------------
Event: time 1599427807.096019, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1599427807.096019, -------------- SYN_REPORT ------------
`

Here's some of it. What happens is it works like the "OnThreshold" gesture instead of like "OnInterval"(even though it's set as "OnInterval"). So once you move it X `pixels`, it presses and releases the button, and won't let you press it again until you move your thumb away from the thumbwheel or move it in the opposite direction. So effectively, a weird tap is more useful than scrolling it like someone normally would.

solarchemist, 2020-09-06:
```
$ sudo evtest 
No device specified, trying to scan all of /dev/input/event*
Available devices:
[... unrelated devices, redacted by me ...]
/dev/input/event6:	Logitech MX Master 3000
/dev/input/event7:	LogiOps Virtual Input
[... unrelated devices, redacted be me ...]
/dev/input/event17:	LogiOps Virtual Input
Select the device event number [0-17]: 6
```
`sudo evtest` shows me two Logiops Virtual input devices and one Logitech MX Master 3000. Only the latter showed any output at all when I operated the mouse, so that's the one I chose. I hope that's what we're looking for.

I [saved the output in this gist](https://gist.github.com/solarchemist/70311cd3b4a88ae3f9dc4c53d046319e), as suggested. Apart from clicking the button a few times initially, I tried to keep the mouse still and took care to only scroll the thumbwheel. 

Most scrolls were not registered, though, just like before. So at least we are consistent ;-)

PixlOne, 2020-09-06:
@solarchemist I see you have two instances of logid running, could you close them and only keep the one with the horizontal scroll config on?

LogiOps Virtual Input is responsible for transmitting diverted events so only the diverted horizontal scrolling and any remapped buttons will appear there.

The output for the MX Master 3000 isn't relevant.

gdiShun, 2020-09-06:
@solarchemist Could you try tapping the wheel like I mentioned? Instead of scrolling it like is natural, tap it in one direction and see if it works better. Probably the same problem for us both if so.

solarchemist, 2020-09-06:
@PixlOne I thought those two processes were weird too, but did not know enough to tell if it was relevant. Anyway, I checked with `ps aux` and indeed found two concurrent processes. I killed both of them and restarted the `logid.service`, there's now only one logid instance and evtest lists only one Logiops virtual device.
```
$ sudo evtest 
Available devices:
[...]
/dev/input/event6:	Logitech MX Master 3000
/dev/input/event7:	LogiOps Virtual Input
[...]
Select the device event number [0-16]: 7
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "LogiOps Virtual Input"
```

Thanks for that explainer about the role of the virtual input. Now I understand why all the "regular" mouse buttons did not register for this device! And I should have been more careful in my first report.

Here's the output from LogiOps Virtual Input when I'm scrolling the thumbwheel left/right at varying amplitudes:
```
Event: time 1599432838.366336, type 2 (EV_REL), code 6 (REL_HWHEEL), value 3
Event: time 1599432838.366336, -------------- SYN_REPORT ------------
Event: time 1599432839.300430, type 2 (EV_REL), code 6 (REL_HWHEEL), value 3
Event: time 1599432839.300430, -------------- SYN_REPORT ------------
Event: time 1599432840.010513, type 2 (EV_REL), code 6 (REL_HWHEEL), value 1
Event: time 1599432840.010513, -------------- SYN_REPORT ------------
Event: time 1599432840.606608, type 2 (EV_REL), code 6 (REL_HWHEEL), value 3
Event: time 1599432840.606608, -------------- SYN_REPORT ------------
Event: time 1599432841.380685, type 2 (EV_REL), code 6 (REL_HWHEEL), value -1
Event: time 1599432841.380685, -------------- SYN_REPORT ------------
Event: time 1599432842.025185, type 2 (EV_REL), code 6 (REL_HWHEEL), value -8
Event: time 1599432842.025185, -------------- SYN_REPORT ------------
Event: time 1599432842.604819, type 2 (EV_REL), code 6 (REL_HWHEEL), value -3
Event: time 1599432842.604819, -------------- SYN_REPORT ------------
Event: time 1599432843.232862, type 2 (EV_REL), code 6 (REL_HWHEEL), value -5
Event: time 1599432843.232862, -------------- SYN_REPORT ------------
Event: time 1599432843.796986, type 2 (EV_REL), code 6 (REL_HWHEEL), value -3
Event: time 1599432843.796986, -------------- SYN_REPORT ------------
Event: time 1599432844.457026, type 2 (EV_REL), code 6 (REL_HWHEEL), value 3
Event: time 1599432844.457026, -------------- SYN_REPORT ------------
Event: time 1599432845.005086, type 2 (EV_REL), code 6 (REL_HWHEEL), value 2
Event: time 1599432845.005086, -------------- SYN_REPORT ------------
Event: time 1599432845.585174, type 2 (EV_REL), code 6 (REL_HWHEEL), value 2
Event: time 1599432845.585174, -------------- SYN_REPORT ------------
Event: time 1599432845.859418, type 2 (EV_REL), code 6 (REL_HWHEEL), value 22
Event: time 1599432845.859418, -------------- SYN_REPORT ------------
Event: time 1599432846.477282, type 2 (EV_REL), code 6 (REL_HWHEEL), value 16
Event: time 1599432846.477282, -------------- SYN_REPORT ------------
Event: time 1599432846.895329, type 2 (EV_REL), code 6 (REL_HWHEEL), value 12
Event: time 1599432846.895329, -------------- SYN_REPORT ------------
Event: time 1599432847.443391, type 2 (EV_REL), code 6 (REL_HWHEEL), value -9
Event: time 1599432847.443391, -------------- SYN_REPORT ------------
Event: time 1599432847.861430, type 2 (EV_REL), code 6 (REL_HWHEEL), value -17
Event: time 1599432847.861430, -------------- SYN_REPORT ------------
Event: time 1599432848.279473, type 2 (EV_REL), code 6 (REL_HWHEEL), value -20
Event: time 1599432848.279473, -------------- SYN_REPORT ------------
Event: time 1599432848.635530, type 2 (EV_REL), code 6 (REL_HWHEEL), value -11
Event: time 1599432848.635530, -------------- SYN_REPORT ------------
```

@gdiShun I'm afraid I don't understand what you mean by tapping the thumbwheel. Tapping it does not feel natural to me, but that's just my opinion. In any case the thumbwheel has no springiness to it at all, so there's no tactile feedback when tapping, right?

gdiShun, 2020-09-06:
@solarchemist Yes, it shouldn't feel natural. But does it function more consistently when you do it though? To repeat what I said in my first post today. For me, the wheel will only be "pressed" once until I physically move my hand/thumb away from the wheel, or if I move it in the opposite direction(left then right or vice versa). So I can move the wheel, using 2 hands, multiple full rotations to the left, but it will only be pressed once. Quickly tapping it makes it work more consistently but obviously feels very unnatural.

rtowsley, 2020-10-02:
Hello,
Today I purchased MX Master 3, great mouse, and I just wanted to thank you for the development of this application to configure it in Linux. This thread helped me determine the correct syntax for the thumbwheel operation, now I have much playing around with the settings to determine what works best for me. First step was to install the logiops-git package from AUR instead of the standard logiops 0.2.2-1 to enable the thumbwheel operation.

I would suggest that updating the example configuration file to include some samples to understand the correct syntax, following the configuration wiki directly is difficult to get the syntax correct.