PixlOne/logiops #109: Error with G502 HERO, "tried to reset, but no reset mechanism was available"

shanewstone, 2020-07-22:
Hello and thank you for bringing Logitech mouse configuration to Linux.

I am trying to configure buttons on a G502 HERO using (for now) the following simple configuration file at `/etc/logid.cfg`,

```
devices: (
{

    name: "G502 HERO Gaming Mouse";

    dpi: 1200;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Keypress";
                keys: ["KEY_D"];
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type: "Keypress";
                keys: ["KEY_A"];
            };
        }
    );
}
);
```

`logid -v` returns

```
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[INFO] Device found: G502 HERO Gaming Mouse on /dev/hidraw1:255
[DEBUG] /dev/hidraw1:255 tried to reset, but no reset mechanism was available.
```

and no changes are made to the mouse.

Any ideas on why this would happen?

PixlOne, 2020-07-23:
That isn't quite the issue, the issue is more that the G502 HERO uses a differrent method to control buttons that logid does not support. (Otherwise it would have printed a list of CIDs)

You should be able to change the DPI though.

shanewstone, 2020-07-24:
Okay, thanks for the reply!

closed 2020-07-24 by shanewstone