PixlOne/logiops #113: Docs: How to find the right CIDS for your device / Adding devices to the documentation.

NiklasBeierl, 2020-08-08:
Hello, thanks for providing this cool package! 

I am just having a bit of trouble using it:
As far as I understood the wiki, the `cid` is used to configure the button that I want to remap. I then headed to the `CIDs` section in hopes of finding the right cid for the "thumb button" on my MX Master 2S. Sadly I couldn't find it there and I got quite frustrated because there where no further hints for figuring it out myself. I also couldn't find it after 15 minutes of googling, I tried `xev` but that didn't give me any working `cid` either.
In my case I got lucky and found a issue here where someone had the same problem. However, I think we should "teach a man how to fish". I am sure there is some way `xev`-like way of figuring out the cids and we can add that to the docs / wiki. 

It might also encourage people to map out their devices and contribute some descriptions to the wiki. I would gladly do so for the MX Master 2S. :)

PixlOne, 2020-08-08:
It's called the Gesture Button and it's `0xc3`. I'll try to add a debug action and use that to help people discover the right CIDs.

milanleonard, 2020-08-16:
As a note here, you can directly modify the MX Master 3 logid.cfg file provided. I don't know if this would be useful to you, but here's my MX Master 2S config that I like to use

https://pastebin.com/0bHpuLL7

The functions are 
1. Thumb down no movement - launch control
2. Thumb down up/down - zoom in/out
3. Thumb left/right - tile window left/right

WorkProg, 2024-04-22:
I'd really like a xev like tool.

As I'll be, through trial and error I guess, map out the various `G*` keys on my G502, would you be opposed to adding device specific subpages to the `CID` wiki section?

ModProg, 2024-04-22:
I'd really like a xev like tool.

As I'll be, through trial and error I guess, map out the various `G*` keys on my G502, would you be opposed to adding device specific subpages to the `CID` wiki section?

WorkProg, 2024-04-22:
Should have maybe done at least 5 more minutes of research, seems the 502 is not actually supported https://github.com/PixlOne/logiops/issues/109

ModProg, 2024-04-22:
Should have maybe done at least 5 more minutes of research, seems the 502 is not actually supported https://github.com/PixlOne/logiops/issues/109