PixlOne/logiops #116: Adjust scroll speed

PolygonHJ, 2020-08-16:
Is there a way that the scroll wheel speed can be adjusted (e.g. number of lines per notch)? 

Sorry if this is a stupid question, I read through the configuration page and issues but couldn't find anything mentioning it. I also absolutely love the project too! Thanks for the great work!

PolygonHJ, 2020-08-17:
Reading through the issues again, I realised today that setting `hires:false` sets the behaviour to what I would expect. My error previously was that I had two instances of `logid` running with different options.

That said, it would still be nice to see an option to customise the scroll speed!

rurickdev, 2020-08-21:
I also had scroll speed issues, When the service is running the scroll is too slow.

rurickdev, 2020-08-25:
Thanks, solved with that.
One think I noted was if a defined `target` and `hires` as true the scroll stoped working.

This is the `hiresscroll` section that make it work like usual, the value I used was 8

```bash
hiresscroll:
	    {
		    hires: true;
		    invert: false;
		    target: false;
		    up: {
		        mode: "Axis";
		        axis: "REL_WHEEL_HI_RES";
		        axis_multiplier: 8;
		    },
		    down: {
		        mode: "Axis";
		        axis: "REL_WHEEL_HI_RES";
		        axis_multiplier: -8;
		    },
	    };
```

denydias, 2020-08-31:
@rurickdev thanks, that helped me.

I confirm that by setting `target: true` there's no scroll at all. Also I see no difference setting `axis_multiplier` to whatever value. All that I get is a minimum scroll of about seven lines in any text editor. I would like to set it to about three lines, but I could not manage to achieve that with any of the documented configurations nor by using lower values in `axis_multiplier` as per @PixlOne recommends above.

My environment:

```console
$ logid -V
v0.2.2-8-gc3419b9
$ modinfo hid_logitech_hidpp|grep vermagic
vermagic:       5.4.61 SMP mod_unload
$ grep -A 17 hiresscroll /etc/logid.cfg
    hiresscroll:
    {
      hires: true;
      invert: false;
      target: false;
      up:
      {
        mode: "Axis";
        axis: "REL_WHEEL_HI_RES";
        axis_multiplier: 3;
      },
      down:
      {
        mode: "Axis";
        axis: "REL_WHEEL_HI_RES";
        axis_multiplier: -3;
      },
    };
```

PS: @PixlOne, thank you so much for this fine piece of software! Your driver is just great for anyone using MX series devices with Linux.

vip9, 2020-11-01:
I've also wanted to make scroll faster, but axis_multiplier seems not working. 
BTW, for brave/chrome/etc I am using https://github.com/gblazex/smoothscroll extension with high step and no animation (I personally does not like smoothing my scroll, just needed make it faster)

healarconr, 2020-12-19:
This is the `hiresscroll` section that works for me:

```
  hiresscroll: {
    hires: false;
    invert: false;
    target: true;
    up: {
        mode: "Axis";
        axis: "REL_WHEEL";
        axis_multiplier: 2;
    },
    down: {
        mode: "Axis";
        axis: "REL_WHEEL";
        axis_multiplier: -2;
    }
  };
```

I did not enable `hires`, enabled `target` and used `REL_WHEEL` as `axis` instead of `REL_WHEEL_HI_RES`. The scrolling speed is 2 times faster, but I feel it sometimes misses the scroll when the wheel is rotated. I feel that this problem does not feel as much if `smartshift` is disabled. Still it is better than having a slow scrolling speed.

denydias, 2020-12-20:
That worked great, @healarconr! Thank you!

mrpandey, 2021-04-01:
Just discovered this project. Awesome work. Just need some help setting up my config.

How can I set the scroll speed of the thumbwheel? Does `axis_multiplier` work for it too?

arudp, 2021-09-10:
> This is the `hiresscroll` section that works for me:
>
> ```
>   hiresscroll: {
>     hires: false;
>     invert: false;
>     target: true;
>     up: {
>         mode: "Axis";
>         axis: "REL_WHEEL"; [...]

For the missing scroll thing, I set `target` to `false` and it worked wonders!

Valeronlol, 2023-05-27:
getting an error: Error at line [axis_multiplier config line number here]: SettingTypeException
Seems axis_multiplier deprecated or doesn't work.

sanderpick, 2023-06-08:
> getting an error: Error at line [axis_multiplier config line number here]: SettingTypeException Seems axis_multiplier deprecated or doesn't work.

`axis_multiplier` must be float, e.g, `1.0` / `-1.0`, otherwise you'll see those `SettingTypeException` logs. afaict, the multiplier only works with `hires: false`, at least on my distro (pop os).

tsmalls33, 2023-11-20:
Hey, I am just now configuring this with my Mx Master 3s. My problem is the fast scrolling in ratchet mode, I would like it to be slower. When I set Hiresscroll to this, my scrolling improves but then all of my gestures stop working... anybody have an idea as to why that might be?
`  hiresscroll: {
    hires: false;
    invert: false;
    target: true;
    up: {
        mode: "Axis";
        axis: "REL_WHEEL";
        axis_multiplier: 2;
    },
    down: {
        mode: "Axis";
        axis: "REL_WHEEL";
        axis_multiplier: -2;
    }
  };`

pank-su, 2024-01-21:
> Hey, I am just now configuring this with my Mx Master 3s. My problem is the fast scrolling in ratchet mode, I would like it to be slower. When I set Hiresscroll to this, my scrolling improves but [...]

I have Mx Master 3S too, this helps me

https://github.com/PixlOne/logiops/issues/116#issuecomment-1583041962

my conf:

```
 hiresscroll:
    {
        hires: false;
        invert: false;
        target: true;
	up: {
            mode: "Axis";
            axis: "REL_WHEEL";
            axis_multiplier: 2.0;
    	},
    	down: {
              mode: "Axis";
              axis: "REL_WHEEL";
              axis_multiplier: -2.0;
    	}
    }; 
```

elijah-potter, 2024-11-30:
As a plus-one: I have a MX Master 2S and this is the config I'm comfortable with:

```
hiresscroll:
{
    hires: false;
    invert: false;
    target: true;
    up: {
        mode: "Axis";
        axis: "REL_WHEEL";
        axis_multiplier: 20.0;
    },
    down: {
        mode: "Axis";
        axis: "REL_WHEEL";
        axis_multiplier: -20.0;
    },
};
```