PixlOne/logiops #118: Allow define prefix keys in gesture actions

leodream, 2020-08-21:
Many task/windows-switching shortcuts require to keep holding the first key and press the second key multiple times to iterate over the full list of candidates. E.g. the "`Ctrl+Tab`" in Firefox, IDEA, PyCharm, "`Alt+Tab`" in Windows and the "task switcher" in KDE.

It would be helpful if it is allowed to define "prefix" keys in the root level of the gesture action.
Something like below.
```
    buttons: (
      {
        cid: 0xc3;
        action =
        {
            type: "Gestures";
            keys: ["KEY_LEFTALT"];
            gestures: (
                {
                    direction: "Left";
                    mode: "OnFewPixels";
                    pixels: 20;
                    action =
                    {
                        type: "Keypress";
                        keys: ["KEY_TAB"];
                    };
                },
```

The idea is to 
- Hold the gesture button =>  hold the prefix key 'LEFT_ALT"
- Move the mouse to left for n pixels => press and release "TAB" n/20 times
- Release the gesture button => release the 'LEFT_ALT" to stop the task switching 

Currently, if I define "KEY_LEFTALT" + "KEY_TAB" in the direction action, I can only switch between 2 particular windows, instead of all windows.

abraha2d, 2020-11-24:
^ try my changes on that PR, let me know if that works.

leodream, 2020-11-30:
Works like a charm!
Found that you also implemented #150 that allows using the wheel in gestures, which is even nicer.

closed 2021-07-27 by leodream