PixlOne/logiops #128: Gesture reaction time

gitthangbaby, 2020-09-07:
The gestures feel delayed vs Windows. All 8 gestures defined for Mx Master 3 have some initial delay.
For example, you do VolumeUp gesture with 0xc3,0xc4 Button+move up:
```
                    {
                        direction: "Up";
                        mode: "OnInterval";
                        interval: 30;
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_VOLUMEUP"];
                        };
                    },

```
...it will perform not after 30 pixels, but let's say after 100 pixels. Next steps will be fine. It's confusing as you need to move mouse more, and then it flies quick, so you need to slow down, and readjust by doing opposite gesture.
The worse reaction is on direction change. When doing VolumeDown after VolumeUp, therefore keep holding Button but change mouse direction, the gap will be massive, like 300 pixels. In the end,  the mouse will travel beyond the mouse pad.

abraha2d, 2020-11-24:
This is by design (at least what I understand by looking in the code). For your first issue of delayed gestures, you can somewhat get around that by setting `threshold: 1` to remove the extra delay at the beginning.

For the second issue, the code is set up to only start the opposite direction gesture once you reach the origin point. For example, if you press the gesture button, and move up 300 pixels, then it will only start the down gesture once you move back down 300 pixels. Fixing this would require modifying the code.

gitthangbaby, 2020-11-24:
Thanks for identifying the mechanisms. I quickly added thresholds and it feels more natural. Funny logic for the second mechanism, which I didn't see before. I'd expect mouse to return to the same place _after_ reverse gesture is done. Tested further there's no such gap between left->up, right->down movements. I think i found the code (blocks with "// Ignore to Origin") and i wonder what was the original plan.
Testing distros for 6months, this particular issue is in my top3 list of things "missing from Windows" every time I made a note. It impacts perhaps the most popular gesture combo Volume+ <-> Volume- and Zoom+ <-> Zoom-. If you agree, would it be possible to make assessment if it's safe to make amendment to this code? Thanks a lot.

abraha2d, 2020-11-24:
^ that PR addresses these issues (so you won't need the `threshold: 1` either). Try it out, lmk what you think.

bdavjones, 2021-08-23:
Sorry to necro this thread but was there ever / is there planned to be any way to ignore the `Ignore to origin ` in gestures.cpp ? 

I've tried blatting the ignore to origin section on the GestureAction.cpp and rebuilding to try and get smooth volume up/down transition like so:

`            //if(new_y) { // Ignore to origin
            auto up = _config.gestures().find(Up);
            if(up != _config.gestures().end())
                up->second->move(-new_y);
            //}`

however it still seems to behave much the same, waiting until the origin point to hit the inversion, which usually ends up in me overshooting!