PixlOne/logiops #129: logid stops working after mouse is idle for a while

hendriktews, 2020-09-09:
Hi,
I use logid for an MX Master 3 connected over bluetooth mostly to make the thumb button a real mouse button. This works nicely (thanks!!), however, logid stops working when the mouse is idle for a while (somewhere between 5 minutes and half an hour). The output of logid gives no indication and the logid process keeps running. I see
```
cert ~ 8> sudo logid -v -c ~/.logid.cfg 
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw1 ignored
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw2:255
[DEBUG] /dev/hidraw2:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```

where the thumb wheel line directly appears after starting and then there is no more output. 

Restarting logid fixes the problem (until the next longer idle time) and meanwhile I have a window manager shortcut for that. Nevertheless it is a nuisance when the favorite mouse button stoped working after returning from the whiteboard.

I would appreciate any hints about how to fix or debug this problem. Thanks!

mankyd, 2020-09-17:
I see somethinmg similar where my mouse wheel acceleration turns off after my desktop idles. It sounds similar to this issue someone reported on reddit a year ago (not using logiops): https://www.reddit.com/r/kde/comments/bk6u5z/scroll_button_speed_slow_after_wake_up_from_sleep/

I would love a fix for this because it is absolutely infuriating. I am not sure where the fault lies, however.

hendriktews, 2020-09-30:
I don't see the problem anymore and I have not changed anything in logiops or its configuration. But I regularly updated my system. This seems to indicate that there was either a bug in the system or a communication problem with logiops that is not triggered any more.

PixlOne, 2020-10-02:
Does this only happen when you connect your mouse over Bluetooth?

mankyd, 2020-10-02:
That is the case for me, yes, though I haven't tested thoroughly with the unify receiver.

hendriktews, 2020-10-05:
pixl <notifications@github.com> writes:

> Does this only happen when you connect your mouse over Bluetooth?

Yes, I connect my mouse over bluetooth. I have not tested with a
different connection. So it happens with bluetooth, but I cannot
tell whether it only happens with bluetooth.

Hendrik

PixlOne, 2020-10-23:
Is this on the latest commit?

hendriktews, 2020-10-25:
As I wrote on Sept 30, the issue went away, apparently without changing logiops. @mankyd: Can you still reproduce? Otherwise I suggest to close this.

ExposedCat, 2021-06-30:
Add `/lib/systemd/system-sleep/logid.sh` with content:
```bash
#!/bin/sh
systemctl restart logid
```
It will restart logid.service on WakeUp, so driver will work after idle and resuming from suspend

PiotrGorczyca, 2024-07-19:
MxMaster 3 for Mac. Manjaro linus. Recently got the same issue where i go make a coffee and come back to a mouse with default settings. Additionally even though i have logiops installed and the logid config is present i have to manually run systemd start logid on every boot, and when the mouse looses its custom settings the systemd restart logid does not do anything. I do not see any processes related to logiops or logid running either.

Yhugi7, 2026-09-16:
Hmm, weird. I still have this issue. M720 Triathlon on LMDE 7, using a unifying receiver. Am I doing something wrong? If I restart the logid service, everything works fine like others have said.

fnxweb, 2026-11-12:
I'm also seeing this with a new MX Master 4 (Fedora 44, logiops 0.3.5) using the receiver widget.   I need to see if I can get any udev debug or something to tell if I can see the mouse waking up, there's nothing obvious in the logs by default.

_edit_  OK, tell a lie, if I look further back I do see:
```
Nov 12 11:00:07 host logid[12323]: [INFO] /dev/hidraw2:2 fell asleep.
Nov 12 11:21:30 host logid[12323]: [INFO] /dev/hidraw2:2 woke up.
```
I'll watch next time to see if that's in the logs when the mouse config is lost.

fnxweb, 2026-11-12:
Yes, when I see that come out in the log, I need to restart the service to get the mouse config back.