PixlOne/logiops #132: Virtual device gets detected as XBox 360 Controller

rockerbacon, 2020-09-18:
When playing Rocket League, my actual controller was stuck as player 2. Steam controller settings listed a second XBox 360 controller I could not identify. Stopping the logid service fixed both issues, Rocket League correctly identified my controller as the only one and Steam no longer listed a "ghost" controller.

PixlOne, 2020-10-02:
This might be because the evdev device for logiops reports that it supports all reports. I'll look into this, thanks.

PixlOne, 2020-10-02:
I can't really selectively enable evdev reports after initialising the device, so I want to try something else. 
Also, could you explain how to reproduce this bug, is this just with Rocket League? When I tried this out on Steam Big Picture, I couldn't reproduce this bug.

Also, try out the following patch and see if that helps:

```patch
From e63820e9c55a8d98c3a83bcfc41059ee0968b396 Mon Sep 17 00:00:00 2001
From: pixl <pixlone@protonmail.com>
Date: Fri, 2 Oct 2020 17:00:16 -0400
Subject: [PATCH] Use Logitech vendor ID for evdev device

---
 src/logid/InputDevice.cpp | 4 ++++
 1 file changed, 4 insertions(+)

diff --git a/src/logid/InputDevice.cpp b/src/logid/InputDevice.cpp
index 30b448b..5f78245 100644
--- a/src/logid/InputDevice.cpp
+++ b/src/logid/InputDevice.cpp
@@ -26,6 +26,9 @@ extern "C"
 #include <libevdev/libevdev-uinput.h>
 }
 
+#define LOGITECH_VENDOR_ID 0x046d
+
+
 using namespace logid;
 
 InputDevice::InvalidEventCode::InvalidEventCode(const std::string& name) :
@@ -42,6 +45,7 @@ InputDevice::InputDevice(const char* name)
 {
     device = libevdev_new();
     libevdev_set_name(device, name);
+    libevdev_set_id_vendor(device, LOGITECH_VENDOR_ID);
 
     ///TODO: Is it really a good idea to enable all events?
     libevdev_enable_event_type(device, EV_KEY);
-- 
2.28.0
```

rockerbacon, 2020-10-16:
The bug should be easy to identify by going to Steam's general controller settings. You should be able to identify an inexistent controller in the list of detected devices and stopping and starting logid should make the device go away and pop back up.

Rocket League just has this weird behaviour of strictly following the XInput device order which forces the actual controller into player 2. I'm not sure having that ghost device can cause problems anywhere else.

For reference, this was seen on a MX Master 2s connected with the 2.4Ghz dongle.

PixlOne, 2020-10-23:
Did the suggested patch help?