PixlOne/logiops #133: logid.cfg is not created

manmohitgrewal1, 2020-09-20:
I have cloned your repo and followed the instructions. But there is not file in `/etc` named logid.cfg
```
~ systemctl status logid
● logid.service - Logitech Configuration Daemon
   Loaded: loaded (/lib/systemd/system/logid.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-09-20 17:35:40 +0530; 35min ago
 Main PID: 28458 (logid)
    Tasks: 7 (limit: 4915)
   CGroup: /system.slice/logid.service
           └─28458 /usr/local/bin/logid

Sep 20 17:35:40 manmohit-Inspiron-7373 systemd[1]: Started Logitech Configuration Daemon.
Sep 20 17:35:40 manmohit-Inspiron-7373 logid[28458]: [ERROR] I/O Error while reading /etc/logid.cfg: FileIOException
Sep 20 17:35:42 manmohit-Inspiron-7373 logid[28458]: [WARN] Error adding device /dev/hidraw1: std::exception
Sep 20 17:51:39 manmohit-Inspiron-7373 systemd[1]: logid.service: Current command vanished from the unit file, execution of the command list

```

```
ls /etc/ | grep logi

login.defs

```

cjlaborde, 2020-09-26:
I am having the exact same problem
```
▶ sudo make install   
[100%] Built target logid
Install the project...
-- Install configuration: ""
-- Up-to-date: /usr/local/bin/logid
-- Up-to-date: /lib/systemd/system/logid.service

```

cjlaborde, 2020-09-26:
Added the file manually and getting [INFO] Device Wireless Mouse MX Master 3 not configured, using default config.
```
▶ sudo logid -v -c /etc/logid.cfg
[INFO] Device Wireless Mouse MX Master 3 not configured, using default config.
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw0:255
[DEBUG] /dev/hidraw0:255 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```

cjlaborde, 2020-09-26:
using configuration from here
https://wiki.archlinux.org/index.php/Logitech_MX_Master
```
devices: (
{
    name: "Logitech Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 30;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F10"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_F7"];
                        };
                    },
#                    {
#                        direction: "Left";
#                        mode: "OnRelease";
#                        action =
#                        {
#                            type: "CycleDPI";
#                            dpis: [50, 500, 1000, 1500, 2000, 3000, 4000];
#                        };
#                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_LEFT"];
                        };
                    },

#                    {
#                        direction: "Right";
#                        mode: "OnRelease";
#                        action =
#                        {
#                            type = "ToggleHiresScroll";
#                        }
#                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA", "KEY_RIGHT"];
                        }
                    },

                    {
                        direction: "None"
                        mode: "NoPress"
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_LEFTCTRL", "KEY_PAGEDOWN"];
            };
        }
    );
},
{
# Another device to configure
name: "Other Logitech USB Receiver";

}
);
```

cjlaborde, 2020-09-26:
Using Ubuntu 18.04

cjlaborde, 2020-09-26:
Ok I changed
```
    name: "Logitech Wireless Mouse MX Master 3";
```
to
```
     name: "Wireless Mouse MX Master 3";
```
and now works

manmohitgrewal1, 2020-09-27:
Fix my problem as well! Thank you for the help :+1:

closed 2020-09-27 by manmohitgrewal1