PixlOne/logiops #134: MX Master 3 - firmware update

adreasnow, 2020-09-22:
Hey PixlOne,
As the title says, my MX Master 3 is used across multiple computers, one of which has Logitech Options and receives firmware updates. After a recent update, button allocation is not working, even though scroll and DPI control is. 

No errors are produced upon running logiops with `-v` and I  haven't changed the config file since the update.

The issue seems to not be limited to logiops, as ratbagd also has the same issue.

Details according to Solaar:
```
Index        : 1
Wireless PID : 4082
Protocol     : HID++ 4.5
Polling rate : 8 ms (125Hz)
Serial       : <redacted>
  Bootloader : BOT 95.01.B0015
  Firmware   : MPM 19.01.B0015
```

Thanks,
Adrea

PixlOne, 2020-09-22:
Run `sudo hidpp-list-features` from [PixlOne/hidpp](https://www.github.com/PixlOne/hidpp) and post the output.

adreasnow, 2020-09-22:
Sorry that took a while, I didn't realise that you had to `sudo cmake` to get it to find libudev.

I hope this is what you need, 

```
# hidpp-list-features /dev/hidraw0  
Logitech USB Receiver (046d:c52b) is a HID++ 1.0 device
Register 0x00 read   3: 00 09 00
Register 0x01 read   3: 00 00 00
Register 0x02 read   3: 00 01 00
Register 0x03 read   3: 00 00 00
Register 0xb3 read  16: c6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
Register 0xb5 read  16: Invalid value (0x03)
Register 0xd0 read   3: 00 00 00
Register 0xd4 read   3: 00 00 7c
Register 0xe6 read   3: 00 00 00
Register 0xf0 read   3: 00 00 00
Register 0xf1 read   3: Invalid value (0x03)
```
```
# hidpp-list-features /dev/hidraw0 -d 1
MX Master 3 (046d:4082) is a HID++ 4.5 device
Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x1d4b] Wireless device status
Feature 0x05: [0x0020] Reset
Feature 0x06: [0x0021] Crypto Identifier
Feature 0x07: [0x0007] Device Friendly Name
Feature 0x08: [0x1000] Battery status
Feature 0x09: [0x1b04] Reprog controls v4
Feature 0x0a: [0x1814] Change host
Feature 0x0b: [0x2250] ?
Feature 0x0c: [0x2201] Adjustable dpi
Feature 0x0d: [0x2110] Smart shift
Feature 0x0e: [0x2121] Hi-res wheel
Feature 0x0f: [0x2150] ?
Feature 0x10: [0x2251] ?
Feature 0x11: [0x00c2] DFUcontrol 3
Feature 0x12: [0x1802] Device reset (hidden, internal)
Feature 0x13: [0x1803] ? (hidden, internal)
Feature 0x14: [0x1806] Configurable device properties (hidden, internal)
Feature 0x15: [0x1813] ? (hidden, internal)
Feature 0x16: [0x1805] OOBState (hidden, internal)
Feature 0x17: [0x1830] ? (hidden, internal)
Feature 0x18: [0x1890] ? (hidden, internal)
Feature 0x19: [0x1891] ? (hidden, internal)
Feature 0x1a: [0x18a1] ? (hidden, internal)
Feature 0x1b: [0x1df3] ? (hidden, internal)
Feature 0x1c: [0x1e00] Enable hidden features (hidden)
Feature 0x1d: [0x1eb0] ? (hidden, internal)
Feature 0x1e: [0x1861] ? (hidden, internal)
Feature 0x1f: [0x9001] ? (hidden, internal)
Feature 0x20: [0x9203] ? (hidden, internal)
Feature 0x21: [0x9205] ? (hidden, internal)
Feature 0x22: [0x9300] ? (hidden, internal)
```

PixlOne, 2020-09-22:
Hmm, everything seems to be fine there. Could you try running `sudo hidpp20-reprog-controls -d <device index> <hidraw path> info` and posting the output? It's from the same repo.

adreasnow, 2020-09-22:
```
# hidpp20-reprog-controls -d 1 /dev/hidraw0 info
Control	Task	Info	Fn	Capabilities	Group	Remap to groups
0x0050	0x0038	mouse	-	-	1	1
0x0051	0x0039	mouse	-	-	1	1
0x0052	0x003a	mouse	-	reprog, divert	3	1+2+3
0x0053	0x003c	mouse	-	reprog, divert	2	1+2
0x0056	0x003e	mouse	-	reprog, divert	2	1+2
0x00c3	0x00a9	mouse	-	reprog, divert	3	1+2+3
0x00c4	0x009d	mouse	-	reprog, divert	3	1+2+3
0x00d7	0x00b4	-	-	divert, virtual	4	-
```

PixlOne, 2020-09-22:
Hmm, I see no reason why it shouldn't be working. Can you post your config file?

adreasnow, 2020-09-22:
At the risk of a long post:

```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 255;
        default_threshold: 255;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    thumbwheel:
    {
        invert: true;
    }
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_UP"];
                        };
                    },
                    #{
                    #    direction: "Left";
                    #    mode: "OnRelease";
                    #    action =
                    #    {
                    #        type: "Keypress";
                    #        keys: ["KEY_LEFTSUPER", "KEY_LEFT"];
                    #    };
                    #},
                    #{
                    #    direction: "Right";
                    #    mode: "OnRelease";
                    #    action =
                    #    {
                    #        type: "Keypress";
                    #        keys: ["KEY_LEFTSUPER", "KEY_RIGHT"];
                    #    };
                    #},
                    {
                        direction: "None"
                        mode: "OnRelease"
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_SCALE"];
                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        },
        {
            # Next tab instead of fwd in history, Comment to default behavior
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_FORWARD"];
            };
        },
        {
            # Previous tab instead of back in history, Comment to default behavior
            cid: 0x53;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_BACK"];
            };
        }
    );
}
);
```

PixlOne, 2020-09-22:
Your config file seems fine too, I really don't know why this is happening.

Is anyone else with an MX Master 3 experiencing this issue?

adreasnow, 2020-09-22:
I keep my system pretty up to date as well, so there could be something that ubuntu 20.04.1 updated that's causing this?
I might try compiling logiops on a live usb and see if it fixes anyhting.

PixlOne, 2020-09-22:
I guess you could try `sudo evtest` and monitor events from `LogiOps Virtual Input`. If the button presses are appearing on the event log, then it's definitely some sort of incompatibility between Ubuntu 20.04.1 and LogiOps.

adreasnow, 2020-09-22:
I just tested and the same issue happens on a clean copy of ubuntu 20.04.1 (i might try 20.04.0 next). `sudo evtest` does show `LogiOps Virtual Input` and it's not reporting any button presses.

adreasnow, 2020-09-22:
I'm going to close this, as the same issue is happening on other, non-linux devices as well, which makes me think that it's a firmware issue. Thank you for your time helping me to troubleshoot the issue though.

closed 2020-09-22 by adreasnow