PixlOne/logiops #135: Erratic scroll speed

thoastbrot, 2020-09-25:
Related to comment https://github.com/PixlOne/logiops/issues/116#issuecomment-683582102 (using the same config values for hires, but without the up/down settings)

I have the same issue, scrolling with a single haptic point on the wheel produces many scrolled lines, including a few lines when just tapping the wheel a bit. I have two connected systems, an arch linux installation and an ubuntu 20.04, the latter with the issue. The arch installation works just fine. Device is an MX Master 3. The config file is a copy of the working installation.

Tried 0.2.2 and master to no avail. I suspect a change in libevdev (ubuntu: 1.9.0, arch 1.9.1) or similar. What's your base OS?

thoastbrot, 2020-09-25:
Eh, immediate solution to my ubuntu is too simple. Set hires to false, and it works as expected... keeping it open, if you would like to inspect the differences between the systems further. Otherwise, just close it.

PixlOne, 2020-10-02:
This could have something to do with your kernel version. In Linux 5.6, the hid-logitech-hidpp driver added support for the MX Master 3, Ubuntu 20.04 uses Linux 5.4 by default. If hid-generic (regular hid driver) is used, then I've noticed that scrolling on hi-res scrolling logitech mice feels weird.

You could try updating your kernel to 5.6 and trying that out, but going forward, I'd like to find some solution that eliminates these kinds of kernel quirks.

thoastbrot, 2020-10-03:
OK, I'll try that. In case you can't get rid of the quirks, maybe add a warning for the combination of driver/kernel version?

thoastbrot, 2020-10-05:
FYI: kernel 5.6 works as expected.