PixlOne/logiops #139: Cannot detect libevdev on CentOS 8.

vu-dinh-hung, 2020-10-16:
CMake could not find the libevdev package, even though I have libevdev, libevdevPlus, and libevdevPlus-devel installed. Here is the error:
```
Checking for module 'libevdev'
--   Package 'libevdev', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:418 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:585 (_pkg_check_modules_internal)
  src/logid/CMakeLists.txt:75 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
```
Am I missing something here, and there's a different package I should have installed for libevdev, or is logiops simply not working on CentOS 8?

weilintuye95, 2020-10-21:
Happens to me on Ubuntu 20.04 LTS as well

`--   No package 'libevdev' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  src/logid/CMakeLists.txt:75 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
`

vu-dinh-hung, 2020-10-21:
@weilintuye95 I actually have logiops on my Ubuntu 20.04 LTS installation. I installed g++ in addition to the packages you mentioned. Maybe ```sudo apt install g++``` would solve your problem.

vu-dinh-hung, 2020-10-21:
As for the original issue, I solved it on CentOS 8 by enabling PowerTools repo, then installing libevdev-devel. 

This was also a duplicate issue, which I didn't realize I had created (spotty internet). Closed now.

closed 2020-10-21 by vu-dinh-hung

weilintuye95, 2020-10-21:
Thanks for the answer @vu-dinh-hung! Didn't work still as I have g++ installed already as well -- glad to see that you found a fix though!