PixlOne/logiops #140: Cannot find module 'libevdev' on CentOS 8

vu-dinh-hung, 2020-10-16:
CMake could not find the libevdev package, even though I have libevdev, libevdevPlus, and libevdevPlus-devel installed. Here is the error:
```
Checking for module 'libevdev'
--   Package 'libevdev', required by 'virtual:world', not found
CMake Error at /usr/share/cmake/Modules/FindPkgConfig.cmake:418 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPkgConfig.cmake:585 (_pkg_check_modules_internal)
  src/logid/CMakeLists.txt:75 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
```
Am I missing something here, and there's a different package I should have installed for libevdev, or is logiops simply not working on CentOS 8?

einsell, 2020-10-18:
I'm on Fedora, but have you tried installing libevdev-devel ?

vu-dinh-hung, 2020-10-21:
@einsell that did the job. I couldn't find libdevdev-devel at first, turns out I needed to enable the PowerTools repo.

closed 2020-10-21 by vu-dinh-hung