PixlOne/logiops #142: Libevdev installed but cmake cannot detect during installation - Ubuntu 20.04.1 LTS

weilintuye95, 2020-10-21:
I'm trying to install logiops and I'm pretty sure all requirement packages are installed, eg. if i run the following:
```
sudo apt install cmake libevdev-dev libudev-dev libconfig++-dev git g++     

```
I get:
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.16.3-1ubuntu1).
g++ is already the newest version (4:9.3.0-1ubuntu2).
git is already the newest version (1:2.25.1-1ubuntu3).
libevdev-dev is already the newest version (1.9.0+dfsg-1).
libconfig++-dev is already the newest version (1.5-0.4build1).
libudev-dev is already the newest version (245.4-4ubuntu3.2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

```
The error is:
```
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /home/linuxbrew/.linuxbrew/bin/cc
-- Check for working C compiler: /home/linuxbrew/.linuxbrew/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/linuxbrew/.linuxbrew/bin/c++
-- Check for working CXX compiler: /home/linuxbrew/.linuxbrew/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
LogiOps Version Number: v0.2.2-31-g4c58224
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /home/linuxbrew/.linuxbrew/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libevdev'
--   No package 'libevdev' found
CMake Error at /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:463 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.16/Modules/FindPkgConfig.cmake:643 (_pkg_check_modules_internal)
  src/logid/CMakeLists.txt:75 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home/weilin/Downloads/logiops/build/CMakeFiles/CMakeOutput.log".
See also "/home/weilin/Downloads/logiops/build/CMakeFiles/CMakeError.log".
```
Any help would be appreciated! Thanks in advance

alexpetrean80, 2020-10-22:
Happens on Fedora Workstation 32 too

PixlOne, 2020-10-23:
For Fedora, try the `libevdev-devel` package.

As for Ubuntu/Debian, I'll look into that.

danmackinlay, 2020-10-23:
@weilintuye95 cmake is picking up your linuxbrew setup. Is that what you want? Try temporarily removing cmake from your path in your shell of choice, then purging the cmake cache

```bash
rm CMakeCache.txt 
```

Now run the install instructions as normal. Does that make it build for you? It does for me.

There is probably a way of makeing this go by setting config variables for CMake or editing `CMakeCache.txt`

weilintuye95, 2020-11-05:
Ah you are right, sorry for the late reply! I i had some old linuxbrew folders from previous installations, so all i did was delete the linuxbrew folder in /home  with `sudo rm -rf /home/linuxbrew`) and everything worked. Thanks @danmackinlay!!

closed 2020-11-05 by weilintuye95

lrettig, 2021-01-31:
Ran into the same issue. In my case, it was finding `pkg-config` inside linuxbrew. Assuming you don't want to totally blow away `/home/linuxbrew`, you can just temporarily remove it from your `PATH` variable when you run `cmake` and it should work.

americanhanko, 2022-02-11:
Had the same issue. I ended up putting the following into my shell profile so that `pkg-config` won't have issues finding `.pc` files in the future, regardless of where they're installed:

```shell
export PKG_CONFIG_PATH="$(/usr/bin/pkg-config --variable pc_path pkg-config)":"$(/home/linuxbrew/.linuxbrew/bin/pkg-config --variable pc_path pkg-config)"
```

Which results in:
```
❯ echo $PKG_CONFIG_PATH | tr : \n
/usr/local/lib/x86_64-linux-gnu/pkgconfig
/usr/local/lib/pkgconfig
/usr/local/share/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/pkgconfig
/usr/share/pkgconfig
/home/linuxbrew/.linuxbrew/lib/pkgconfig
/home/linuxbrew/.linuxbrew/share/pkgconfig
/home/linuxbrew/.linuxbrew/Homebrew/Library/Homebrew/os/linux/pkgconfig
```

After that, removing `CMakeCache.txt` and re-running `cmake ..` worked:

```
logiops/build on  master via △ v3.16.3
❯ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
LogiOps Version Number: v0.2.2-70-gdbe2b28
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /home/linuxbrew/.linuxbrew/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'libevdev'
--   Found libevdev, version 1.9.0
-- Checking for module 'systemd'
--   Found systemd, version 245
-- Checking for module 'libconfig'
--   Found libconfig, version 1.5
-- Checking for module 'libudev'
--   Found libudev, version 245
-- systemd units will be installed at /lib/systemd/system
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ehanko/dev/github.com/PixlOne/logiops/build
```