PixlOne/logiops #143: Swap horizontal and vertical scroll

dsmaynard, 2020-10-23:
I can't find any suggestions for how to do this in the documentation, but I'm curious if it's possible to swap the vertical and horizontal scroll wheels. That is, make the thumb scroll wheel be the main vertical scroll, the the middle scroll be horizontal. This seems like it should be easy, but I'm been fighting with he settings and haven't had any luck (on Arch).

PixlOne, 2020-10-23:
Remapping the horizontal scroll wheel is only supported on the MX Master 3 (and even then, I'm not sure if it works). As for the vertical scroll wheel, you can set the up/down axis to `REL_HWHEEL_HI_RES` or `REL_HWHEEL` to set it as the horizontal scroll wheel on most mice. This is in the `hiresscroll` section in the config page on the wiki.

Hope that helps.

dsmaynard, 2020-10-23:
Thanks for the quick reply. This would be for the MX master 3. Honestly I'm less interested in changing the middle scroll wheel -- I really just want to add vertical scroll capabilities to the thumb wheel. I'd be fine with both wheels being vertical if that was possible. But it sounds like adding vertical scroll to the thumb is challenging, even of the MX master 3?

felagund, 2020-11-10:
I have MX Vertical adn that only has one wheel. Could I set it up so that when I press a button (either the DPI switch or more likely one of the back/forward buttons), scrolling happens horizontally? The config file seems to enable just configuring a function of one button, not combinations.