PixlOne/logiops #144: Failing to load configuration at boot

rtowsley, 2020-10-24:
Hello, I am on Manjaro, and logiops is being told to start as a a service at boot time via "sudo systemctl enable logid". 

What I have found, is that the configuration will only load properly if the mouse is "awake" at boot time, for example I need to ensure that I wiggle the mouse when I boot the computer. If I fail to do so, logid doesn't see that the mouse is available, and the configuration is not loaded when I finally do wiggle the mouse. The solution is simply "sudo systemctl restart logid" however there must be a way to make this step unnecessary. 

Thanks!

viperML, 2021-06-06:
I had a similiar problem when booting, the systemd service started but it showed the warning:

` [WARN] Error adding device /dev/hidraw3: Invalid function ID`

I also read about a solution in #204 to change from multi-user.target to graphical.target, but that didn't solve it.
Restarting the service everytime I boot the computer seemed to solve the issue. 

In the end I recompiled my kernel to include all Logitech related stuff, instead of building them as modules.

```
Device Drivers --->
    HID support --->
        Special HID Drivers --->
            Logitech devices
                ...
```

System:
- Arch Linux
- Kernel linux-zen based, version 5.12.9-zen1
- Logiops 0.2.3 (AUR package logiops version 0.2.3-1)

rtowsley, 2021-06-07:
Hello,
This is a fairly old post. I haven't had this issue in a while, I suspect it was fixed in a more recent version. I am currently running version r206.a0687c8-1 installed via AUR (logiops-git package) without encountering this issue at boot anymore.

brainplot, 2021-06-13:
I was getting the same exact issue with an MX Master 3 for Mac. I worked around the issue by editing the main service with `systemctl edit --full logid.service` to have the following content:
```
[Unit]
Description=Logitech Configuration Daemon
Wants=bluetooth.target

[Service]
Type=simple
ExecStart=/usr/bin/logid
User=root
ExecReload=/bin/kill -HUP $MAINPID
Restart=on-failure

[Install]
WantedBy=graphical.target
```
and then create a `systemd.timer` for the servivce with `systemctl edit --force --full logid.timer`, with the following content in it:
```
[Unit]
Description=Start LogiOps With Some Delay

[Timer]
OnBootSec=5

[Install]
WantedBy=timers.target
```

This way it seems to work on boot. I'm on Arch Linux.

Perhaps this can give @PixlOne an insight on what the issue is and how to properly fix it (?)

wooparadog, 2021-06-16:
@brainplot can you provide the svg produced by the following command? 

```
$ systemd-analyze plot > systemd-analyze.svg
```

brainplot, 2021-06-17:
I can upload it if you want but I've later realized that I didn't actually fix anything. I still need to wiggle my mouse for `logid` to work, just like @rtowsley initially described. My half-workaround only gives me a bit more time to move my mouse before the service tries to start.