PixlOne/logiops #145: Thumbwheel / Horizontal wheel on MX Master

DyonR, 2020-10-27:
For more than 30 minutes I've been trying to get the thumbwheel working on my MX Master...  
I've looked at #106, but that seems to be related only to the MX Master 3  
  
My goal is to have my thumbwheel influence my volume levels.  

Right now this is my thumbwheel **config**, taken from #106 :
```
thumbwheel:
{
	divert: true
	invert: false
	left: {	
		mode: "OnInterval"
		pixels: 1
		action: { 
			type: "Keypress"
			keys: ["KEY_VOLUMEUP"]
		}
	}
	right: {
		mode: "OnInterval"
		pixels: 1
		action: { 
			type: "Keypress"
			keys: ["KEY_VOLUMEDOWN"] 
		}
	}
};
```  

**evtest** on my thumbwheel value -1 is moving it up (right, if using horizontal scroll), and 1 being down (left):
```
Event: time 1603757361.144572, type 2 (EV_REL), code 6 (REL_HWHEEL), value -1
Event: time 1603757361.144572, type 2 (EV_REL), code 12 (REL_HWHEEL_HI_RES), value -120
Event: time 1603757361.144572, -------------- SYN_REPORT ------------
Event: time 1603757362.376575, type 2 (EV_REL), code 6 (REL_HWHEEL), value 1
Event: time 1603757362.376575, type 2 (EV_REL), code 12 (REL_HWHEEL_HI_RES), value 120
Event: time 1603757362.376575, -------------- SYN_REPORT ------------
```

**logid -v DEBUG**
```
[INFO] Device found: Wireless Mouse MX Master on /dev/hidraw6:1
[DEBUG] /dev/hidraw6:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
```

**hidpp-list-devices**
```
/dev/hidraw6: Logitech USB Receiver (046d:c52b) HID++ 1.0
/dev/hidraw6 (device 1): MX Master (046d:4069) HID++ 4.5
```  

**hidpp-list-features /dev/hidraw6 -d 1**
```
MX Master (046d:4069) is a HID++ 4.5 device
Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x1d4b] Wireless device status
Feature 0x05: [0x0020] Reset
Feature 0x06: [0x0021] Crypto Identifier
Feature 0x07: [0x1000] Battery status
Feature 0x08: [0x1806] Configurable device properties (hidden, internal)
Feature 0x09: [0x1814] Change host
Feature 0x0a: [0x1b04] Reprog controls v4
Feature 0x0b: [0x2201] Adjustable dpi
Feature 0x0c: [0x2100] Vertical scrolling
Feature 0x0d: [0x2110] Smart shift
Feature 0x0e: [0x2121] Hi-res wheel
Feature 0x0f: [0x6501] Gesture 2
Feature 0x10: [0x00c2] DFUcontrol 3
Feature 0x11: [0x1813] ? (hidden, internal)
Feature 0x12: [0x1830] ? (hidden, internal)
Feature 0x13: [0x1890] ? (hidden, internal)
Feature 0x14: [0x1891] ? (hidden, internal)
Feature 0x15: [0x18a1] ? (hidden, internal)
Feature 0x16: [0x18c0] ? (hidden, internal)
Feature 0x17: [0x1df3] ? (hidden, internal)
Feature 0x18: [0x1e00] Enable hidden features (hidden)
Feature 0x19: [0x1eb0] ? (hidden, internal)
Feature 0x1a: [0x1803] ? (hidden, internal)
Feature 0x1b: [0x1861] ? (hidden, internal)
Feature 0x1c: [0x9001] ? (hidden, internal)
Feature 0x1d: [0x9200] ? (hidden, internal)
Feature 0x1e: [0x9202] ? (hidden, internal)
Feature 0x1f: [0x1805] OOBState (hidden, internal)
```
  
Maybe I am just retarded, but I am quite new to all of this :)

brandesjj, 2020-11-16:
Did you find a solution? I have the same issue.

DyonR, 2020-11-16:
> Did you find a solution? I have the same issue.

Unfortunately, no 😞

brandesjj, 2020-11-16:
> > Did you find a solution? I have the same issue.
> 
> Unfortunately, no

Alright. I'll probably switch to xbindkeys for volume control then. Thanks anyways!

nonetrix, 2020-12-21:
One of my friends is having this issue can you try running xev and pressing it

nonetrix, 2020-12-21:
Might be able to just do it in a key binding manager or something

nswerhun, 2020-12-25:
> One of my friends is having this issue can you try running xev and pressing it

`xev` shows nothing. It's like I'm not pressing anything at all; nothing registers.

turkino, 2020-12-30:
@DyonR I have been trying to configure my Logitech MX Master 3 for 2 days with logiops, because solaar does not provide specific button configuration, and I have not worked a lot with the solaar-git version. Anyway, right now, I can control the volume with my thumbwheel. The problem I have, is that I have to lift my finger everytime I increase/decrease 5% of my volume. For example, if I want to increase the volume from 0% to 15%, I have to scroll 1 time, lift my thumb, scroll a second time, lift my thumb, scroll a third time. Here is the thumbwheel configuration:

    thumbwheel:
    {
        divert: true;
        invert: false;
        left:
        {
            mode: "OnInterval";
            interval: 1;
            action =
            {
                type: "Keypress";
                keys: ["KEY_VOLUMEDOWN"];
            };
        },
        right:
        {
            mode: "OnInterval";
            interval: 1;
            action =
            {
                type: "Keypress";
                keys: ["KEY_VOLUMEUP"];
            };
        };
    };

I changed "interval" variable with "pixels", worked the same, from what I experienced. From wiki: ":" and "=" are used for defining variables and are interchangeable. (e.g. name: "foo"; is the same as name = "foo";). The only difference I see that can be affecting, is that i have a lot of commas "," and semicolons ";". From my experience. sometimes the debug mode of logiops does not tell you about errors when leaving out comas or semicolons in the configuration lines.

turkino, 2020-12-30:
@DyonR Also, this is my logiops debug, and the thumbwheel is there:

[20-12-30|13:37:41|turkino@argon:/home/turkino]>> logid -v
[WARN] Error adding device /dev/hidraw3: open failed: Permission denied
[DEBUG] All workers were busy for 500ms, running task in new thread.
[INFO] Detected receiver at /dev/hidraw0
[DEBUG] All workers were busy for 500ms, running task in new thread.
[DEBUG] All workers were busy for 500ms, running task in new thread.
[WARN] Error adding device /dev/hidraw2: std::exception
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw0:1
[DEBUG] /dev/hidraw0:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        |
[DEBUG] 0x51 |         |         | YES        |
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
[WARN] Error adding device /dev/hidraw1: std::exception

turkino, 2020-12-30:
@DyonR Also, this is my logiops debug, and the thumbwheel is there:

turkino>> logid -v
[WARN] Error adding device /dev/hidraw3: open failed: Permission denied
[DEBUG] All workers were busy for 500ms, running task in new thread.
[INFO] Detected receiver at /dev/hidraw0
[DEBUG] All workers were busy for 500ms, running task in new thread.
[DEBUG] All workers were busy for 500ms, running task in new thread.
[WARN] Error adding device /dev/hidraw2: std::exception
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw0:1
[DEBUG] /dev/hidraw0:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        |
[DEBUG] 0x51 |         |         | YES        |
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
[WARN] Error adding device /dev/hidraw1: std::exception

turkino, 2020-12-30:
@DyonR The same person you got the configuration from, same the same thing I'm saying. When you configure the thumbwheel this way, you have to lift the finger everytime:

"Here's some of it. What happens is it works like the "OnThreshold" gesture instead of like "OnInterval"(even though it's set as "OnInterval"). So once you move it X pixels, it presses and releases the button, and won't let you press it again until you move your thumb away from the thumbwheel or move it in the opposite direction. So effectively, a weird tap is more useful than scrolling it like someone normally would."

https://github.com/PixlOne/logiops/issues/106#issuecomment-687908986

Gorzen, 2021-01-04:
Indeed, I have converged to a setup almost identical to yours and it also works with a weird tap. Continuous scrolling doesn't redo the action :disappointed:

Gorzen, 2021-01-04:
By changing a bit the code I managed to make it "work" but I have no idea if that's the proper way to go about it. It feels a bit wonky but it might give some ideas.

There was a check in `src/logid/features/ThumbWheel.cpp` to see if the direction has changed, I simply removed it and with `interval: 2;` the speed is close to manageable but if I increase it more it seems to have weird behavior.

```diff
diff --git a/src/logid/features/ThumbWheel.cpp b/src/logid/features/ThumbWheel.cpp
index 57bb055..b8077b2 100644
--- a/src/logid/features/ThumbWheel.cpp
+++ b/src/logid/features/ThumbWheel.cpp
@@ -152,13 +152,11 @@ void ThumbWheel::_handleEvent(hidpp20::ThumbWheel::ThumbwheelEvent event)
                 opposite_scroll = _config.rightAction();
             }
 
-            if(direction != _last_direction) {
-                if(opposite_scroll)
-                    opposite_scroll->release();
-                if(scroll_action) {
-                    scroll_action->press(true);
-                    scroll_action->move(direction * event.rotation);
-                }
+            if(opposite_scroll)
+                opposite_scroll->release();
+            if(scroll_action) {
+                scroll_action->press(true);
+                scroll_action->move(direction * event.rotation);
             }
 
             _last_direction = direction;
```

Starvexx, 2021-01-25:
@Gorzen Thank you so much, This worked exactly as I expected. It indeed feels a little wonky, but its not too bad

marxlaml, 2021-03-11:
@Gorzen Your two last contributions are everything I needed to solve (or work around) this problem. I doubt this fix can be put in a PR though.. (right?)

closed 2021-03-14 by PixlOne

itsKaytertot, 2025-10-24:
I was able to do this with the config section below, didn't need to modify the thumbwheel file, it honestly feels less wonky than it did on Windows

```
thumbwheel:
     {
        divert: true;
        invert: false;
        left:
        {
            threshold: 100;
            interval: 3;
            direction: "Left";
            mode: "OnInterval";
            action =
            {
                type: "Keypress";
                keys: ["KEY_VOLUMEDOWN"];
            };
        };
        right:
        {
            threshold: 100;
            interval: 3;
            direction: "Right";
            mode: "OnInterval";
            action =
            {
                type: "Keypress";
                keys: ["KEY_VOLUMEUP"];
            };
        };
}; 
```

PR #215 (2021-03-14): Fix ThumbWheel feature direction bug
Should fix #145 for the MX Master 3

Thank you @Gorzen for this change, this is basically your change just cleaned up a bit.
commit 48399a1d (2021-03-14): Fix ThumbWheel feature direction bug

Should fix #145 for the MX Master 3