PixlOne/logiops #148: Driver does not work after suspend

lowski, 2020-11-07:
Hey!
Thank you for this driver. It makes my Logitech MX Master 2S configurable for Linux.

I encountered one issue though: after suspend  (on ElementaryOS), drivers seems to stop working (even though the service is still active and running) - i.e. scroll configuration is lost. I have to restart the service so everything works as defined in the file.

Any idea what can cause this issue?

gobbedy, 2020-11-14:
I second that this driver is awesome. Thank you so much.

I have a similar issue with my MX Master 3. Although it's not after a suspend. I have my system set up to never lock/suspend/hibernate.

I left my appartment and came back 7 hours later, and the buttons of my mouse was unresponsive. Similarly to the previous commentor, the logid service was still running.

After restarting the service, the mouse buttons became responsive.

Not a huge deal and this is the first time it happens after a week of use. Overall a fantastic driver that I'm grateful for.

lowski, 2020-11-15:
@gobbedy for me it happens 90% of time after leaving computer for ~1-2 hours. The most annoying thing is scroll, as it changes to some very low value and to a scroll single line of text on the page, I have to spin a wheel many times. `service logid restart` solves the problem.

gobbedy, 2020-11-15:
@lowski Ah yes, that happens to me too now that you mention it. Suddenly my main scroller will become extremely slow. Since the main scroller works without logiops I didn't associate this issue with this driver, but it must surely be the same issue. Thanks for the tip I'll try restarting the service next time!

gobbedy, 2020-11-15:
@lowski confirmed! I just had the slow scroller issue happen to me. Restarting the logid service fixed it. Thank you!

I have a ton of messages of this nature in my syslog, not sure if related:
```Nov 15 10:42:53 kaligg logid[628]: [WARN] Error adding device /dev/hidraw2: std::exception```

lowski, 2020-11-15:
> I have a ton of messages of this nature in my syslog, not sure if related:
`Nov 15 10:42:53 kaligg logid[628]: [WARN] Error adding device /dev/hidraw2: std::exception`

I have them as well but not sure if they are related to this issue as I've seen them even after OS boot.

gobbedy, 2020-11-15:
Going through syslog, this is the only error (as opposed to warning) I have related to logid:
```/var/log/syslog:Nov 15 09:47:51 kaligg logid[634]: [ERROR] Failed to add device 1 to receiver on /dev/hidraw3: _readReport read failed: Input/output error```

gobbedy, 2020-11-16:
It's just happened again, and my log again contains the warning from me previous comment:
```Nov 16 13:14:08 kaligg logid[628]: [WARN] Error adding device /dev/hidraw2: std::exception```

That's twice in a row now that this message happened shortly before the problem arose. At the very least, I cannot rule out that this warning is related.

gobbedy, 2020-11-16:
Woah. I just restarted the service, and the issue remained.

Sure enough I looked at my log, and the same warning just popped up:
```Nov 16 13:24:30 kaligg logid[3787]: [WARN] Error adding device /dev/hidraw2: std::exception```

I then restart the service again. This time no message, and no scroll issue.

I'm now quite convinced that the warning explains the problem.

lowski, 2020-11-23:
@gobbedy thanks for further investigation. I actually think that both issues are related as by "it does not work" here, I mean the scrolling issue you also encountered and described in #156 :wink:

lukasz-gosiewski, 2020-12-06:
I think I've got the exact same issue on my ArchLinux machine. In case any diagnostic help is needed - feel free to ping me. Good luck with a fix!

lowski, 2020-12-26:
@gobbedy @lukasz-gosiewski I've recently switched to https://github.com/pwr-Solaar/Solaar, it's definitely not as customizable as `logiops` but at least scroll keeps working after suspend (with `--restart-on-wake-up` option). I know it doesn't solve the issue but thought I will share it.

tomekand11, 2021-01-06:
i just did a custom basic udev rule to restart logid after sleep or reconection in /etc/udev/rules.d

`ACTION=="add", SUBSYSTEM=="bluetooth", KERNEL=="hci0:[0-9]*", RUN+="mx.sh"`

add script in /lib/udev so u dont need to add full path

lowski, 2021-01-06:
@tomekand11 thanks for the tip! What's the content of `mx.sh`?

tomekand11, 2021-01-07:
Hi @lowski, yeah is just an simple restart of logid, 

`#!/bin/bash  

systemctl restart logid
`

Sothatsit, 2021-02-11:
I have also been running into this issue constantly. I get the same issue with the `std::exception` line, followed by extremely slow scrolling.

Unfortunately for me, just restarting my logid service does not seem to solve the issue...

NewRedsquare, 2021-03-22:
> Hi, @lowski yeah is just an simple restart of logid,
> 
> mx.sh:
> 
> `#!/bin/bash`
> `systemctl restart logid`
> 
> then just restart udev with:
> `sudo udevadm control --reload-rules ; udevadm trigger`

This seems to work for me +1

ExposedCat, 2021-06-30:
https://github.com/PixlOne/logiops/issues/129#issuecomment-871666675