PixlOne/logiops #153: How to adjust the horizontal scroll BUTTON speed for M720?

yizdu, 2020-11-13:
First of all, I'm very grateful to the author for creating this project.
I have a little problems when using it. 
I'm using the Logitech M720, which doesn't have thumb wheel but has a 4-way wheel. The wheel can be press to the side to achieve left and right scroll. Whether I use logiops or not, my horizontal scroll button is very slow, and I wonder if there is any way to adjust its speed.
I did try add the `thumbwheel` option in the `logid.cfg` like this:
```    
thumbwheel:
    {
        divert: true;
        invert: false;
        left: {
            mode: "Axis";
            axis: "REL_HWHEEL_HI_RES";
            axis_multiplier: -10;
        },
        right: {
            mode: "Axis";
            axis: "REL_HWHEEL_HI_RES";
            axis_multiplier: 10;
        };
    }
```
But it doesn't work.
Then I tried to remap the left and right scroll button. The left scroll button named `button 6` in `xev`, the `cid` is `0x5b`. I tired to simulate the left scroll event first, use the `evtest` capture the input event, the output is:
```
Event: time 1605262840.231548, type 2 (EV_REL), code 6 (REL_HWHEEL), value -1
Event: time 1605262840.231548, type 2 (EV_REL), code 12 (REL_HWHEEL_HI_RES), value -120
Event: time 1605262840.231548, -------------- SYN_REPORT ------------
```
And then I realized that I can't simulate it with `Keypress` option. 
So is there a way to adjust the speed of the horizontal scroll button? Maybe I should use the `axis` mode to set a gesture, but I did read the configuration page, but still don't know how. 
`imwheel` can set the horizontal scrolling speed, but I'm looking for a simpler (i.e. only logiops) solution.
Any help would be appreciated, thanks.

tsnieman, 2020-12-12:
I'm also interested in a solution for adjusting the thumbwheel scroll speed. I would have assumed that I could use some sort of `hires` or `multiplier` property, but I haven't been able to figure it out either.

S-trace, 2021-05-08:
Please take a look at the pull request https://github.com/PixlOne/logiops/pull/243 - it allows to map mouse buttons to axis changes (use `REL_HWHEEL_HI_RES` or `REL_HWHEEL` for vertical scrolling and  `REL_HWHEEL_HI_RES` or `REL_HWHEEL` for horizontal scrolling).

weizhizhu, 2022-05-13:
Can you provide me with a M720 configuration file，thank！！！！

yizdu, 2022-05-20:
> Can you provide me with a M720 configuration file，thank！！！！

My blog might answer your questions
https://segmentfault.com/a/1190000039985213