PixlOne/logiops #156: "Error adding device" -- need to restart logid service to fix

gobbedy, 2020-11-18:
Hello and thank you for your wonderful driver.

This issue is related to the problem I described [here](https://github.com/PixlOne/logiops/issues/148) issue but I thought I'd open my own issue rather than continue to hijack someone else's.

The following warning appears to be the problem:
```[WARN] Error adding device /dev/hidraw2: std::exception```

I always get the error when I run ```sudo logid -v``` (bold emphasis is mine):
```
[DEBUG] Unsupported device /dev/hidraw4 ignored
**[WARN] Error adding device /dev/hidraw2: std::exception**
[DEBUG] Unsupported device /dev/hidraw0 ignored
[INFO] Detected receiver at /dev/hidraw1
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw1:1
[DEBUG] /dev/hidraw1:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```
The warning occurs no matter what my ```logid.cfg``` is; or indeed whether I have one or not.

Still, in case it matters, I got the above output with the following ```/etc/logid.cfg```:
```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    buttons: (
        {
            # Forward button
            cid: 0x56;
            action =
            {
                type :  "Keypress";
                keys: ["KEY_VOLUMEUP"];
            };
        }
    );
}
);
```

Note that when I run the above command (```sudo logid -v```), the warning appears harmless: the driver still works without issue.

However, when logid is started as a service, the same warning appears at random, and when it does, I get the scrolling issues described in the issue 148.

I then have to restart the service in order to get my mouse working properly again.

This happens on average every 2 hours or so, enough to be annoying lol.

I'd be grateful for any help resolving this issue.

Here is the output of my ```uname -v```:
```#1 SMP Debian 5.7.6-1kali2 (2020-07-01)```

Thanks in advance!

fastcat, 2021-10-04:
I have a very similar issue -- mine shows up when disconnecting & reconnecting the receiver ... which in my case is really when switching systems on my KVM, but the same thing happens if I manually unplug & re-plug the receiver.

I specifically see the "running it interactively with `-v` avoids the problem" ... and in fact I found that setting the _systemd_ service to use `-v` _also_ works around the issue!

leviable, 2021-11-05:
I'm having the same or similar issue as @fastcat, specifically getting logid to work correctly with my Master 2S after using a KVM to switch between work stations. Bouncing the service rarely works, but manually running logid with `-v` makes it work most of the time.

I suspect there is some sort of timeout that needs to happen first. On this particular work station (Ubuntu 20.04.3), sometimes Ill remote desktop into it, and sometimes switch from RDP to the actual box with my KVM. If I don't wait ~20 seconds between killing the RDP session and logging into the box via the KVM, my sound wont work, as there is a 20 second timeout that needs to pop. I suspect something similar is occurring here.

DimitriDaSilva, 2021-12-09:
I'm having the same issue. Restarting the service also fixes it for me

hbarcelos, 2022-07-29:
Same issue here. MX Master 3, connected through bluetooth.

jandresfg, 2022-08-13:
Same issue here. MX Master 3, connected through bluetooth. This keeps me from launching successfully at startup, gotta always restart the service manually for it to work...

NoChance777, 2022-08-17:
Same here. The device is MX 3 Master 3 Mac, connected via Bluetooth. Adding 'After=bluetooth.target' in service file does not solve the problem.

gnos28, 2022-08-17:
same issue for me as well
mx master 3 bluetooth / ubuntu 22

very likely because bluetooth mouse isn't automatically paired when system boot (you have to click any mouse button to "activate it")

might be solved with a tweak in service file like @NoChance777 said but I'm not enough comfortable to play with that

wmbruckner, 2022-11-04:
Having the same issue here when starting (Kubuntu 22.04), modified logid.service did not fix it.
However, if I do not touch the mouse for like some 20 secs after booting to desktop, logid runs just fine.

dszmaj, 2022-11-24:
same  problem here with master 3 on fedora 37

im-calvin, 2023-01-21:
Just to bump this thread I'm running into the same issue with Mx Master 3 on Fedora 37 using exclusively Bluetooth.

I've tried some of the fixes in the linked thread but the main issue is that when the laptop goes to sleep / mouse goes to sleep I run into the following error: `[WARN] Error adding device /dev/hidraw1: Invalid function ID`.
This error doesn't show up when I either restart or fully shut down the system, only when something goes to sleep.

https://github.com/PixlOne/logiops/issues/204#issuecomment-841861932

im-calvin, 2023-04-27:
I fixed the issue by removing the Solaar package which I had already installed on my system. I presume the issue was that they were both fighting for the mouses' config.
Hope this helps others!

fastcat, 2023-04-27:
> I fixed the issue by removing the Solaar package

You're my hero for today :)

I still see the `[WARN] Error adding device /dev/hidraw13: std::exception` error in the logs ... but the mouse still gets properly reconfigured on insert now.

im-calvin, 2023-04-28:
Yeah I get that same error as well, seems like the drivers work though.

I initially installed Solaar to monitor the battery life, kinda sucks that they aren't compatible with each other. Maybe a potential improvement?

fastcat, 2023-04-28:
I maybe spoke too soon yesterday. Disabling/removing solaar _does_ make things better, but it doesn't fix everything. When switching away & back with my KVM, I'm now getting the button-remapping working right where it sometimes wouldn't before, but my scroll wheel is _sometimes_ not getting reconfigured for the correct scrolling speed (but sometimes is).

im-calvin, 2023-04-28:
I'll look into it when I get a chance, I haven't actually had a chance to use my Linux device all that much but I'll see if I get the same behaviour.

im-calvin, 2023-05-09:
Following up, I've been seeing some buggy behaviour with the service. Sometimes when the service is not on none of my binds are working (including scroll wheel which I guess used to be fixed with the `Solaar` package). Not too sure what the fix is atm.
Seems to be reproducible whenever I turn my Bluetooth on my device on / off? When I power cycle the mouse it fixed it.
When I set my device to suspend then it also stops working.

im-calvin, 2023-05-09:
I seem to have possibly fixed the issue by changing `hirresscroll: { hires: false }` to `true` in my config file.
Not sure if this is a permanent fix but I found the solution through #116

raymondshiner, 2023-11-08:
Any updates on this? I am having the same problem. As far as i can tell, i have uninstalled Solaar, but it still wont work on reboot. I have to restart service via CLI to get gesture button func.

raymondshiner, 2023-11-08:
So not a fix, but a workaround. The AI Overlord suggested I make a seperate timer service to restart logid shortly after restart. It works now i have gestures after logging in, without having to manually restart.

https://chat.openai.com/share/1393c511-d92b-41f8-b019-742d9cda2ef4

arthurbarros, 2023-11-20:
> So not a fix, but a workaround. The AI Overlord suggested I make a seperate timer service to restart logid shortly after restart. It works now i have gestures after logging in, without having to manually restart.
> 
> https://chat.openai.com/share/1393c511-d92b-41f8-b019-742d9cda2ef4

Worked for me, not a fix but I will take the workaround. @raymondshiner thank you

c-kr, 2024-01-11:
Had the same problem with MX Master 3 via bluetooth. A better approach instead of starting with a delay is a udev rule that starts logid after the hidraw device appears:

```
cat /etc/udev/rules.d/50-restart_logid.rules 
ACTION=="add", KERNEL=="hidraw0", RUN+="/usr/bin/systemctl --no-block restart logid.service"
```

This allows to disable the logid.service and have the daemon just started once, right after connection without  a need to restart.

Tested on arch with the logiops AUR package

lukash, 2024-11-21:
The above didn't work for me, had to use (MX Anywhere 3):
```
ACTION=="add", KERNELS=="hci0", SUBSYSTEM=="bluetooth", RUN+="/usr/bin/systemctl --no-block restart logid.service"
```

lexouma, 2025-08-25:
[c-kr](https://github.com/c-kr) solution's worked for me! 
The only thing was that by looking at the logid logs, my Master MX 3s is found on `hidraw8`, and so after creating the `/etc/udev/rules.d/50-restart_logid.rules` file I put this in : 
```
ACTION=="add", KERNEL=="hidraw8", RUN+="/usr/bin/systemctl --no-block restart logid.service"
```

Using hidraw0 wasn't working, as I think that it is specific to your computer.