PixlOne/logiops #158: Make error with ThumbWheel.cpp.o

ightnapovial, 2020-11-23:
Attempting to build from source (cloned repository, not release package) throws the below error when running 'make' on Kubuntu 20.10:

`[ 21%] Building CXX object src/logid/CMakeFiles/logid.dir/features/ThumbWheel.cpp.o
/home/user/Downloads/logiops/src/logid/features/ThumbWheel.cpp: In static member function ‘static std::shared_ptr<logid::actions::Action> logid::features::ThumbWheel::Config::_genAction(logid::Device*, libconfig::Setting&, const string&)’:
/home/user/Downloads/logiops/src/logid/features/ThumbWheel.cpp:230:36: error: no match for ‘operator[]’ (operand types are ‘libconfig::Setting’ and ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’})
  230 |         auto& a_group = config_root[name];
      |                                    ^
In file included from /home/user/Downloads/logiops/src/logid/features/../actions/gesture/../Action.h:22,
                 from /home/user/Downloads/logiops/src/logid/features/../actions/gesture/Gesture.h:21,
                 from /home/user/Downloads/logiops/src/logid/features/ThumbWheel.h:23,
                 from /home/user/Downloads/logiops/src/logid/features/ThumbWheel.cpp:19:
/usr/include/libconfig.h++:223:13: note: candidate: ‘libconfig::Setting& libconfig::Setting::operator[](const char*) const’
  223 |   Setting & operator[](const char *name) const;
      |             ^~~~~~~~
/usr/include/libconfig.h++:223:36: note:   no known conversion for argument 1 from ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’} to ‘const char*’
  223 |   Setting & operator[](const char *name) const;
      |                        ~~~~~~~~~~~~^~~~
/usr/include/libconfig.h++:224:13: note: candidate: ‘libconfig::Setting& libconfig::Setting::operator[](int) const’
  224 |   Setting & operator[](int index) const;
      |             ^~~~~~~~
/usr/include/libconfig.h++:224:28: note:   no known conversion for argument 1 from ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’} to ‘int’
  224 |   Setting & operator[](int index) const;
      |                        ~~~~^~~~~
/home/user/Downloads/logiops/src/logid/features/ThumbWheel.cpp: In static member function ‘static std::shared_ptr<logid::actions::Gesture> logid::features::ThumbWheel::Config::_genGesture(logid::Device*, libconfig::Setting&, const string&)’:
/home/user/Downloads/logiops/src/logid/features/ThumbWheel.cpp:247:36: error: no match for ‘operator[]’ (operand types are ‘libconfig::Setting’ and ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’})
  247 |         auto& g_group = config_root[name];
      |                                    ^
In file included from /home/user/Downloads/logiops/src/logid/features/../actions/gesture/../Action.h:22,
                 from /home/user/Downloads/logiops/src/logid/features/../actions/gesture/Gesture.h:21,
                 from /home/user/Downloads/logiops/src/logid/features/ThumbWheel.h:23,
                 from /home/user/Downloads/logiops/src/logid/features/ThumbWheel.cpp:19:
/usr/include/libconfig.h++:223:13: note: candidate: ‘libconfig::Setting& libconfig::Setting::operator[](const char*) const’
  223 |   Setting & operator[](const char *name) const;
      |             ^~~~~~~~
/usr/include/libconfig.h++:223:36: note:   no known conversion for argument 1 from ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’} to ‘const char*’
  223 |   Setting & operator[](const char *name) const;
      |                        ~~~~~~~~~~~~^~~~
/usr/include/libconfig.h++:224:13: note: candidate: ‘libconfig::Setting& libconfig::Setting::operator[](int) const’
  224 |   Setting & operator[](int index) const;
      |             ^~~~~~~~
/usr/include/libconfig.h++:224:28: note:   no known conversion for argument 1 from ‘const string’ {aka ‘const std::__cxx11::basic_string<char>’} to ‘int’
  224 |   Setting & operator[](int index) const;
      |                        ~~~~^~~~~
make[2]: *** [src/logid/CMakeFiles/logid.dir/build.make:219: src/logid/CMakeFiles/logid.dir/features/ThumbWheel.cpp.o] Error 1
make[2]: Leaving directory '/home/user/Downloads/logiops/build'
make[1]: *** [CMakeFiles/Makefile2:94: src/logid/CMakeFiles/logid.dir/all] Error 2
make[1]: Leaving directory '/home/user/Downloads/logiops/build'
make: *** [Makefile:130: all] Error 2`

ightnapovial, 2020-11-23:
Sorry for the formatting.

JMoerman, 2020-11-23:
Just came across this project and also ran into this issue. Using the GitHub blame view it was easy to find the culprit though: 28a93b2df431e36b0e1a84171b5a8a88a3c5a0c1.

lowski, 2020-11-23:
I encountered the same problem with latest version.

PixlOne, 2020-11-23:
I've reverted this commit, so it should be working now.

lowski, 2020-11-24:
@PixlOne thanks! I confirm that latest version compiles without issues.

closed 2023-05-14 by PixlOne