PixlOne/logiops #165: [ERROR] Could not create input device: No such file or directory

oliverwelter, 2020-11-28:
I installed logiops version 0.2.2 on a Gentoo system using https://gpo.zugaina.org/sys-apps/logiops
The daemon is started via systemd but only shows an error:

`logid -V`
`v0.2`

`systemctl status logid` :
```logid.service - Logitech Configuration Daemon
     Loaded: loaded (/lib/systemd/system/logid.service; enabled; vendor preset:>
     Active: active (running) since Sat 2020-11-28 11:10:33 UTC; 34min ago
   Main PID: 2247 (logid)
      Tasks: 2 (limit: 37912)
     Memory: 3.7M
     CGroup: /system.slice/logid.service
             └─2247 /usr/bin/logid
```
Using the command directly
`logid -v`
`[ERROR] Could not create input device: No such file or directory`

I copied https://github.com/PixlOne/logiops/blob/master/logid.example.cfg to `/etc/logid.cfg`

MarcWillem, 2020-12-01:
same for me.

cat /dev/input/by-id/usb-Logitech_USB_Receiver-if02-event-mouse yields events on mouse move / key presses.

wdiazux, 2021-01-20:
I have the same problem, I don't know if its possible to solve it with udev rules, I tried some rules that I found in other issues but didn't work.

wdiazux, 2021-01-22:
After trial and error, I found that I was missing uinput module in my kernel, this is my ebuid:
logiops-9999.ebuild 
```
# Copyright 2021 Gentoo Authors
# Distributed under the terms of the GNU General Public License v2

EAPI=7

inherit linux-info cmake

DESCRIPTION="Unofficial driver for Logitech devices"
HOMEPAGE="https://github.com/PixlOne/logiops"

if [[ "${PV}" == 9999 ]] ; then
        inherit git-r3
        EGIT_REPO_URI="https://github.com/PixlOne/logiops.git"
        S="${WORKDIR}/${PN}-${PV}"
else
        SRC_URI="https://github.com/PixlOne/${PN}/archive/v${PV}.tar.gz -> ${P}.tar.gz"
fi

LICENSE="GPL-3"
SLOT="0"
KEYWORDS="~amd64 ~x86"
RESTRICT="mirror"

RDEPEND="dev-libs/libevdev
        virtual/libudev
        dev-libs/libconfig"

DEPEND="${RDEPEND}"
BDEPEND="dev-util/cmake"

pkg_pretend() {
        local CHECK_CONFIG="~HID_LOGITECH ~HID_LOGITECH_HIDPP ~INPUT_UINPUT"
        check_extra_config
}

pkg_postinst() {
        elog "NOTE: The installed systemd daemon is called 'logid'."
        elog "WARNING: No configuration file is included!"
        elog "See https://github.com/PixlOne/logiops/wiki/Configuration"
}
```

oliverwelter, 2021-03-12:
Thank you very much!
Configuring the kernel using the uinput module solved the issue.

closed 2021-03-12 by oliverwelter