PixlOne/logiops #166: No virtual keyboard

kris7t, 2020-11-28:
Since I updated by Arch installation to systemd 247, `sway` does not recognize the virtual keyboard device created by logiops. Thus, keypress events bound to mouse buttons in `logid.conf` are not recognized.

However, no error message is printed to the system journal or to the standard output (even when running `logid -v`).

PixlOne, 2020-11-29:
Is this just on sway? Does this happen with other Wayland desktop environments? What about X11?

I'll look into this when I have time, thanks.

kris7t, 2020-12-01:
I had a bit of time and also tried out GNOME on Wayland and on X11. In both cases, pressing the thumb button on my mouse didn't trigger any keystroke, despite it being bound to a key combination in `logid.conf`. So it seems the uinput device isn't recognized by GNOME either, irrespective of the windowing system.

aragon999, 2020-12-14:
I just looked into the code, and currently cannot see any problem. The following code:
https://github.com/PixlOne/logiops/blob/b554d32cf3a38f701465dbe8e588676be6352cfb/src/logid/InputDevice.cpp#L118-L122
is executed with the (in my opinion) correct arguments (without returning an error). 

Also the creation seems to throw no error, so everything seems to be according to the `libevdev` documentation:
https://www.freedesktop.org/software/libevdev/doc/1.0/group__uinput.html

So maybe there is a problem with systemd 247 and virtual keyboards on Wayland.

As said by the original reporter I cannot see an virtual keyboard device in `sway` using `swaymsg -t get_inputs`.

kris7t, 2020-12-14:
Strangely enough, I am running my own virtual keyboard for other purposes (https://github.com/kris7t/keyboard_remap) which does get correctly recognized under systemd 247.

aragon999, 2020-12-14:
Very strange, I can see the keyboard events in`sudo libinput debug-events`:
```
-event22  KEYBOARD_KEY     +6.182s	*** (-1) pressed
 event22  KEYBOARD_KEY     +6.182s	*** (-1) released
```

But they seem to not reach the window manager.

What is strange (I do not know if it can be related to the issue though), when I press a key I have the following debug output:
```
-event3   KEYBOARD_KEY     +6.744s	*** (-1) pressed
 event3   KEYBOARD_KEY     +6.824s	*** (-1) pressed
```

i.e. a different type of event though.

kris7t, 2020-12-14:
@aragon999 I think the names `event22`, `event3`, etc. just refer to the various evdev devices in your system.

My conjecture would be that the problem is a matter of metadata: the uinput device created by logiops doesn't get recognized as an input device in udev, so compositors don't open it as a keyboard. Compare what `udevadm info --name /dev/input/event<n>` sees about my actual keyboard:

```
P: /devices/platform/i8042/serio0/input/input4/event4
N: input/event4
L: 0
S: input/by-path/platform-i8042-serio-0-event-kbd
E: DEVPATH=/devices/platform/i8042/serio0/input/input4/event4
E: DEVNAME=/dev/input/event4
E: MAJOR=13
E: MINOR=68
E: SUBSYSTEM=input
E: USEC_INITIALIZED=21773931
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_BUS=i8042
E: ID_SERIAL=noserial
E: ID_PATH=platform-i8042-serio-0
E: ID_PATH_TAG=platform-i8042-serio-0
E: LIBINPUT_DEVICE_GROUP=11/1/1:isa0060/serio0
E: DEVLINKS=/dev/input/by-path/platform-i8042-serio-0-event-kbd
E: TAGS=power-switch
E: CURRENT_TAGS=power-switch
```

or my own uinput device from https://github.com/kris7t/keyboard_remap:

```
P: /devices/virtual/input/input27/event20
N: input/event20
L: 0
E: DEVPATH=/devices/virtual/input/input27/event20
E: DEVNAME=/dev/input/event20
E: MAJOR=13
E: MINOR=84
E: SUBSYSTEM=input
E: USEC_INITIALIZED=22307588
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_SERIAL=noserial
E: LIBINPUT_DEVICE_GROUP=3/1/1:py-evdev-uinput
E: TAGS=power-switch
E: CURRENT_TAGS=power-switch
```

with the logiops uinput device:

```
P: /devices/virtual/input/input24/event17
N: input/event17
L: 0
E: DEVPATH=/devices/virtual/input/input24/event17
E: DEVNAME=/dev/input/event17
E: MAJOR=13
E: MINOR=81
E: SUBSYSTEM=input
E: USEC_INITIALIZED=22048163
E: ID_INPUT=1
E: ID_INPUT_MOUSE=1
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_SERIAL=noserial
E: TAGS=power-switch
E: CURRENT_TAGS=power-switch
```

In particular, `LIBINPUT_DEVICE_GROUP` is missing from the logiops device.

Note that `libinput list-devices` simply iterates over `/dev/input` without querying udev metadata, so it correctly sees the logiops device.

simokron, 2020-12-14:
Hey @kris7t

Since already in your OP, you pointed out that a systemd upgrade was to blame, I guess you have confirmed that downgrading just this package addresses the issue? This was the case for me in X11, at least (downgrade from 247.1 to 246.6; see #177).

(as for what causes it, that's a bit beyond my abilities but your last post regarding metadata seems convincing)

kris7t, 2020-12-14:
@simokron If possibly, I'd like to avoid downgrading to systemd 246, but I'll try that later if nothing else helps.

I'm currently trying to get libevdev set the appropriate udev metadata on the uinput device. Following https://github.com/gvalkov/python-evdev/blob/00c0fa76dc589b3d2f72349d35adeb17b03d84cf/evdev/uinput.py#L146 seems to be the way to go, but there are some complications (`python-evdev` doesn't use libedev but communicates with the kernel directly, so some things are slightly different). For some reason, even if I specify

```c
    libevdev_set_id_vendor(device, 0x1);
    libevdev_set_id_product(device, 0x1);
    libevdev_set_id_version(device, 0x1);
    libevdev_set_id_bustype(device, BUS_USB);
    libevdev_set_phys(device, "logiops-uinput");
    libevdev_set_name(device, name);
```

the `LIBINPUT_DEVICE_GROUP` doesn't get set properly.

simokron, 2020-12-14:
@kris7t hmm.. could there be an upstream bug somewhere? Also, any hypothesis why it worked before 247? Seems weird that it would suddenly start breaking if that value wasn't indicated before.

By the way, here is my working `udevadm info --name /dev/input/event<n>` for the logiops button on 246:

```
P: /devices/virtual/input/input44/event22
N: input/event22
L: 0
E: DEVPATH=/devices/virtual/input/input44/event22
E: DEVNAME=/dev/input/event22
E: MAJOR=13
E: MINOR=86
E: SUBSYSTEM=input
E: USEC_INITIALIZED=128467245
E: ID_INPUT=1
E: ID_INPUT_MOUSE=1
E: ID_INPUT_KEY=1
E: ID_INPUT_KEYBOARD=1
E: ID_SERIAL=noserial
E: TAGS=:power-switch:
```

So indeed, it was missing all along.

aragon999, 2020-12-14:
@kris7t thank you for the detailed message, I was not aware of that layer, but that makes sense.

Maybe it is then somehow related to that change: https://github.com/systemd/systemd/blob/28423d9a75cf3e740997545ca13429a93b27181c/NEWS#L5-L87

I will try to look into that later (if I have time to work on this again :-) )

kris7t, 2020-12-14:
This still remains extremely strange: after reading through the libevdev source code and realizing that it doesn't actually call the `UI_SET_PHYS` ioctl and calling that manually on the uinput fd instead, I can add the `E: LIBINPUT_DEVICE_GROUP=0/0/0:logiops-uinput` metadata to the logiops device. However, that's still not enough to be recognized as an input device. Also, I wonder why I can't set the bus type, vendor id and product id, despite libevdev actually calling the associated `UI_DEV_SETUP` ioctl. In fact, if I call it manually, my settings will get overwritten by libevdev.

@aragon999 Interesting find. As far as I can understand, tag handling changes somewhat. Unfortunately, I'm not sure how are udev tags used by libinput. For what it's worth, the only tag I can see in `udevadm` is `power-switch`, which seems hardly relevant.

aragon999, 2020-12-14:
I just create an example program, where everything works as expected:

```main.cpp
#include <iostream>
#include <string.h>

extern "C"
{
#include <libevdev/libevdev.h>
#include <libevdev/libevdev-uinput.h>
}

void test_input()
{
    int err;
    struct libevdev *dev;
    struct libevdev_uinput *uidev;

    std::cout << "Create new device" << std::endl;
    dev = libevdev_new();
    libevdev_set_name(dev, "virtual keyboard");

    std::cout << "Created new keyboard: " << libevdev_get_name(dev) << std::endl;

    libevdev_enable_event_type(dev, EV_KEY);
    libevdev_enable_event_code(dev, EV_KEY, KEY_A, NULL);

    std::cout << "Enabled KEY_A for: " << libevdev_get_name(dev) << std::endl;

    err = libevdev_uinput_create_from_device(dev,
            LIBEVDEV_UINPUT_OPEN_MANAGED,
            &uidev);

    if (err != 0) {
        std::cerr << "error rc: << " -err << strerror(-err) << std::endl;
    }

    // Wait to display the char
    getchar();

    std::cout << "Send Keypress for KEY_A on: " << libevdev_get_name(dev) << std::endl;
    libevdev_uinput_write_event(uidev, EV_KEY, KEY_A, 1);
    libevdev_uinput_write_event(uidev, EV_SYN, SYN_REPORT, 0);

    std::cout << "Send Release for KEY_A on: " << libevdev_get_name(dev) << std::endl;
    libevdev_uinput_write_event(uidev, EV_KEY, KEY_A, 0);
    libevdev_uinput_write_event(uidev, EV_SYN, SYN_REPORT, 0);

    std::cout << "Finished sending keys: " << libevdev_get_name(dev) << std::endl;

    libevdev_uinput_destroy(uidev);
}

int main() {
    std::cout << "Start test" << std::endl;
    test_input();
    std::cout << "Finished test" << std::endl;

    return 0;
}
```

Compile with: `g++ -levdev test-input.cpp  -I/usr/include/libevdev-1.0 -o test-input`.

The create input device also has no group assigned:
```
udevadm info --name /dev/input/event18
P: /devices/virtual/input/input40/event18
N: input/event18
L: 0
E: DEVPATH=/devices/virtual/input/input40/event18
E: DEVNAME=/dev/input/event18
E: MAJOR=13
E: MINOR=82
E: SUBSYSTEM=input
E: USEC_INITIALIZED=278891039
E: ID_INPUT=1
E: ID_INPUT_KEY=1
E: TAGS=power-switch
E: CURRENT_TAGS=power-switch
```

So maybe it is related how the window manager interacts with `udev`, but yes the tag in my case is also only `power-switch`.

kris7t, 2020-12-14:
@aragon999 Indeed, this test program also works for me. So it's not a metadata problem after all!

closed 2020-12-21 by PixlOne

whot, 2021-01-07:
@kris7t kernel evdev itself has 16 bit scancodes (up to `KEY_MAX`). Wayland supports 32-bit keycodes but X only supports 8-bit keycodes *and* it uses an offset of 8. So any evdev code above 247 cannot be used in X. As wayland has become more popular we've started adding some of the higher keycodes for multimedia keys to the default XKB maps. Those won't work in X but they will still work in Wayland.

For debugging, `libinput debug-events --show-keycodes` will also print out the keycodes (don't type passwords while it is running). That should help verify that the key codes being sent are correct. From my attempts at reproducing this with https://github.com/PixlOne/logiops/issues/166#issuecomment-744592017 in an Arch VM with systemd 247, everything at the kernel/libinput level appears correct.

jasht1, 2023-10-28:
I'm not sure if this is the same issue or if I should start a new issue thread.
Similar to the issue described here;
- none of my key press bindings work, 
- the cursor stops moving whenever a gesture button is held as expected but no action is taken. 
- non gesture buttons do nothing either. 
- The keystrokes are appearing as expected in `libinput debug-events --show-keycodes` but have no effect beyond that.
- I am using a wayland environment in POP os.
however in my case logid functions perfectly if I launch it manually in the terminal with `sudo logid`.
I'm still very new to Linux (only 4 days in) so I would greatly appreciate any insight on this.
## Configuration & further info
- Pop!_OS 22.04 LTS
- Wireless Mouse MX Master
- /etc/logid.cfg:
```logid.cfg
devices: (
{
    name: "Wireless Mouse MX Master";
    smartshift:
    {
        on: true;
        threshold: 30;
        torque: 50;
    };
    thumbwheel: 
    {
        divert: false;
        invert: false;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1000;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA","KEY_PAGEUP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTMETA","KEY_PAGEDOWN"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL","KEY_LEFTMETA","KEY_PAGEUP"];
                        };
                    },
[... 117 lines cut ...]
   Main PID: 878 (logid)
      Tasks: 7 (limit: 18212)
     Memory: 1.7M
        CPU: 968ms
     CGroup: /system.slice/logid.service
             └─878 /usr/bin/logid

Oct 28 03:00:12 pop-os systemd[1]: Started Logitech Configuration Daemon.
Oct 28 03:00:12 pop-os logid[878]: [WARN] Error adding device /dev/hidraw3: open failed: No such file or directory
Oct 28 03:00:12 pop-os logid[878]: [WARN] Error adding device /dev/hidraw4: open failed: No such file or directory
~
~
~
~
~
~
~
~
~
~
lines 1-13/13 (END)
```
~about 40 times over

so `sudo systemctl enable --now logid` seems to have worked in launching logid on boot. I'm not sure if the `[WARN]`s about hidraw3 & hidraw4 are anything to worry about as my mouse is on hidraw2:255.

commit 911e91ee (2020-12-21): Enable fewer EV_KEY events in InputDevice

Fixes #166

Starting with systemd 247 (?), desktop environments fail to recognize
the virtual input device if it has too many enabled libinput events.

It seems like KEY_ROTATE_LOCK_TOGGLE as the highest enabled event is a
good limit. This is the highest evdev event mapped by xkb, so the
usefulness higher events is limited, anyways.
PR #182 (2020-12-21): Fix virtual keyboard
Fixes #166

Starting with systemd 247 (?), desktop environments fail to recognize the virtual input device if it has too many enabled libinput events.

It seems like `KEY_ROTATE_LOCK_TOGGLE` as the highest enabled event is a good limit. This is the highest evdev event mapped by xkb, so the usefulness higher events is limited, anyways.

Kudos to @aragon999 for helping debuggin the issue, especially for the MWE in https://github.com/PixlOne/logiops/issues/166#issuecomment-744592017!

Looks like I was mistaken in https://github.com/PixlOne/logiops/issues/166#issuecomment-7445 [... cut]