PixlOne/logiops #172: Sticky key after gesture

vjuge, 2020-12-02:
Hi,

There is a problem of some key being kept pressed after gesture, this is annoying with some application because it results to unexpected shortcuts. 
Test scenario :
screen one > press gesture button + direction up/down > release gesture button > focus on Chrome > scroll page with wheel > **the page zoom in / out**
--> it seems the "ctrl" key is still pressed after gesture
If I press/release the Ctrl key before the scroll, the problem doesn't happen (seems to "reinitialize" the status ?)

My setup is: 
* keyboard Logitech MX Keys
* keyboard layout as 'French (Macintosh)
* swapped Left Alt with Left Ctrl (with gnome-tweaks > Keyboard and Mouse > Additional Layout Options > Ctrl position)  
* `logid.cfg` :
```
devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: true;
        threshold: 13;
    };
    hiresscroll:
    {
        hires: false;
        invert: false;
        target: false;
    };
    dpi: 1200;

    buttons: (
        {
            cid: 0xc3;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Up";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_UP"];
                        };
                    },
                    {
                        direction: "Down";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN"];
                        };
                    },
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action = 
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_LEFT"];
                        };
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress"
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_RIGHT"];
                        };
                    },
                    {
                        direction: "None"
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_FN", "KEY_F1"];
                        }
                    }
                );
            };
        },
        {
            cid: 0xc4;
            action =
            {
                type = "ToggleSmartshift";
            };
        }
    );
}
);
```

jixbo, 2022-01-14:
I am having this exact issue, it only happens with my gesture to changes desktops, so at the beginning I didn't think the issue was related to logiops (only gestures with CTRL key).
Those are the gestures causing the issue for me:
```                    
                   {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_RIGHT"];
                        };
                   },
                   {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_LEFTMETA", "KEY_LEFT"];
                        };
    
               }
```
Did you manage to fix it @vjuge?
Any idea what could it be @PixlOne?

shafinhasnat, 2024-11-17:
To achieve this gesture, I had to change keyboard shortcuts manually as following - 
```
Move to workspace on the right ---> Super+Right
Move to workspace on the left ---> Super+left
```

And my config  -
```
devices: ({
  name: "MX Master 3S";

  smartshift: {
    on: true;
    threshold: 100;
  };

  hiresscroll: {
    hires: true;
    invert: false;
    target: false;
  };

  dpi: 1500; // max=4000

  buttons: (
    // Forward button
    {
      cid: 0x56;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_FORWARD" ];
            }
          },

          {
            direction: "Up";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_PLAYPAUSE" ];
            }
          },

          {
            direction: "Down";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA" ];
            }
          },

          {
            direction: "Right";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_NEXTSONG" ];
            }
          },

          {
            direction: "Left";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_PREVIOUSSONG" ];
            }
          }
        );
      };
    },

    // Back button
    {
      cid: 0x53;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_BACK" ];
            }
          }
        );
      };
    },

    // Gesture button (hold and move)
    {
      cid: 0xc3;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA" ];
            }
          },

         {
           direction: "Right";
           mode: "OnRelease";
           action = {
             type: "Keypress";
             keys: [ "KEY_LEFTMETA", "KEY_LEFT"];
           }
         },

         {
           direction: "Left";
           mode: "OnRelease";
           action = {
             type: "Keypress";
             keys: [ "KEY_LEFTMETA", "KEY_RIGHT" ];
           }
		  },

		  {
            direction: "Up";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA", "KEY_UP" ];
            }
		  },
		  
		  {
            direction: "Down";
            mode: "OnRelease";
            action = {
              type: "Keypress";
              keys: [ "KEY_LEFTMETA", "KEY_DOWN" ];
            }
          }
        );
      };
    },
	
    // Top button
    {
      cid: 0xc4;
      action = {
        type: "Gestures";
        gestures: (
          {
            direction: "None";
            mode: "OnRelease";
            action = {
              type: "ToggleSmartShift";
            }
          },

          {
            direction: "Up";
            mode: "OnRelease";
            action = {
              type: "ChangeDPI";
              inc: 1000,
            }
          },

          {
            direction: "Down";
            mode: "OnRelease";
            action = {
              type: "ChangeDPI";
              inc: -1000,
            }
          }
        );
      };
    }
  );
});
```