PixlOne/logiops #174: MX Anywhere 3: SmartShift feature not found, cannot use ToggleSmartShift action

maaattes, 2020-12-05:
Hi,
first of all I would like to thank you for this driver. I really really appreciate your efforts.

#### Issue:
```
logid -v
[INFO] Device found: MX Anywhere 3 on /dev/hidraw3:255
[WARN] /dev/hidraw3:255: SmartShift feature not found, cannot use ToggleSmartShift action.
```
Somehow I can't bind ToggleSmartShift to a button. Using the default config, the button works as expected.

#### My Setup
`/etc/logid.cfg`  
```
devices: (
{
    name: "MX Anywhere 3";
    smartshift:
    {
        on: true;
        threshold: 127;
        default_treshold: 127;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };
    dpi: 1200;

    buttons: (
        {
            cid: 0xc4;
            action =
            {
                type: "Gestures";
                gestures: (
                    {
                        direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_LEFTCTRL", "KEY_PAGEUP"];
                        }
                    },
                    {
                        direction: "Right";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
                            keys: ["KEY_RIGHTCTRL", "KEY_PAGEDOWN"];
                        }
                    },
                    {
                        direction: "None";
                        mode: "OnRelease";
                        action =
                        {
                            type: "ToggleSmartshift";
                        }
                    },
                );
            };
        }
    );
}
);
```

```
logid -V
v0.2
```
(In reality, it's v0.2.2, or am I wrong here? `sudo dnf info logiops`  gives `Version      : 0.2.2`)

```
uname -srv
Linux 5.9.11-200.fc33.x86_64 #1 SMP Tue Nov 24 18:18:01 UTC 2020
```
(Fedora 33 using Wayland, MX Anywhere 3 connected via Bluetooth)

vjuge, 2021-01-01:
Hi,
Question for the community: 
can you describe how you manage to build for fedora ? What are the packages to install ?
thanks

Staindk, 2021-04-12:
Same issue here with an MX Anywhere 3 - did you ever find some solution @maaattes?

I also seem to not be able to map any keys to any of the buttons? I can successfully map e.g. DPI switching, but trying to map a key to one of the mouse buttons just does nothing (i.e. the button does nothing, not even the default action).

I'm on Gnome / Wayland / Arch, also connected via Bluetooth.

maaattes, 2021-04-13:
> Same issue here with an MX Anywhere 3 - did you ever find some solution @maaattes?
No, in the end, I've sold the mouse because of it's MagSpeed scroll wheel. It is very difficult to make a middle click without scrolling some lines at the same time. Even when the threshold is set to the highest level. As a user of CAD and GIS software this mouse just does not work.

Staindk, 2021-04-13:
Ah okay, thanks.

I'm still getting used to the scroll wheel. I think the fact that I don't need the `hiresscroll` function helps - with it enabled the wheel does get way too sensitive.

channeladam, 2021-05-09:
Same issue

leios76, 2021-07-21:
[patch.tar.gz](https://github.com/PixlOne/logiops/files/6857563/patch.tar.gz)

With this patch smartshift will be work on mx anywhere3.

thhart, 2021-08-06:
The above patch is working well. Here is how to apply:
1. Download and extract the file into the source folder
2. Enter following command:
patch -p1 < 0001-support-smartshift-on-mx-anywhere3.patch
3. Follow the standard make install instructions

gabrielbdsantos, 2021-08-07:
I'm having the same issue here on my MX Vertical. I have already tried the patch shared by @leios76, but it just didn't work for me.

It seems the patch sets a new definition for the smart shift (SMART_SHIFT_V2 = 0x2111). Perhaps on the MX Vertical the feature index could be something different? I tried to list all feature using the [hidpp](https://github.com/PixlOne/hidpp) library.

`hidpp-list-features` outputs:

```
Logitech MX Vertical Advanced Ergonomic Mouse (046d:c08a) is a HID++ 4.5 device
Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x0020] Reset
Feature 0x05: [0x0021] Crypto Identifier
Feature 0x06: [0x0007] Device Friendly Name
Feature 0x07: [0x1000] Battery status
Feature 0x08: [0x1002] ?
Feature 0x09: [0x1b04] Reprog controls v4
Feature 0x0a: [0x1814] Change host
Feature 0x0b: [0x1815] Hosts info
Feature 0x0c: [0x2250] ?
Feature 0x0d: [0x2100] Vertical scrolling
Feature 0x0e: [0x2130] Low-res wheel
Feature 0x0f: [0x2201] Adjustable dpi
Feature 0x10: [0x00c2] DFUcontrol 3
Feature 0x11: [0x1803] ? (hidden, internal)
Feature 0x12: [0x1806] Configurable device properties (hidden, internal)
Feature 0x13: [0x1813] ? (hidden, internal)
Feature 0x14: [0x1805] OOBState (hidden, internal)
Feature 0x15: [0x1830] ? (hidden, internal)
Feature 0x16: [0x1890] ? (hidden, internal)
Feature 0x17: [0x1891] ? (hidden, internal)
Feature 0x18: [0x18a1] ? (hidden, internal)
Feature 0x19: [0x1e00] Enable hidden features (hidden)
Feature 0x1a: [0x1eb0] ? (hidden, internal)
Feature 0x1b: [0x1861] ? (hidden, internal)
Feature 0x1c: [0x18c0] ? (hidden, internal)
```
Well, smart shift isn't listed at all. Does that mean that the MX Vertical does not support smart shift?

leios76, 2021-08-08:
> I'm having the same issue here on my MX Vertical. I have already tried the patch shared by @leios76, but it just didn't work for me.
>
> It seems the patch sets a new definition for the smart shift [...]

It seems that MX Vertical doesn't have magnetic wheel. To use smart shift feature, it must have 2110 or 2111 feature.

output with My MX Anywhere3:

MX Anywhere 3 (046d:4090) is a HID++ 4.5 device
Feature 0x01: [0x0001] Feature set
Feature 0x02: [0x0003] Device FW version
Feature 0x03: [0x0005] Device name
Feature 0x04: [0x1d4b] Wireless device status
Feature 0x05: [0x0020] Reset
Feature 0x06: [0x0021] Crypto Identifier
Feature 0x07: [0x0007] Device Friendly Name
Feature 0x08: [0x1004] ?
Feature 0x09: [0x1b04] Reprog controls v4
Feature 0x0a: [0x1814] Change host
Feature 0x0b: [0x2250] ?
Feature 0x0c: [0x2201] Adjustable dpi
**Feature 0x0d: [0x2111] ?**
Feature 0x0e: [0x2121] Hi-res wheel
Feature 0x0f: [0x2251] ?
Feature 0x10: [0x00c2] DFUcontrol 3
Feature 0x11: [0x1802] Device reset (hidden, internal)
Feature 0x12: [0x1803] ? (hidden, internal)
Feature 0x13: [0x1806] Configurable device properties (hidden, internal)
Feature 0x14: [0x1812] ? (hidden, internal)
Feature 0x15: [0x1805] OOBState (hidden, internal)
Feature 0x16: [0x1830] ? (hidden, internal)
Feature 0x17: [0x1890] ? (hidden, internal)
Feature 0x18: [0x1891] ? (hidden, internal)
Feature 0x19: [0x18a1] ? (hidden, internal)
Feature 0x1a: [0x1e00] Enable hidden features (hidden)
Feature 0x1b: [0x1eb0] ? (hidden, internal)
Feature 0x1c: [0x1861] ? (hidden, internal)
Feature 0x1d: [0x9001] ? (hidden, internal)
Feature 0x1e: [0x9205] ? (hidden, internal)
Feature 0x1f: [0x9300] ? (hidden, internal)

gabrielbdsantos, 2021-08-08:
> It seems that MX Vertical doesn't have magnetic wheel. To use smart shift feature, it must have 2110 or 2111 feature.

Thanks.

closed 2023-05-08 by PixlOne