PixlOne/logiops #175: Build failed on Ubutun 20.04.1 LTS

PierrickG34, 2020-12-06:
Hi all,
I'm trying to build the application but I have this problem when I launch `make` :
```
[...]/logiops/build$ make  
Scanning dependencies of target logid
[  1%] Building CXX object src/logid/CMakeFiles/logid.dir/logid.cpp.o
[  3%] Building CXX object src/logid/CMakeFiles/logid.dir/util/log.cpp.o
[  5%] Building CXX object src/logid/CMakeFiles/logid.dir/InputDevice.cpp.o
[  6%] Building CXX object src/logid/CMakeFiles/logid.dir/DeviceManager.cpp.o
[  8%] Building CXX object src/logid/CMakeFiles/logid.dir/Device.cpp.o
[ 10%] Building CXX object src/logid/CMakeFiles/logid.dir/Receiver.cpp.o
[ 11%] Building CXX object src/logid/CMakeFiles/logid.dir/Configuration.cpp.o
[ 13%] Building CXX object src/logid/CMakeFiles/logid.dir/features/DPI.cpp.o
[ 15%] Building CXX object src/logid/CMakeFiles/logid.dir/features/SmartShift.cpp.o
[ 16%] Building CXX object src/logid/CMakeFiles/logid.dir/features/HiresScroll.cpp.o
[ 18%] Building CXX object src/logid/CMakeFiles/logid.dir/features/RemapButton.cpp.o
[ 20%] Building CXX object src/logid/CMakeFiles/logid.dir/features/DeviceStatus.cpp.o
[ 21%] Building CXX object src/logid/CMakeFiles/logid.dir/features/ThumbWheel.cpp.o
[ 23%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/Action.cpp.o
[ 25%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/NullAction.cpp.o
[ 26%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/KeypressAction.cpp.o
[ 28%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/ToggleHiresScroll.cpp.o
[ 30%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/ToggleSmartShift.cpp.o
[ 31%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/CycleDPI.cpp.o
[ 33%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/ChangeDPI.cpp.o
[ 35%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/GestureAction.cpp.o
[ 36%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/ChangeHostAction.cpp.o
[ 38%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/Gesture.cpp.o
[ 40%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/ReleaseGesture.cpp.o
[ 41%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/ThresholdGesture.cpp.o
[ 43%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/IntervalGesture.cpp.o
[ 45%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/AxisGesture.cpp.o
[ 46%] Building CXX object src/logid/CMakeFiles/logid.dir/actions/gesture/NullGesture.cpp.o
[ 48%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/Error.cpp.o
[ 50%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/raw/DeviceMonitor.cpp.o
[ 51%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/raw/RawDevice.cpp.o
[ 53%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/dj/Receiver.cpp.o
[ 55%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/dj/ReceiverMonitor.cpp.o
[ 56%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/dj/Error.cpp.o
[ 58%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp/Device.cpp.o
[ 60%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp/Report.cpp.o
[ 61%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp10/Error.cpp.o
[ 63%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp10/Device.cpp.o
[ 65%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/Device.cpp.o
[ 66%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/Error.cpp.o
[ 68%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/Feature.cpp.o
[ 70%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/EssentialFeature.cpp.o
[ 71%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Root.cpp.o
[ 73%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/FeatureSet.cpp.o
[ 75%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/DeviceName.cpp.o
[ 76%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/Reset.cpp.o
[ 78%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/AdjustableDPI.cpp.o
[ 80%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/SmartShift.cpp.o
[ 81%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ReprogControls.cpp.o
[ 83%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/HiresScroll.cpp.o
[ 85%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ChangeHost.cpp.o
[ 86%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/WirelessDeviceStatus.cpp.o
[ 88%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/hidpp20/features/ThumbWheel.cpp.o
[ 90%] Building CXX object src/logid/CMakeFiles/logid.dir/backend/dj/Report.cpp.o
[ 91%] Building CXX object src/logid/CMakeFiles/logid.dir/util/workqueue.cpp.o
[ 93%] Building CXX object src/logid/CMakeFiles/logid.dir/util/worker_thread.cpp.o
[ 95%] Building CXX object src/logid/CMakeFiles/logid.dir/util/task.cpp.o
[ 96%] Building CXX object src/logid/CMakeFiles/logid.dir/util/thread.cpp.o
[ 98%] Building CXX object src/logid/CMakeFiles/logid.dir/util/ExceptionHandler.cpp.o
[100%] Linking CXX executable ../../logid
/home/linuxbrew/.linuxbrew/bin/ld : CMakeFiles/logid.dir/actions/Action.cpp.o : dans la fonction « logid::actions::Action::makeAction(logid::Device*, libconfig::Setting&) » :
Action.cpp:(.text+0x1ae) : référence indéfinie vers « libconfig::Setting::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >[abi:cxx11]() const »
/home/linuxbrew/.linuxbrew/bin/ld : CMakeFiles/logid.dir/actions/KeypressAction.cpp.o : dans la fonction « logid::actions::KeypressAction::Config::Config(logid::Device*, libconfig::Setting&) » :
KeypressAction.cpp:(.text+0x328) : référence indéfinie vers « libconfig::Setting::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >[abi:cxx11]() const »
/home/linuxbrew/.linuxbrew/bin/ld : CMakeFiles/logid.dir/actions/GestureAction.cpp.o : dans la fonction « logid::actions::GestureAction::Config::Config(logid::Device*, libconfig::Setting&) » :
GestureAction.cpp:(.text+0x1243) : référence indéfinie vers « libconfig::Setting::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >[abi:cxx11]() const »
/home/linuxbrew/.linuxbrew/bin/ld : GestureAction.cpp:(.text+0x139d) : référence indéfinie vers « libconfig::Setting::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >[abi:cxx11]() const »
/home/linuxbrew/.linuxbrew/bin/ld : CMakeFiles/logid.dir/actions/ChangeHostAction.cpp.o : dans la fonction « logid::actions::ChangeHostAction::Config::Config(logid::Device*, libconfig::Setting&) » :
ChangeHostAction.cpp:(.text+0x467) : référence indéfinie vers « libconfig::Setting::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >[abi:cxx11]() const »
/home/linuxbrew/.linuxbrew/bin/ld : CMakeFiles/logid.dir/actions/gesture/Gesture.cpp.o:Gesture.cpp:(.text+0x632) : encore plus de références indéfinies suivent vers « libconfig::Setting::operator std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >[abi:cxx11]() const »
collect2: error: ld returned 1 exit status
make[2]: *** [src/logid/CMakeFiles/logid.dir/build.make:955 : logid] Erreur 1
make[1]: *** [CMakeFiles/Makefile2:94 : src/logid/CMakeFiles/logid.dir/all] Erreur 2
make: *** [Makefile:130 : all] Erreur 2
```
 
Any suggestions on how to make this step work?

Thank you :)

rootwork, 2021-01-25:
I'm guessing this is addressed in #151 -- you need to run `sudo apt install build-essential` because it was left out of the original instructions for Ubuntu. I have installed this successfully on Linux Mint 20 (based on Ubuntu 20); I happened to have already had `build-essential` installed, so ping me if you need help.

ZegertBoele, 2021-02-26:
@rootwork This did it for me on a Debian 10. I think it should be in the readme.

PierrickG34, 2021-03-02:
Hi @rootwork, thanks for this command but it's still doesn't work :(. Same problem as in the beginning for me

rootwork, 2021-03-07:
@PierrickG34 What OS are you on?

emilcarr, 2021-03-07:
Looks like you're missing libconfig. Try `sudo apt install libconfig-dev`

PierrickG34, 2021-03-08:
@rootwork 
```
$ lsb_release -a

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
```

@emilcarr Nope, I've already install this one before. I try with this : ```sudo apt install cmake libevdev-dev libudev-dev libconfig++-dev build-essential``` but it's the same error.

kbialek, 2021-05-02:
I came across this same problem and solved it by building the binary in a docker container. Here is a gist showing how to do it https://gist.github.com/kbialek/625bfe7c3a285529989374b9febd5ac8