PixlOne/logiops #177: cid: 0xc3 remap ignored

simokron, 2020-12-08:
Hi,

I'm having a rather peculiar issue. I have two nigh identical Arch Linux installs - one on the work laptop and one on my personal PC. I have the exact same `logid.cfg` file, but on the laptop my remap of the thumb button is ignored. I've had this setup for months without issues, but I did change a lot of hardware recently, so I can't say for sure when it started occurring.

Here is my config file:
```
devices: ({
  name: "Wireless Mouse MX Master 3";

  dpi: 1000;

  buttons: (

    // Make thumb button 10.
    { cid: 0xc3; action = { type: "Keypress"; keys: ["KEY_PLAYPAUSE"]; }; },

  );
});
```

For debugging, I changed the keypress to e.g. `KEY_O`, and on the stationary PC that will happily enter "o" into whatever text field I have focused, but on the laptop it does nothing. I can change the DPI on both computers though, so it's definitely recognising the device and such.

I've tried with the same USB receiver connected to the USB dock in my monitor, and with unique receivers but the result is the same, i.e., it works on one computer but not the other.

Here is the verbose output on the working computer:
```
[~]$ sudo logid -v                                                                                                                 
[DEBUG] Unsupported device /dev/hidraw1 ignored
[DEBUG] Unsupported device /dev/hidraw0 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[DEBUG] Unsupported device /dev/hidraw5 ignored
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
[DEBUG] Unsupported device /dev/hidraw8 ignored
[DEBUG] Unsupported device /dev/hidraw9 ignored
[WARN] Error adding device /dev/hidraw3: std::exception
[INFO] Detected receiver at /dev/hidraw2
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw2:1
[DEBUG] /dev/hidraw2:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
```
and on the busted laptop with the same receiver:
```
[~]$ sudo logid -v                                                                                                   (130) 
[INFO] Detected receiver at /dev/hidraw0
[INFO] Device found: Wireless Mouse MX Master 3 on /dev/hidraw0:1
[DEBUG] /dev/hidraw0:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0xc3 | YES     |         | YES        | YES
[DEBUG] 0xc4 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[DEBUG] Thumb wheel detected (0x2150), capabilities:
[DEBUG] timestamp | touch | proximity | single tap
[DEBUG] YES       | YES   | YES       | YES       
[DEBUG] Thumb wheel resolution: native (18), diverted (120)
[DEBUG] Unsupported device /dev/hidraw2 ignored
[DEBUG] Unsupported device /dev/hidraw3 ignored
[DEBUG] Unsupported device /dev/hidraw4 ignored
[WARN] [DEBUG] Unsupported device /dev/hidraw5 ignored
Error adding device /dev/hidraw1: std::exception
[DEBUG] Unsupported device /dev/hidraw6 ignored
[DEBUG] Unsupported device /dev/hidraw7 ignored
```

I'm at a complete loss as to what can cause this.. do you have any recommendations for a deeper investigation?

Thanks,

Necoro, 2020-12-12:
I'd like to add a "me too" to this issue. For me it's also the very same laptop, where the button remapping stopped working. (I only use it when sitting at the desk, which is not so often -- so I cannot pinpoint an actual change).

What has happened in the meantime were multiple kernel upgrades.

@simokron: Are the kernel versions used on your systems also identical?

kbtz, 2020-12-13:
Here the output of `sudo libinput debug-events` shows that the proper key is being fired from the virtual input but it seems that this event source is not being used by my WM. Anyone using wayland?

simokron, 2020-12-14:
> I'd like to add a "me too" to this issue. For me it's also the very same laptop, where the button remapping stopped working. (I only use it when sitting at the desk, which is not so often -- so I cannot pinpoint an actual change).
> 
> What has happened in the meantime were multiple kernel upgrades.
> 
> @simokron: Are the kernel versions used on your systems also identical?
> 
> logid output:
> 
> ```
> # logid -v -c /etc/logid.cfg
> [DEBUG] Unsupported device /dev/hidraw9 ignored
> [DEBUG] Unsupported device /dev/hidraw0 ignored
> [DEBUG] Unsupported device /dev/hidraw2 ignored
> [DEBUG] Unsupported device /dev/hidraw3 ignored
> [DEBUG] Unsupported device /dev/hidraw1 ignored
> [DEBUG] Unsupported device /dev/hidraw4 ignored
> [INFO] Device found: MX Vertical Advanced Ergonomic Mouse on /dev/hidraw5:255
> [DEBUG] /dev/hidraw5:255 remappable buttons:
> [DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
> [DEBUG] 0x50 |         |         | YES        | 
> [DEBUG] 0x51 |         |         | YES        | 
> [DEBUG] 0x52 | YES     |         | YES        | YES
> [DEBUG] 0x53 | YES     |         | YES        | YES
> [DEBUG] 0x56 | YES     |         | YES        | YES
> [DEBUG] 0xd7 | YES     |         |            | YES
> [DEBUG] 0xfd | YES     |         | YES        | YES
> ```
> 
> My logid.cfg:
> 
> ```
> devices: (
>     {
>         name: "MX Vertical Advanced Ergonomic Mouse"
>         buttons: (
>             {
>                 cid: 0xfd
>                 action = {
>                     type: "Keypress"
>                     # maps to logical button XF86AudioPlay
>                     keys: ["KEY_PLAYPAUSE"]
>                 }
>             }
>         )
>     }
> )
> ```

Thanks for the input! I am indeed using the exact same kernel versions (now 5.9.14.arch1-1).

simokron, 2020-12-14:
> Here the output of `sudo libinput debug-events` shows that the proper key is being fired from the virtual input but it seems that this event source is not being used by my WM. Anyone using wayland?

Thanks for the information! Indeed, the correct key is being issued over here but ignored further down the line. I'm using fresh bspwm and sxhkd builds from their respective GitHub pages (so X still).

p1otr, 2020-12-14:
I'm using Sway and I have exactly the same symptoms (libinput sees the right keys but they vanish before my WM can see them). Note that this happened for me few weeks ago probably after some other part of my system was updated. I didn't update logid until yesterday.

Additional info: I  had to update (perfectly working fine before) name from "MX Master 2S" to "Wireless Mouse MX Master 2S". I have cid: 0xc4 mapped to BTN_MIDDLE and even though this button doesn't change scroll mode anymore, it doesn't work as middle button as well (other than in libinput)

aragon999, 2020-12-14:
Maybe this is also a duplicate of #166 

Does the remap work when you use an action like: `type = "ToggleSmartshift";`?

@simokron Maybe you have one the one system Wayland enabled, and on the other X11? Maybe this is also due to the graphics vendor or so.

simokron, 2020-12-14:
> Maybe this is also a duplicate of #166
> 
> Does the remap work when you use an action like: `type = "ToggleSmartshift";`?
> 
> @simokron Maybe you have one the one system Wayland enabled, and on the other X11? Maybe this is also due to the graphics vendor or so.

You're most likely right that it's a duplicate. Sorry!

`type = "ToggleSmartshift";` works and actually does toggle smartshift.

As for the Wayland connection, I'm not sure to be honest. I do have some Wayland packages installed (on both machines), but they are requirements from Mesa ans such. Though as far as I'm aware, bspwm has no support for Wayland at the moment I use the same dotfiles, so it shouldn't be that. I can with confidence at least say that if that is the case, it is not intentional on my part.

True that it could be graphics vendor though. They both use nVidia, but the laptop is a workstation with a Quadro card, and the desktop has a 3070. The CPUs are also different (Intel on the laptop and AMD Ryzen on the desktop).

p1otr, 2020-12-14:
> Does the remap work when you use an action like: `type = "ToggleSmartshift";`?
yes it does

aragon999, 2020-12-14:
@simokron what does 
```
echo $XDG_SESSION_TYPE
```
say on the two different machines?

Unfortunately I do not know `bspwm` but I guess this should be properly set by the windows manager.

What systemd version are you running? Are you using the same version on both systems?

simokron, 2020-12-14:
Inspired by the connection to #166, I downgraded to systemd to 246.6, which instantly addresses the issue. So for sure systemd 247 is to blame on my machine. Now to figure out why the heck it only happens on one of them and if it can be fixed.

simokron, 2020-12-14:
> @simokron what does
> 
> ```
> echo $XDG_SESSION_TYPE
> ```
> 
> say on the two different machines?
> 
> Unfortunately I do not know `bspwm` but I guess this should be properly set by the windows manager.
> 
> What systemd version are you running? Are you using the same version on both systems?

Laptop & desktop:
```[~]$ echo $XDG_SESSION_TYPE                                                                                                
tty```

They were both running systemd 247.1-3.

simokron, 2020-12-14:
I'm closing this as it is almost certainly a duplicate of #166. Please follow that issue for updates, and consider downgrading systemd for the time being.

closed 2020-12-14 by simokron