PixlOne/logiops #181: Keypress does not seem to work

aragon999, 2020-12-14:
I am using logid with sway in Archlinux, but I seem to be not able to get the Keypress to work, either in the gestures or for other buttons, for example the following config the commented part works (toggles smartshift) while the `Keypress` action type does not work:
```
        {
            cid: 0xc4;
            action =
            {
                type: "Keypress";
                keys: ["KEY_A"];
            }
#            action =
#            {
#                type: "ToggleSmartshift";
#            };
        }
```

Is there anything I can help with debugging? Or can you point me to some code where I can start debugging?

kris7t, 2020-12-14:
Probably a duplicate of https://github.com/PixlOne/logiops/issues/166

aragon999, 2020-12-14:
Yes that seems to be the case. Sorry I did not see this issue before. I will close this issue as a duplicate.

closed 2020-12-14 by aragon999