PixlOne/logiops #186: MX Master 3 on Ubuntu 20.10 main wheel scrolling is really slow

dszmaj, 2020-12-26:
Hey,

first of all I'd like to thank you for this marvel of an open source software, I'm able to use all features of my mouse :)

But the problem is that when I have hires disabled, scrolling through pages in chrome or firefox is very slow, so much so that it barely moves the page. When I enable hires it improves a lot, but only so it is acceptable, I'd like it to work faster to match windows experience.

Is there any way to configure it?

Here is my current config with the current master build:
```

devices: ({
  name: "Wireless Mouse MX Master 3";

  smartshift: { on: true; threshold: 10; };

  hiresscroll: {
    hires: true;
    invert: false;
    target: false;
  };

  thumbwheel: {
    divert: true;
    invert: false;
    left: {
      mode: "OnInterval";
      pixels: 1;
      action: {
        type: "Keypress";
        keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_UP"];
      };
    };
    right: {
      mode: "OnInterval";
      pixels: 1;
      action: {
        type: "Keypress";
        keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_DOWN"];
      };
    };
  };

  // 4000 max for MX Master 3.
  dpi: 1800;

  buttons: (

    // Thumb button.
    { cid: 0xc3; action = { type: "Keypress"; keys: ["KEY_LEFTCTRL", "KEY_W"]; }; },
    // Top button.
    { cid: 0xc4; action = { type: "ToggleSmartshift"; }; },
    // Forward button.
    { cid: 0x54; action = { type: "Keypress"; keys: ["BTN_BACK"]; }; },
    // Back button.
    { cid: 0x57; action = { type: "Keypress"; keys: ["BTN_FORWARD"]; }; }

  );
});
```

dial25sd, 2021-01-02:
Had a similar problem with Firefox on Manjaro and solved it by manually setting an `axis_multiplier`.

See my config below:
`    hiresscroll: {
        hires: true;
        invert: false;
        target: true;
        up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 3;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -3;
        },
    };`

thaidh, 2021-01-12:
Same problem on Ubuntu 20.04 after upgrade kernel 5.04 --> 5.08. Fix pls !!!

marcom48, 2021-02-08:
Hey there,

Solved the same issue on Ubuntu 20.04! It ended up setting the hires field to true in the hiresscroll setting.
Further I restart the logid.service on boot `sudo systemctl restart logid` (try `logid.service` in place of `logid` if that doesn't work). See [https://askubuntu.com/questions/889632/startup-script-with-sudo-in-ubuntu-16-10](https://askubuntu.com/questions/889632/startup-script-with-sudo-in-ubuntu-16-10) for help with running the command on boot.

Below is my config, I hope this helps!

```devices: (
{
    name: "Wireless Mouse MX Master 3";
    smartshift:
    {
        on: false;
        threshold: 100;
    };
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
    };    

    dpi: 2000;
    buttons: (
            {
                cid: 0xc3;
                action =
                {
                    type: "Gestures";
                    gestures: (
                        {
                            direction: "Left";
                            mode: "OnRelease";
                            action =
                            {
                                type: "Keypress";
                                keys: ["KEY_F1"];
                            };
                        }
                    );
                };
            },
        {
                cid: 0x53;
                action =
                {
                    type: "Keypress";
                    keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_T"];
                };
            },
        {
                cid: 0x54;
                action =
                {
                    type: "Keypress";
                    keys: ["KEY_LEFTCTRL", "KEY_LEFTALT", "KEY_T"];
                };
            },
        {
                cid: 0x56;
                action =
                {
                    type: "Keypress";
                    keys: ["KEY_COPY"];
                };
            },
            {
                cid: 0x57;
                action =
                {
                    type: "Keypress";
                    keys: ["KEY_COPY"];
                };
            }

    );
        
    }
);

```

bflorat, 2022-10-29:
Same here (MX3 under Ubuntu 22.04). Very slow scrolling from times to times. I fix this by turning the mouse off and on again (using the physical power button on the lower side).

[image]

reeseovine, 2023-05-08:
> Had a similar problem with Firefox on Manjaro and solved it by manually setting an `axis_multiplier` (and also tweaking a bit with FF's internal scrolling config).
> 
> See my config below:
> 
> ```
>    hiresscroll: {
>         hires: true;
>         invert: false;
>         target: true;
>         up: {
>             mode: "Axis";
>             axis: "REL_WHEEL_HI_RES";
>             axis_multiplier: 3;
>         },
>         down: {
>             mode: "Axis";
>             axis: "REL_WHEEL_HI_RES";
>             axis_multiplier: -3;
>         },
>     };
> ```

this broke after i just updated today. not sure what version i was on before.

aimfeld, 2024-04-06:
Same problem, restarting with `sudo systemctl restart logid` and turning off/on mouse fixes it. I have to do this after every reboot which is annoying.
I have this in my logid.cfg:

    // A lower threshold number makes the wheel switch to free-spin mode
    // quicker when scrolling fast. A high threshold like 1000 makes wheel never free-spin.
    // smartshift: { on: true; threshold: 20; };
    smartshift: { on: true; threshold: 1000; };

    // Set scroll wheel speed
    // https://github.com/PixlOne/logiops/issues/116#issuecomment-680307475
    hiresscroll:
    {
        hires: true;
        invert: false;
        target: false;
        up: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: 96;
        },
        down: {
            mode: "Axis";
            axis: "REL_WHEEL_HI_RES";
            axis_multiplier: -96;
        },
    };

diasdev, 2024-05-29:
> Same problem, restarting with `sudo systemctl restart logid` and turning off/on mouse fixes it. I have to do this after every reboot which is annoying. I have this in my logid.cfg:
> 
> ```
> // A lower threshold number makes the wheel switch to free-spin mode
> // quicker when scrolling fast. A high threshold like 1000 makes wheel never free-spin.
> // smartshift: { on: true; threshold: 20; };
> smartshift: { on: true; threshold: 1000; };
> 
> // Set scroll wheel speed
> // https://github.com/PixlOne/logiops/issues/116#issuecomment-680307475
> hiresscroll:
> {
>     hires: false;
>     invert: false;
>     target: false;
>     up: {
>         mode: "Axis";
>         axis: "REL_WHEEL";
>         axis_multiplier: 3;
>     },
>     down: {
>         mode: "Axis";
>         axis: "REL_WHEEL";
>         axis_multiplier: -3;
>     }
> };
> ```

Did you find a solution to this issue?

aimfeld, 2024-05-30:
> Did you find a solution to this issue?

@diasdev Unfortunately not, I have been too lazy.