PixlOne/logiops #187: Should ship with udev rules tagging created virtual input devices as uaccess

fastcat, 2020-12-29:
The virtual input device(s) created by logid are not, by default on Debian anyways, accessible to normal users. This results in synthesized events from e.g. gestures not working in a default setup.

One workaround is to add the user to the `input` group, though this grants that user more permissions than may be desired.

The robust solution AFAICT is to ship udev rules that tag the created virtual device(s) with `uaccess`, so that systemd/udevd will chmod the input devices on the fly based on the user logged into an active seat (display, etc.).

wormyrocks, 2021-01-04:
Do you have an example udev rule that accomplishes this, just for my own setup? Thanks!

fastcat, 2021-01-04:
I haven't worked out the device matching rules for logiops yet, but for the overall structure of what this would look like, Solaar has this: https://github.com/pwr-Solaar/Solaar/blob/master/rules.d/42-logitech-unify-permissions.rules or the UHK agent has another simpler one, which may be a better model for logiops: https://github.com/UltimateHackingKeyboard/agent/blob/master/rules/50-uhk60.rules

Looking at `udevadm info -a /dev/input/...` for the mouse & event devices logid creates, I think the matchers to use are:
`SUBSYSTEMS=="input" ATTRS{name}=="LogiOps Virtual Input"`

I think this should match both, but I haven't tested this yet.