PixlOne/logiops #188: "Fixed scroll wheel mode" for MX Master 3

GabAlpha, 2020-12-30:
Hi everyone,
comparing this software with logitech options I realized that the function to fix the mouse wheel is missing.

[image]

I don't know if you can add it as a configuration.
Thanks

PixlOne, 2021-01-07:
This is equivalent to disabling SmartShift on the config.

```
smartshift:
{
    on: false;
};
```

GabAlpha, 2021-01-10:
Oh thanks, I noticed at boot the scroll wheel always puts itself in freespin mode.
I'd rather it was on ratchet mode, can I change it?

BerenLuth, 2021-01-10:
I don't know if there's such an option, but I guess you can just put the smartshift on with the highest threshold value (it should be 255) so it should behave like a normal ratchet mode unless you try to spin it really really hard, but i guess at that a bit of smart shift won't hurt!

GabAlpha, 2021-01-18:
> I don't know if there's such an option, but I guess you can just put the smartshift on with the highest threshold value (it should be 255) so it should behave like a normal ratchet mode unless you try to spin it really really hard, but i guess at that a bit of smart shift won't hurt!

Thanks!

maxikd, 2021-02-13:
Just toggle SmartShift off and press the mouse button below the main scroll wheel. Don't this work?

NimaSakha, 2022-04-30:
if you go on the point and scroll tab turn smart shift off the go on the mouse tab and click on the button that on top of the mouse click the image not the actual mouse then scroll to the bottom and select none

eroldru, 2024-07-22:
> Just toggle SmartShift off and press the mouse button below the main scroll wheel. Don't this work?

3 years later, this worked for me. Thanks a ton. The wheel was driving me crazy, I probably pressed the button by accident.