PixlOne/logiops #191: Can't duplicate letters or numbers with a press of a button

ValentinPifu, 2021-01-09:
Hello,

Just installed Logiops and it's working so great.
But it's a bug somewhere. 
I tried to bind a gesture to write a phrase but if it's a duplicate letter/number, it'll won't show the duplicate letter/number.

`{
			direction: "Left";
                        mode: "OnRelease";
                        action =
                        {
                            type: "Keypress";
			    keys: ["KEY_T", "KEY_E", "KEY_S", "KEY_T", "KEY_1", "KEY_2", "KEY_1", "KEY_2"];
                        }
		    },
`

It's show this: TES12.

Thanks.
Have a nice day.

chamabreu, 2021-01-16:
I am searching a similar function.
I want to enter my emailadress via a gesture.
Tried to bind multiple actions or different keys: arguments, but this of course gives syntax error.

jfk88, 2021-01-18:
You can use xbindkeys with xdotools. It'll be cleaner as well.

chamabreu, 2021-01-19:
> You can use xbindkeys with xdotools. It'll be cleaner as well.

Hi thank you for that.
I often read about these, but never tried them cause I did not need it.
Now they are perfect for me - thanks.

@ValentinPifu This should help you also.
xbindkeys holds the shortcut which fires a "xdotool type 'myemailadress'".
This outputs the given string.
Now just fire the shortcut you defined with the logid action.

If you got problems with special characters like @ or € then look for setxkbmap. this helps to get the right layout.

ValentinPifu, 2021-01-19:
> Hi thank you for that.
> I often read about these, but never tried them cause I did not need it.
> Now they are perfect for me - thanks.
>
> @ValentinPifu This should help you also.
> xbindkeys holds the [...]

Can you help me with this? How did you do this? I installed xbindkeys and xdotools and more?

chamabreu, 2021-01-19:
- install xbindkeys and xdotools
xbindkeys lets you create custom shortcuts which then can for example shoot a terminal command.
You can find the first Steps under Configuration on this page: https://wiki.archlinux.org/index.php/Xbindkeys

for xdotool you can find here enough examples
https://wiki.ubuntuusers.de/xdotool/

the rest is try and error. but i guess this is another topic and not logiops anymore.

closed 2022-01-03 by vp-lab