PixlOne/logiops #192: Low-battery notifications

arigit, 2021-01-16:
Recent convert from solaar to logid. One feature I miss is a visual indication/reminder of low battery in my desktop environment (gnome shell)

Feature request: support configuration for:

- "low-battery notification threshold" e.g. 5%, 10%
- notification mode: once, recurrent
- recurrent notification period: 1 hour, 2, 6, 24 etc hours

Once the threshold is crossed, notifications would kick-in; with support for GTK/gnome shell and KDE.
Something simplistic that works, /usr/bin/notify-send "logid: low-battery threshold crossed"

maxikd, 2021-02-13:
I'm running Pop!_OS 20.10 and it gives me notifications when I start the PC. I also can check the battery status on Settings' Power tab:
[image]

arigit, 2021-02-14:
@maxikd thanks, I thought for a second you might be using bluetooth but I just checked in my system (ubuntu 20.04.2) and yes it's there already! In my case it shows 100% after a month of light use (mx master 3) but it does seem it works. Closing this issue.

closed 2021-02-14 by arigit

kris7t, 2021-02-14:
@arigit It seems hidpp doesn't report exact battery percentages. See e.g. the output of `upower -d`:

```
Device: /org/freedesktop/UPower/devices/battery_hidpp_battery_8
  native-path:          hidpp_battery_8
  model:                Wireless Mouse MX Master 3
  serial:               <<REDACTED>>
  power supply:         no
  updated:              Sun 14 Feb 2021 07:42:38 PM CET (51 seconds ago)
  has history:          yes
  has statistics:       yes
  mouse
    present:             yes
    rechargeable:        yes
    state:               fully-charged
    warning-level:       none
    battery-level:       full
    percentage:          100% (should be ignored)
    icon-name:          'battery-full-charged-symbolic'
```

The `(should be ignored)` note says that the percentage is not exact, only the `battery-level`. However, as the battery discharges, it will eventually turn to `low`, which is probably what Gnome reports to you.

This kernel patch (coming in 5.12) may be of interest: https://git.kernel.org/pub/scm/linux/kernel/git/hid/hid.git/commit/?h=for-next&id=e037acf0b1aed31cb5f3b09ccb602b4768c133d5 However, Filipe Laíns notes that:

> I tested this patch with a MX Anywhere 3, which supports the new feature. Since
> I don't have any device that doesn't support the state_of_charge parameter of
> this feature, I forced the MX Anywhere 3 to use the level information, instead
> of battery percentage, to test that part of the implementation.
> 
> I also tested with a MX Master 3, which supports the Battery Level Status
> (0x1000) feature, and a G703 Hero, which supports the Battery Voltage (0x1001)
feature, to make sure nothing broke there.

So it may be the case that the MX Master 3 just doesn't support reporting any more precise charge information

arigit, 2021-02-14:
@kris7t  great insight, thanks. I do recall that solaar was only able to report 100 / 50 / 20 / 10% battery levels for my older mx master 2, and nothing in-between, so perhaps this will be consistent with those 4 states. While I see 100% (and upower reports battery_level = full) in my case after 4 weeks of light use, in @maxikd screenshot above we see 10% so it seems gnome's indicator could be consistent with what solaar shows - I'll try and post an update here once I notice significant discharge

maxikd, 2021-02-22:
I use both the MX Master 3 and MX Keys for work and my personal PC. I use Windows for work and it does show the percentage on Windows' Settings. I'll try to check tomorrow what percentage the mouse and keyboard are, but I think they report more than just 100 / 50 / 20 / 10%

maxikd, 2021-02-23:
Specifically now I can't say if it has more than those levels. I'm on 50% on MX Keys and 20% on MX Master 3. But I'll keep checking for MX Keys on the following days.

[image]

maxikd, 2021-02-26:
Today I got the 5% notification

[image]

maxikd, 2021-02-26:
I think I got 5% on Linux too

[image]