PixlOne/logiops #194: M720 features are "not supported"

thansk, 2021-01-21:
I am running Ubuntu 20.10
I built it following the instructions in the repo.

My config file is the same as the example one with the name changed to `M720 Triathlon Multi-Device Mouse`

Here is the output:
```
[DEBUG] Unsupported device /dev/hidraw0 ignored
[INFO] Detected receiver at /dev/hidraw1
[INFO] Device found: M720 Triathlon Multi-Device Mouse on /dev/hidraw1:1
[WARN] /dev/hidraw1:1: DPI feature not found, cannot use CycleDPI action.
[WARN] /dev/hidraw1:1: SmartShift feature not found, cannot use ToggleSmartShift action.
[DEBUG] /dev/hidraw1:1 remappable buttons:
[DEBUG] CID  | reprog? | fn key? | mouse key? | gesture support?
[DEBUG] 0x50 |         |         | YES        | 
[DEBUG] 0x51 |         |         | YES        | 
[DEBUG] 0x52 | YES     |         | YES        | YES
[DEBUG] 0x53 | YES     |         | YES        | YES
[DEBUG] 0x56 | YES     |         | YES        | YES
[DEBUG] 0x5b | YES     |         | YES        | YES
[DEBUG] 0x5d | YES     |         | YES        | YES
[DEBUG] 0xd0 | YES     |         | YES        | YES
[DEBUG] 0xd7 | YES     |         |            | YES
[WARN] M720 Triathlon Multi-Device Mouse: CID 0xc3 does not exist.
[WARN] M720 Triathlon Multi-Device Mouse: CID 0xc4 does not exist.
[WARN] Error adding device /dev/hidraw2: std::exception
```

Am I doing something wrong?

idantene, 2021-01-22:
Hey, this actually helped me set up my M720!
Perhaps should be noted in the CIDs wiki, but for M720 the gesture button is actually at 0x00d0 (and not 0x00c3).

Friedrich42, 2021-05-30:
@PixlOne perhaps this should be closed

closed 2021-05-30 by PixlOne