PixlOne/logiops #196: MX Master 3 thumb button only send the Tab key

djfjeff, 2021-01-22:
I am not using logiops yet as I am still discovering the functionality of the mouse. One thing that I saw that was different on my system if the behaviour of the thumb button.

Everywhere I see that the thumb button should send CTRL+ALT+TAB but on my system (Fedora 33), xev report it only sending the Tab button
~~~
KeyPress event, serial 34, synthetic NO, window 0x2e00001,
    root 0x539, subw 0x0, time 166023709, (128,8), root:(2599,671),
    state 0x50, keycode 23 (keysym 0xff09, Tab), same_screen YES,
    XLookupString gives 1 bytes: (09) "	"
    XmbLookupString gives 1 bytes: (09) "	"
    XFilterEvent returns: False
~~~

I did not find anywhere that it should be the expected behaviour. Is it something from a new firmware (just bought the mouse)? And with that, should logiops works still?

GitJamz, 2021-02-05:
Im also having problems with the mx master 3 thumb button.
I have used logid for about 6 months. I can say that the thumb button HAS worked fine for a long time.
About 2-3 months ago it stopped working for about a month, and then started working again.
Now it is back to the same behavior. 

I dont know if it is related to software updates in the Distro or something? I run PopOS based on Ubuntu, with KDE Desktop.
Now as a test I disabled logid and I notice that the general shortcut for "display all windows on desktop" actually does not work so in my case it seems to not be a logid problem.

djfjeff, 2021-02-05:
Can you try to run `xev` from a terminal, click the mouse thumb button in it and paste the result here? I just want to see what key your thumb button is reporting on your side.

GitJamz, 2021-02-05:
> Can you try to run `xev` from a terminal, click the mouse thumb button in it and paste the result here? I just want to see what key your thumb button is reporting on your side.

Actually I started to troubleshoot and found that the mproblem was desktop composition in my KDE desktop.
I went to settings hardware display and clicked something there.
Now it works, so it was totally unrelated to logid